pschatzmann/arduino-audio-tools #903: ESP32 use internal ADC's and Internal DAC's at the same time

Radio-Builder, 2023-06-18:
I am using an ESP32 Devkit C and am trying to read audio from the internal DAC and pass it straight through to the internal ADC. ( Later I want to add a FIR filter in the middle ) but I can not get it to work. There are no examples so this is what I tried...

`/**
 * @file streams-analog-i2s.ino
 * @author Phil Schatzmann
 * @brief see https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/streams-adc-i2s/README.md
 * 
 * @author Phil Schatzmann
 * @copyright GPLv3
 */

#include "AudioTools.h"

AudioInfo info(44100,2, 16);
AnalogAudioStream in; 
AnalogAudioStream out;                        
StreamCopy copier(out, in); // copy in to out
ConverterAutoCenter<int16_t> converter;

// Arduino Setup
void setup(void) {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // TX on I2S_NUM_1 
  auto cfgTx = out.defaultConfig(TX_MODE);
  cfgTx.port_no = 0;
  cfgTx.copyFrom(info);
  out.begin(cfgTx);

  // RX automatically uses port 0 with pin GPIO34
  auto cfgRx = in.defaultConfig(RX_MODE);
  cfgTx.port_no = 0;
  cfgRx.copyFrom(info);
  in.begin(cfgRx);
 
}

// Arduino loop - copy data 
void loop() {
  //copier.copy(converter);
}`

**If I set "cfgTx.port_no = 1;" I get a core panic** 

[I] AudioTypes.h : 85 - sample_rate: 44100
[I] AudioTypes.h : 86 - channels: 2
[I] AudioTypes.h : 87 - bits_per_sample: 16
[I] AnalogAudioBase.h : 74 - analog left output pin: 25
[I] AnalogAudioBase.h : 75 - analog right output pin: 26
E (43) I2S: i2s_check_cfg_validity(1553): I2S built-in ADC/DAC only support on I2S0
E (50) I2S: i2s_driver_install(1975): I2S configuration is invalid
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400f503a  PS      : 0x00060e33  A0      : 0x800d5078  A1      : 0x3ffb25b0  
A2      : 0x3ffb9144  A3      : 0x3ffc2b48  A4      : 0x3ffc2b4c  A5      : 0x00060e23  
A6      : 0x00060e20  A7      : 0x00000001  A8      : 0x00000001  A9      : 0x00000000  
A10     : 0xffffffef  A11     : 0x3ffb88d4  A12     : 0x3ffb25c0  A13     : 0x0000000c  
A14     : 0x007bdb9c  A15     : 0x003fffff  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x400864f1  LEND    : 0x40086501  LCOUNT  : 0xfffffffb  

Backtrace: 0x400f5037:0x3ffb25b0 0x400d5075:0x3ffb25d0 0x400d5c29:0x3ffb2600 0x400d5e75:0x3ffb2630 0x400d1f26:0x3ffb2670 0x400d2691:0x3ffb26d0 0x400d2797:0x3ffb2730 0x400d40b2:0x3ffb2820

**If I set "cfgTx.port_no = 0;" I get a 2 Errors**
E (96) I2S: register I2S object to platform failed
[E] AnalogAudioESP32.h : 75 - begin - i2s_driver_install

This is the LOG

ELF file SHA256: 394f42e5f6294096

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[I] AnalogAudioBase.h : 59 - I2S_MODE_DAC_BUILT_IN
[I] AnalogAudio.h : 58 - bool audio_tools::AnalogAudioStream::begin(audio_tools::AnalogConfig)
[I] AnalogAudioESP32.h : 48 - virtual bool audio_tools::AnalogDriverESP32::begin(audio_tools::AnalogConfig)
[I] AudioTypes.h : 85 - sample_rate: 44100
[I] AudioTypes.h : 86 - channels: 2
[I] AudioTypes.h : 87 - bits_per_sample: 16
[I] AnalogAudioBase.h : 74 - analog left output pin: 25
[I] AnalogAudioBase.h : 75 - analog right output pin: 26
[I] AnalogAudioESP32.h : 106 - TX_MODE
[I] AnalogAudioBase.h : 56 - I2S_MODE_ADC_BUILT_IN
[I] AnalogAudio.h : 58 - bool audio_tools::AnalogAudioStream::begin(audio_tools::AnalogConfig)
[I] AnalogAudioESP32.h : 48 - virtual bool audio_tools::AnalogDriverESP32::begin(audio_tools::AnalogConfig)
[I] AudioTypes.h : 85 - sample_rate: 44100
[I] AudioTypes.h : 86 - channels: 2
[I] AudioTypes.h : 87 - bits_per_sample: 16
[I] AnalogAudioBase.h : 77 - input pin1: 34
[I] AnalogAudioBase.h : 79 - input pin2: 0
E (96) I2S: register I2S object to platform failed
[E] AnalogAudioESP32.h : 75 - begin - i2s_driver_install

pschatzmann, 2023-06-18:
There are no examples because this does not work!
Both the internal DAC and the internal ADC are working only on port 0, so you can't use them at the same time

closed 2023-06-18 by pschatzmann

Radio-Builder, 2023-06-18:
OK,,, Thanks very much for your reply.

pschatzmann, 2023-06-18:
Instead of the output to the internal DAC you could try to use the PWM output

Radio-Builder, 2023-06-18:
Thanks for the idea... It would be a solution for me.

This is what I tried... but when I use your sinewave code I get 0-100% approx change in the PWM outout but when I use the internal ADC it does vary with the input 0-1V Sinewave but only about 5% , varies only approx +/- 2-3%  around 90%.

```/**
 * @file streams-generator-pwm.ino
 * @author Phil Schatzmann
 * @brief see https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/streams-generator-pwm/README.md 
 * @author Phil Schatzmann
 * @copyright GPLv3
 */
 
#include "AudioTools.h"

//Pins pins = {22, 23};
AudioInfo info(8000, 1, 16);
//SineWaveGenerator<int16_t> sineWave(32000); // subclass of SoundGenerator with max amplitude of 32000
//GeneratedSoundStream<int16_t> sound(sineWave);  // Stream generated from sine wave

AnalogAudioStream in; 
PWMAudioOutput pwm;                  
StreamCopy copier(pwm, in);    // copy in to out

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Warning);  

  // setup sine wave
  //sineWave.begin(info, N_B4);

  // setup PWM output
  auto config = pwm.defaultConfig();
  config.copyFrom(info);
  //config.resolution = 11;
  pwm.begin(config);

  //   // RX automatically uses port 0 with pin GPIO34
  auto cfgRx = in.defaultConfig(RX_MODE);
  cfgRx.copyFrom(info);
  in.begin(cfgRx);

}

void loop(){
  copier.copy();
}```

pschatzmann, 2023-06-18:
Did you try to display the output of the AnalogAudioStream.? 
I think you need to center the singal...