pschatzmann/arduino-audio-tools #908: Error with AnalogAudioStream

zaretskii, 2023-06-26:
Greetings. 
I have a Lolin s2 mini board (**ESP32-s2**). 
When I try to compile the "_arduino-audio-tools/examples/examples-basic-api/base-adc-serial/**base-adc-serial.ino**_" sketch from the library examples, I get the error: 

```
C:\Temp\.arduinoIDE-unsaved2023526-2928-7kzf0t.6vbab\sketch_jun26a\sketch_jun26a.ino:18:1: error: ' AnalogAudioStream' does not name a type
 AnalogAudioStream adc;
 ^~~~~~~~~~~~~~~~~
C:\Temp\.arduinoIDE-unsaved2023526-2928-7kzf0t.6vbab\sketch_jun26a\sketch_jun26a.ino:24:15: error: 'adc' was not declared in this scope
 auto config = adc.defaultConfig(RX_MODE);
               ^~~
C:\Temp\.arduinoIDE-unsaved2023526-2928-7kzf0t.6vbab\sketch_jun26a\sketch_jun26a.ino:25:1: error: 'config' does not name a type; did you mean 'confstr'?
 config.pin_data = 3;
 ^~~~~~
 confstr
C:\Temp\.arduinoIDE-unsaved2023526-2928-7kzf0t.6vbab\sketch_jun26a\sketch_jun26a.ino: In function 'void setup()':
C:\Temp\.arduinoIDE-unsaved2023526-2928-7kzf0t.6vbab\sketch_jun26a\sketch_jun26a.ino:33:3: error: 'adc' was not declared in this scope
   adc.begin(config);
   ^~~
C:\Temp\.arduinoIDE-unsaved2023526-2928-7kzf0t.6vbab\sketch_jun26a\sketch_jun26a.ino: In function 'void loop()':
C:\Temp\.arduinoIDE-unsaved2023526-2928-7kzf0t.6vbab\sketch_jun26a\sketch_jun26a.ino:39:16: error: 'adc' was not declared in this scope
   size_t len ​​= adc.readBytes(buffer, max_buffer_len);
                ^~~

exit status 1

Compilation error: 'AnalogAudioStream' does not name a type.
```

I didn't change anything in the example. 
I would be very grateful for tips.

pschatzmann, 2023-06-26:
Only the regular ESP32 has proper ADC and DAC support: 
https://github.com/pschatzmann/arduino-audio-tools/wiki/Supported-Architectures

closed 2023-06-26 by pschatzmann

zaretskii, 2023-06-26:
Thanks for the answer. 
I was inattentive while studying the library.