pschatzmann/arduino-audio-tools #911: Feature request: Example for how create a HLS stream m3u8 player

DrewBatchelor, 2023-07-02:
Hi Phil,
Thank you for all the work you have put into making this amazing library, and thank you for generously sharing it. 
It is amazing. 

I have been using it to make my main kitchen radio, but last month BBC have changed how they stream their internet radio, so mp3 streams are no longer available, only m3u8 like this one: 
"http://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/uk/sbr_high/ak/bbc_6music.m3u8"

I can see that you have created a [HLS player in experiments](https://github.com/pschatzmann/arduino-audio-tools/blob/main/src/Experiments/HLSStream.h), which looks like it will solve this problem, but I can't quite work out how to use it. 

Would you please share an example that shows how to use this to create a simple streaming music player. Something like the [URL player example ](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-player/player-url-i2s/player-url-i2s.ino) for playing a m3u8 live stream.

Many thanks 

Drew

pschatzmann, 2023-07-03:
BBC decided to provide the audio as m4a. Unfortunatly I do not have any codec which would support this and information how to parse this is really hard to find.

Any help is appreciated...

DrewBatchelor, 2023-07-03:
Hi Phil, 
Thank you for the reply, this level of coding is well beyond my understanding, 
In searching for a solution, I found 2 possible similar projects that might be of help.

Frank Boesing AudioLibrary on a Teensy looks promising: https://github.com/FrankBoesing/Arduino-Teensy-Codec-lib
"Audiolibrary plugin, plays up to 320 kbps MP3, MP4(AAC), **M4A(AAC),** AAC-Raw or FLAC in software without external hardware - with only 48MHz."
(Frank B [described the](https://forum.pjrc.com/threads/27059-MP3-Player-Lib-with-example?p=58198&viewfull=1#post58198)  "*.m4a .... file format is horror :-) ( "ISO14496-14" if anyone wants to study it)"

And Yokohama-Miyazawa on a M5 Stack has and M5HLSPlayer: https://github.com/Yokohama-Miyazawa/M5HLSPlayer
Although I'm not clear if this one is also solving the m4a Problem.

Kind Regards 
Drew

joba-1, 2023-07-03:
3 attempts at any help:

Espressif provides an aac decoder with their ADF: https://github.com/espressif/esp-adf-libs/blob/master/esp_codec/include/codec/aac_decoder.h

m4a is just an envelope for a codec, usually aac. Have not checked for what bbc actually uses.

faad is one open source project that can decode m4a and aac, so probably contains code to study the difference.

pschatzmann, 2023-07-04:
Yes, I also found franks approach most promising, but I am not sure how reliable it will be. A test file that I donwloaded did not seem to have the same atoms that he was relying on. He also has an easier problem because he can rely on the file API when doing the parsing. Currently I am in the hospital,, so do not expect to see any quick progress.

pschatzmann, 2023-09-10:
Things are not working completely yet, but I decided on the following design:
- HLSStream: provides the stream of encoded data. Usually with Content-Type: video/MP2T, so this is of the mpeg-ts (=MTS) format.
- MTSDecoder: is demuxing the AAC data using the https://github.com/pschatzmann/arduino-tsdemux library. This can be just forwarded to a AAC decoder to provide pcm data.

A draft implemntation of these classes have been committed.

pschatzmann, 2023-09-12:
> faad is one open source project that can decode m4a and aac, so probably contains code to study the difference
Just made a test by sending the data directly to FAAD: it can't handle it! I am getting a Decoding error: Channel coupling not yet implemented. So there seems to be no direct path w/o mpeg-ts.

pschatzmann, 2023-09-12:
Today I spent some time to investigate why the sketch above is not working: I am stunned - the same code with the same data on the desktop is working, but on the ESP32 it's not.

This will be difficult to find...

pschatzmann, 2023-09-13:
https://github.com/frejoel/tsdemux/issues/10

DrewBatchelor, 2023-09-13:
@pschatzmann Thank you. Thank you.

DrewBatchelor, 2023-12-12:
Hi Phil, I'm not sure which example file this is, is it this one and is it working enough to be useful?
https://github.com/pschatzmann/arduino-audio-tools/blob/ae72af696865c42bd3e929953bb36016189529b5/examples/examples-stream/streams-url_mts-hex/streams-url_mts-hex.ino#L12
Thank you

pschatzmann, 2023-12-12:
No hls does not work properly yet

pschatzmann, 2024-02-13:
HLS is not working (reliably) yet. 
I was coming to the conclusion that it will be too slow when using a simple sequential approach.

pschatzmann, 2024-02-13:
The example sends the output to the display! So there can not be any sound...
It might be no problem on an ESP32, but then the solution is not portable any more!

FatherMarco1971, 2024-02-13:
_The example sends the output to the display! So there can not be any sound..._
Yes i'm trying to stream  it i2s just to try it and study. But anyway you say the result it will be not good enough

I see your point regarding portability.

z-l-p, 2024-04-28:
I am also very interested in HLS streams (mainly because BBC discontinued their Icecast MP3 streams). A quick note: Aside from issues of portability, the M5HLSPlayer repo also seems quite broken and hasn't had any commits in a year. HLS is a tough one!

closed 2025-03-19 by pschatzmann

lidense, 2025-03-19:
> I finally got my MTSDecoder working correctly! BBC is reporting video/MP2T which stands for MPEG-TS (MTS). My MTSDecoder is just extracting the ADTS formatted AAC. You can choose to forward it to the ADTS decoder which filters out the invalid aac packets, but this is usually not necessary.
>
> I have updated the [buffered example with the Multidecoder](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-communication/hls/hls-buffer-i2s/hls-buffer-i2s.ino) to handle this case as well
>
> https://github.com/pschatzmann/arduino-audio-tools/issues/911#issuecomment-2737881035

Great, thanks for your hard work.
Dave

----- Mail d'origine -----

z-l-p, 2025-03-20:
This is wonderful! I look forward to adding multidecoder to my web radio receiver to decode BBC streams. BBC recently announced website / app changes for listeners outside the UK, but they didn't specifically mention their raw streams. I assume they will geo-block them soon: https://www.bbc.co.uk/sounds/help/questions/listening-outside-the-uk/outside-uk-changes

pschatzmann, 2025-03-20:
In regard of BBC, I found [this link.](https://gist.github.com/bpsib/67089b959e4fa898af69fea59ad74bc3)...

lidense, 2025-03-20:
OK thanks, actually some of these links seem to change over time which requires reprogramming every few months...

----- Mail original -----