pschatzmann/arduino-audio-tools #915: CodecMP3Helix compile error

AlfredoCubitos, 2023-07-06:
Hi,

I want to compile the following code:
```
#include "BluetoothA2DPSink.h"
#include "AudioTools.h"
#include "AudioCodecs/CodecMP3Helix.h"

BluetoothA2DPSink a2dp_sink;

AnalogAudioStream analog; // final output of decoded stream
EncodedAudioStream dec(&analog, new MP3DecoderHelix()); // Decoding stream

void data_received_callback(const uint8_t *data, uint32_t length) {
  //Serial.println("Data packet received");
  int16_t *samples = (int16_t*) data;
  uint32_t sample_count = length/2;
}

void setup() {
  Serial.begin(115200);
  Serial.println("Bluetooth Audio Receiver");

    static const i2s_config_t i2s_config = {
        .mode = (i2s_mode_t) (I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_DAC_BUILT_IN),
        .sample_rate = 44100, // corrected by info from bluetooth
        .bits_per_sample = (i2s_bits_per_sample_t) 16, /* the DAC module will only take the 8bits from MSB */
        .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
        .communication_format = (i2s_comm_format_t)I2S_COMM_FORMAT_STAND_MSB,
        .intr_alloc_flags = 0, // default interrupt priority
        .dma_buf_count = 8,
        .dma_buf_len = 64,
        .use_apll = false
    };

    AudioLogger::instance().begin(Serial, AudioLogger::Info);

    a2dp_sink.set_i2s_config(i2s_config);
    a2dp_sink.start("MyMusic");
   // a2dp_sink.set_on_data_received(data_received_callback);

}

void loop() {
}
```
I get this error:

```
In file included from /tmp/.arduinoIDE-unsaved202366-16013-1cm8w8g.8u1w/sketch_jul6b/sketch_jul6b.ino:3:
/Devel/Arduino/scripts/libraries/audio-tools/src/AudioCodecs/CodecMP3Helix.h: In member function 'virtual void audio_tools::MP3DecoderHelix::setNotifyAudioChange(audio_tools::AudioInfoDependent&)':
/Devel/Arduino/scripts/libraries/audio-tools/src/AudioCodecs/CodecMP3Helix.h:127:65: error: invalid conversion from 'void (*)(MP3FrameInfo&)' {aka 'void (*)(_MP3FrameInfo&)'} to 'libhelix::MP3InfoCallback' {aka 'void (*)(_MP3FrameInfo&, void*)'} [-fpermissive]
             if (mp3!=nullptr)  mp3->setInfoCallback(infoCallback);
                                                                 ^
In file included from /Devel/Arduino/scripts/libraries/audio-tools/src/AudioCodecs/CodecMP3Helix.h:6,
                 from /tmp/.arduinoIDE-unsaved202366-16013-1cm8w8g.8u1w/sketch_jul6b/sketch_jul6b.ino:3:
/Devel/Arduino/scripts/libraries/libhelix/src/MP3DecoderHelix.h:51:46: note:   initializing argument 1 of 'void libhelix::MP3DecoderHelix::setInfoCallback(libhelix::MP3InfoCallback, void*)'
         void setInfoCallback(MP3InfoCallback cb, void* caller=nullptr){
                              ~~~~~~~~~~~~~~~~^~

exit status 1

Compilation error: exit status 1
```
Any idea where the problem is?

I'm trying to send bluetooth data to an analog amplifier PAM8403.
The example `Output to the Internal DAC` does not work. :-( Means, there is no useful sound output.
Can you please give an example how to use `a2dp_sink` data with an analog amplifier?

pschatzmann, 2023-07-07:
Make sure you use the latest versions of the AudioTools and LibHelix. I guess you are using one updated version with an old incompatible one...

closed 2023-07-07 by pschatzmann

AlfredoCubitos, 2023-07-07:
I installed the zip from the git source repository and not from  release repro. 
After installing the release version it works :-)
Thank you ! :+1: