pschatzmann/arduino-audio-tools #920: icecast stream error

AlfredoCubitos, 2023-07-11:
Hi,
I tried to play this URL http://icecast.ndr.de/ndr/ndr2/niedersachsen/mp3/128/stream.mp3
I received the following error:
```
[W] HttpHeader.h : 233 - Waiting for data...
[W] HttpRequest.h : 174 - no CONTENT_LENGTH found in reply
[E] URLStream.h : 347 - Error code recieved ... stop waiting for reply
```
But this URL "http://stream.srg-ssr.ch/m/rsj/mp3_128" works fine

Any idea whats going wrong?

pschatzmann, 2023-07-11:
Obviously the url does not provide a proper reply. You will need to test this with something like Advanced REST client in order to find out what additional information you might need to provide...

AlfredoCubitos, 2023-07-11:
well I've got something like this:
`Redirected to: http://f131.rndfnk.com/ard/ndr/ndr2/niedersachsen/mp3/128/stream.mp3?cid=01FBQ2CWDYWJHGF4QAJ0SVV730&sid=2SQYi2aSNg5XC2d2u55L9kQfFEM&token=-cWDHJepmVaL9lIorPo9M3ZaMcJuvvc0mYVkkcWuLP4&tvf=R3WBylDlcBdmMTMxLnJuZGZuay5jb20`

I suppose this is a authentication problem with the webtoken, therefore I get no content.
what do you think about this?

pschatzmann, 2023-07-11:
Did you try to enter the redirected URL?
You can also set the log level to debug to see if the redirection is working properly...

pschatzmann, 2023-07-11:
Did you try https://icecast.ndr.de/ndr/ndr2/niedersachsen/mp3/128/stream.mp3 ? The http url seems to be invalid
You can also set the log level to debug to see if the redirects are working...

AlfredoCubitos, 2023-07-12:
yes.
As you can see, it is as I supposed. The client have to make a bearer authentication before getting the stream.
When you call `http://icecast.ndr.de/ndr/ndr2/niedersachsen/mp3/128/stream.mp3` the request is always redirected to the icecast url by the server. The redirecting url has a query parameter `token`. The client has to take this token, put it in to the request header to the given url as `Authorization Bearer <token>`.

This authorization technology is relatively new and is increasingly being used. The advantage is that on the server side the client session handling is simplified, because you can do it without cookies. And it prevents user session stealing.

More information about this you can find [here](https://auth0.com/docs/secure/tokens/access-tokens/use-access-tokens)

So, what we we need is a webclient which can handle that....

```
[I] AnalogAudio.h : 58 - bool audio_tools::AnalogAudioStream::begin(audio_tools::AnalogConfig)
[I] AnalogAudioESP32.h : 48 - virtual bool audio_tools::AnalogDriverESP32::begin(audio_tools::AnalogConfig)
[I] AudioTypes.h : 85 - sample_rate: 44100
[I] AudioTypes.h : 86 - channels: 2
[I] AudioTypes.h : 87 - bits_per_sample: 16
[I] AnalogAudioBase.h : 74 - analog left output pin: 25
[I] AnalogAudioBase.h : 75 - analog right output pin: 26
[I] AnalogAudioESP32.h : 106 - TX_MODE
Gespeicherte Station 0 von 25
[I] AudioPlayer.h : 408 - virtual void audio_tools::AudioPlayer::setMetadataCallback(void (*)(audio_tools::MetaDataType, const char*, int), audio_tools::ID3TypeSelection)
[I] AudioSource.h : 231 - virtual bool audio_tools::AudioSourceURL::setMetadataCallback(void (*)(audio_tools::MetaDataType, const char*, int), audio_tools::ID3TypeSelection)
[I] AudioPlayer.h : 412 - Using ICY Metadata
[I] AudioPlayer.h : 351 - setVolume(1.000000)
[I] VolumeStream.h : 191 - setVolume: 1.000000
[I] VolumeStream.h : 191 - setVolume: 1.000000
[I] AudioTypes.h : 85 - sample_rate: 0
[I] AudioTypes.h : 86 - channels: 0
[I] AudioTypes.h : 87 - bits_per_sample: 16
[I] MetaDataID3.h : 564 - virtual void audio_tools::MetaDataID3::begin()
[I] AudioSource.h : 175 - selectStream: 0/3 -> http://icecast.ndr.de/ndr/ndr2/niedersachsen/mp3/128/stream.mp3
[I] URLStream.h : 89 - virtual bool audio_tools::URLStream::begin(const char*, const char*, MethodID, const char*, const char*): http://icecast.ndr.de/ndr/ndr2/niedersachsen/mp3/128/stream.mp3
[I] Url.h : 74 - Url::parse
[I] Url.h : 112 - pathStart->21
[I] Url.h : 113 - QueryStart->-1
[I] Url.h : 114 - url->http://icecast.ndr.de/ndr/ndr2/niedersachsen/mp3/128/stream.mp3
[I] Url.h : 115 - host->icecast.ndr.de
[I] Url.h : 116 - protocol->http
[I] Url.h : 117 - path->/ndr/ndr2/niedersachsen/mp3/128/stream.mp3
[I] Url.h : 118 - port->80
[I] URLStream.h : 325 - void audio_tools::URLStream::login()
.
[I] URLStream.h : 300 - WiFiClient
[I] HttpRequest.h : 227 - process connecting to host icecast.ndr.de port 80
[I] HttpRequest.h : 213 - connected 1 timeout 60000
[I] HttpRequest.h : 238 - Free heap: 198408
[I] HttpHeader.h : 361 - HttpRequestHeader::setValues - path: /ndr/ndr2/niedersachsen/mp3/128/stream.mp3
[I] HttpHeader.h : 127 - Content-Length 0
[I] HttpHeader.h : 258 - HttpHeader::write
[I] HttpHeader.h : 370 - HttpRequestHeader::write1stLine
[I] HttpHeader.h : 385 - -> GET /ndr/ndr2/niedersachsen/mp3/128/stream.mp3 HTTP/1.1
[I] HttpHeader.h : 193 -  -> Icy-MetaData: 1 
[I] HttpHeader.h : 193 -  -> Content-Length: 0 
[I] HttpHeader.h : 193 -  -> Host: icecast.ndr.de 
[I] HttpHeader.h : 193 -  -> Connection: close 
[I] HttpHeader.h : 193 -  -> Accept: audio/mpeg 
[I] HttpHeader.h : 301 -  -> <CR LF> 
[I] HttpRequest.h : 275 - Request written ... waiting for reply
[I] HttpHeader.h : 226 - HttpHeader::read
[W] HttpHeader.h : 233 - Waiting for data...
[I] HttpHeader.h : 163 - HttpHeader::readLine -> HTTP/1.1 302 Found
[I] HttpHeader.h : 163 - HttpHeader::readLine -> Date: Wed, 12 Jul 2023 08:07:20 GMT
[I] HttpHeader.h : 163 - HttpHeader::readLine -> Content-Length: 0
[I] HttpHeader.h : 163 - HttpHeader::readLine -> Connection: close
[I] HttpHeader.h : 163 - HttpHeader::readLine -> Access-Control-Allow-Methods: GET, OPTIONS, HEAD
[I] HttpHeader.h : 163 - HttpHeader::readLine -> Access-Control-Allow-Origin: *
[I] HttpHeader.h : 163 - HttpHeader::readLine -> Cache-Control: no-cache, no-store
[I] HttpHeader.h : 163 - HttpHeader::readLine -> Location: http://f121.rndfnk.com/ard/ndr/ndr2/niedersachsen/mp3/128/stream.mp3?cid=01FBQ2CWDYWJHGF4QAJ0SVV730&sid=2SSoFbaQxE7mLqOeAT8pWdEiQDh&token=nItTCUV94X4G4OmS3QoNfQttUEpJ-htbgCAjKxfYlvI&tvf=Su7oJe4jcRdmMTIxLnJuZGZuay5jb20
[I] HttpHeader.h : 163 - HttpHeader::readLine -> 
[W] URLStream.h : 267 - Redirected to: http://f121.rndfnk.com/ard/ndr/ndr2/niedersachsen/mp3/128/stream.mp3?cid=01FBQ2CWDYWJHGF4QAJ0SVV730&sid=2SSoFbaQxE7mLqOeAT8pWdEiQDh&token=nItTCUV94X4G4OmS3QoNfQttUEpJ-htbgCAjKxfYlvI&tvf=Su7oJe4jcRdmMTIxLnJuZGZuay5jb20
[... 32 lines cut ...]
[I] HttpHeader.h : 163 - HttpHeader::readLine -> <!DOCTYPE html>
[I] HttpHeader.h : 163 - HttpHeader::readLine -> <html lang="en">
[I] HttpHeader.h : 163 - HttpHeader::readLine ->   <head>
[I] HttpHeader.h : 163 - HttpHeader::readLine ->     <meta charset="UTF-8">
[I] HttpHeader.h : 163 - HttpHeader::readLine ->     <meta name="viewport" content="width=device-width,initial-scale=1">
[I] HttpHeader.h : 163 - HttpHeader::readLine -> 	<link rel="stylesheet" href="/css/rsas.css">
[I] HttpHeader.h : 163 - HttpHeader::readLine ->     <title>Unauthorized</title>
[... cut ...]
[I] HttpHeader.h : 163 - HttpHeader::readLine -> 			<p class="mute">Error code: 401 Unauthorized</p>
[I] HttpHeader.h : 163 - HttpHeader::readLine -> 		</div>
[I] HttpHeader.h : 163 - HttpHeader::readLine -> 	</div>
[I] HttpHeader.h : 163 - HttpHeader::readLine ->   </body>
[I] HttpHeader.h : 163 - HttpHeader::readLine -> </html
[W] HttpRequest.h : 174 - no CONTENT_LENGTH found in reply
[I] URLStream.h : 110 - size: 0
[I] URLStream.h : 341 - virtual bool audio_tools::URLStream::waitForData()
[I] URLStream.h : 343 - Request written ... waiting for reply
```

pschatzmann, 2023-07-12:
The request length was limited to 200 characters. In your case this was getting longer, so it was cut off to
tvf=Su7oJe4jcRdmMTIxLnJuZGZuay5jb
instead
tvf=Su7oJe4jcRdmMTIxLnJuZGZuay5jb20
I increased to limit to 300 chars

closed 2023-07-12 by pschatzmann

AlfredoCubitos, 2023-07-12:
aaah, ok it works :-) :+1: