pschatzmann/arduino-audio-tools #927: Where in the galaxy is StarWars30.h?

joshco, 2023-07-16:
I'm trying to use the example in the README

````
#include "AudioTools.h"
#include "StarWars30.h"

uint8_t channels = 2;
uint16_t sample_rate = 22050;
....
````

But I can't locate StarWars30.h.  Where can I find that?

pschatzmann, 2023-07-16:
Why don't you just use the example ? https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-stream/streams-memory_raw-i2s

closed 2023-07-16 by pschatzmann

joshco, 2023-07-16:
That's a great idea.  I will try that.

dhionnystrauss, 2025-01-24:
Hello Prof. Schatzmann, 
A very nice example I could run nicely. I attached a Photo of the setup for ESP32 and DAC Board UDA1334. 
I was trying to convert a  .WAV File to Raw using the same format for StarWars30.h but could not do it.
Could you please indicate how to convert the File, the software recommended for the File conversion, and the Exact Specifications of the .h file?
Thank you so much. 
Regards,
Dhionny Strauss

[image]

[image]