pschatzmann/arduino-audio-tools #939: portaudio.h is missing

wuppergit, 2023-07-23:
File portaudio.h is missing needed from PortAudioStream.h (e.g.  examples -> examples-desktop -> generator)

pschatzmann, 2023-07-23:
I can't reproduce your issue.
What exectly did you try to do ?

closed 2023-07-23 by pschatzmann

wuppergit, 2023-07-23:
Sorry; I was a little bit short.
It's a compile error

"Arduino: 1.8.19 (Windows 10), Board: "ESP32 Dev Module, Disabled,
Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz
(WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None, Disabled"
Mehrere Bibliotheken wurden für "WiFi.h" gefunden
 Benutzt:
C:\Users\udobr_9swkb8k\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.9\libraries\WiFi
 Nicht benutzt:
C:\Users\udobr_9swkb8k\Documents\arduino-vers-1.8.19\arduino-1.8.19\libraries\WiFi
In file included from
C:\Users\udobr_9swkb8k\Documents\Arduino\libraries\arduino-audio-tools-main\examples\examples-desktop\generator\generator.ino:2:

C:\Users\udobr_9swkb8k\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/PortAudioStream.h:8:10:
fatal error: portaudio.h: No such file or directory
 #include "portaudio.h"
          ^~~~~~~~~~~~~
compilation terminated.

exit status 1

Fehler beim Kompilieren für das Board ESP32 Dev Module.

No issues while compiling other sketches.

Great work

pschatzmann, 2023-07-23:
I think you forgot to read the Readme.
This is supposed to be built on the desktop with cmake and not in Arduino!