pschatzmann/arduino-audio-tools #942: Problems with example "streams-stk_loop-audiokit.ino"

KontinuumLab, 2023-07-24:
Hi.
Thanks for checking my recent issue on the STK repository.
Made me realize that you're using the examples from "arduino-audio-tools", and NOT from "Arduino-STK"
(I was trying to run "MemoryLoop" from "Arduino-STK")

So now I tried to compile the example you linked in your other reply, but it's also not working. 
( This is "streams-stk_loop-audiokit.ino")

Here's the error message:
```
In file included from C:\Users\PC\AppData\Local\Temp\.arduinoIDE-unsaved2023624-3032-1f6naqd.ehjl\streams-stk_loop-audiokit\streams-stk_loop-audiokit.ino:12:
c:\Users\PC\Documents\Arduino\libraries\audio-tools\src/AudioLibs/AudioKit.h:4:10: fatal error: AudioKitHAL.h: No such file or directory
 #include "AudioKitHAL.h"
          ^~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1
```
So "AudioKit.h" is looking for "AudioKitHAL.h", and I haven't been able to find this file anywhere.

Thanks in advance for any help, and sorry to keep bugging you with this.
Jeppe
(Just a reminder, I'm trying to make instruments using a Wemos Lolin32 Lite, and a PCM5102 DAC.)

KontinuumLab, 2023-07-24:
Sorry for the confusion.
My original issue, for the Arduino-STK example, was posted on the arduino-stk repository. But in your reply you linked an example from this repository, so now I posted here.
I'm still confused about the connection between STK and the audio tools, and I might be in over my head trying to make this library work for my project, but I'll keep trying to figure it out with the info you provided.
Thanks so much,
Jeppe

pschatzmann, 2023-07-24:
Ok, my mistake. I did not notice that this was originally filed for [arduino-stk](https://github.com/pschatzmann/arduino-stk).
This was my first project which used to be stand alone.

Later on I decided to make it part of the Audio Tools so that I do not need to maintain the i2s output functionality in multiple projects

closed 2023-08-06 by pschatzmann