pschatzmann/arduino-audio-tools #946: FormatConverterStream crashes at end of file

LinusHeu, 2023-07-26:
```ino
#include <AudioTools.h>
#include "StarWars30.h"

AudioInfo infoIn(22050, 1, 16);
AudioInfo infoOut(22050,1,24);
I2SStream i2s;
MemoryStream music(StarWars30_raw, StarWars30_raw_len);
FormatConverterStream converter(music);
StreamCopy copier(i2s, converter);

void setup(){
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN,LOW);
    Serial.begin(115200);
    while(!Serial);
    delay(5000);
    Serial.println("Hallo");
    digitalWrite(LED_BUILTIN,HIGH);

    AudioLogger::instance().begin(Serial, AudioLogger::Debug);

    converter.begin(infoIn,infoOut);
    auto config = i2s.defaultConfig(TX_MODE);
    config.copyFrom(infoOut);
    if(!i2s.begin(config)){
      while(1){
        
      }
    }

}

void loop(){
    if (!copier.copy()){
      i2s.end();
      digitalWrite(LED_BUILTIN, LOW);//this LED never turns off
      stop();
    }
}
```
Hardware: Raspberry Pi Pico W, Adafruit MAX98357

It plays the file fine, but at the end I think it crashes the Pico. It creates ear-deafening noise and the LED stays on.

Debug-Log:
[device-monitor-230726-213752shortened.log](https://github.com/pschatzmann/arduino-audio-tools/files/12176734/device-monitor-230726-213752shortened.log)

LinusHeu, 2023-07-27:
Based on where the log stops, I suspect this assert might be it:
https://github.com/pschatzmann/arduino-audio-tools/blob/f8c5c0b350e7966587dbc7a85541c3aa97cc93a7/src/AudioTools/ResampleStream.h#L52

pschatzmann, 2023-07-27:
Did you try to comment it out ?

LinusHeu, 2023-07-27:
Just tried it. There is no ear-deafening noise anymore. Instead, there is silence with a pop about every second and the LED still doesn't turn off, indicating that playback doesn't properly stop. 

https://github.com/pschatzmann/arduino-audio-tools/blob/f8c5c0b350e7966587dbc7a85541c3aa97cc93a7/src/AudioTools/ResampleStream.h#L48-L57

To be honest, I don't understand how that entire Converter, Resampling, etc part of your library works, but it seems like at least that read_size in line 54 requires the assert. Maybe there need to be some more changes to have it properly handle the end of streams (have it properly handle only reading <x bytes when it expected x).

pschatzmann, 2023-07-27:
I think you can try to avoid this problem by applying the converter on the output side. 
This should also be more efficient...

PR #950 (2023-08-02): FormatConverterStream: fix crash at end of file
Fixes #946 according to your suggestion https://github.com/pschatzmann/arduino-audio-tools/issues/946#issuecomment-1654111367

To be honest, I'm not familiar with the entire code enough to fully say whether this is just a hacky workaround or a perfect fix.
However, with my example, I can confirm it works!

Code to reproduce: See issue
Before: See issue
After:
Plays entire file fine. log ends reasonably. Ctrl+F counting all
`[I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops` lines
+one additional line `[I] StreamCopy [... cut]
closed 2023-08-02 by pschatzmann