pschatzmann/arduino-audio-tools #948: AudioLogger::begin() takes a Stream instead of Print object

roleroz, 2023-07-30:
I'm working on a logging library that will handle both serial as well as network logging, and I'm trying to use that with this library. For this I'm injecting it into AudioLogger::instance().begin(). While doing this I realized that the first parameter of that function (where I need to add my new logging library object) is a Stream object instead of a Print object. Stream inherits all the output functionality from Print, and as far as I can tell this object is only planned to be an output system. This makes me wonder if there is any reason for it to be a Stream object instead of a Print one

If using a Print object is a possibility, I'll happily write a small PR to change the object type

pschatzmann, 2023-07-30:
I agree that Print would be better here.

roleroz, 2023-07-30:
https://github.com/pschatzmann/arduino-audio-tools/pull/949 created for this

closed 2023-07-31 by roleroz