pschatzmann/arduino-audio-tools #952: A2DPStream needs too much program memory

framik80, 2023-07-31:
I'm trying to use the library to use the ESP32 to play different files from the SD card on a BT speaker depending on the button pressed.

My little sketch works with your May 20 library (91% program memory)
But when I used the current version, the memory is exceeded (109% program memory).

It seems that A2DPStream is too big now.

> The sketch uses 1439525 bytes (109%) of program memory. The maximum is 1310720 bytes.

pschatzmann, 2023-07-31:
I don't understand why you file this as an issue: You obviously just fogot to set the right partition scheme

closed 2023-07-31 by pschatzmann

gauravacad, 2023-09-06:
Better you set the partition scheme as No OTA (2 MB app/2 MD spiff) 
Am i right ?

pschatzmann, 2023-09-06:
Only if you intent to use SPIFF. 
I am usually using Huge APP