pschatzmann/arduino-audio-tools #955: error after upload fresh library

aqildad-create, 2023-08-04:
every thing was working on previous version but after upload fresh library  getting below 

[image]

pschatzmann, 2023-08-04:
Please don't provide pictures but the proper stacktrace in text. 
W/o line numbers I can't really help!

aqildad-create, 2023-08-04:
All right but I don't have the text right now because I have delete the latest version and copying previous version for trouble shooting . 
But latest version have some channel conversing issues I guess.

aqildad-create, 2023-08-04:
I confirm now that old version is again working.

pschatzmann, 2023-08-05:
I am closing this issue since I can't reproduce it myself. 
I guess it might be related to [this correction](https://github.com/pschatzmann/arduino-audio-tools/commit/1a91735712700df53e954341ccc101f22b1f9240)...

Feel free to reopen it when you are able to provide the necessary information.

Or even better, if you see where it is crashing in your use case you can try to provide the correction yourself and share it

closed 2023-08-05 by pschatzmann