pschatzmann/arduino-audio-tools #957: Heavy repo

thewh1teagle, 2023-08-05:
When cloning the repo, it's cloning like 150MB, and it takes a while.
repos should contains only code

pschatzmann, 2023-08-05:
To my knowledge it contains only code!
Any suggestion or pull request to improve is welcome...

LinusHeu, 2023-08-05:
The streams-memory examples have a few audio files turned into .h files. But I really appreciate that they're included because having such example files in this repo makes testing way easier. (Don't have to convert audio samples yourself, and in case something doesn't work you know that it's not a problem with the file.)

pschatzmann, 2023-08-05:
Here is an oberview of the content with the corresponding sizes:

closed 2023-08-05 by pschatzmann

thewh1teagle, 2023-08-05:
Just clone the repo as is. you will see it's weight
```
$ git clone https://github.com/pschatzmann/arduino-audio-tools
$ du -sh arduino-audio-tools/
346M    arduino-audio-tools/
```

It's not just from the files you list, as you can see `./git/objects/` is the most heavy files because it contains files from history which removed but already in the history of git files

pschatzmann, 2023-08-05:
The git directory is in my list. 
So, what would you propose ?

dromer, 2024-07-22:
@pschatzmann use git-lfs for large binaries that are not code.

When you modify these files in git they are in the history three times. Once to add, once to remove and another to add the new version. This happens on every change so the repo will increase a lot in size even though very little changes or code was added.

pschatzmann, 2024-07-22:
I deleted the old unused files which have been deleted a long time ago with [bfg](https://rtyley.github.io/bfg-repo-cleaner/)

This was bringing it back to some reasonable size...

dromer, 2024-07-22:
Yes, bfg is the way to go to clean up history.

Of course the downside is that any downstream clones of the repo will have to re-clone and it doesn't solve the problem long term.
As long as you don't modify these examples and test files too often it should be ok I suppose, but otherwise I really recommend using git lfs for storing such files.

dromer, 2024-07-22:
Biggest files in the current history:

```
02e90f5fcbe1  740KiB examples/examples-audiokit/streams-memory_pcm-mixer-audiokit/guitar.h
0379b4d748bb  746KiB examples/examples-player/player-littlefs-i2s/data/file_example_MP3_700KB.mp3
13a8d777f8fd  977KiB jupyter/guitar-delay.wav
2277da267b84  977KiB jupyter/guitar-fuzz.wav
36d211d7c520  977KiB jupyter/guitar-pitchshift.wav
6924dcf11f74  977KiB jupyter/guitar-tremolo.wav
9879a4867271  977KiB jupyter/guitar-distortion.wav
dfc7eb892acb  977KiB jupyter/guitar-boost.wav
f948aab4474f  977KiB jupyter/guitar.wav
7fc61203bb2e  996KiB jupyter/guitar.raw
7efedd198eae  1.0MiB examples/examples-audiokit/streams-memory_pcm-mixer-audiokit/drums.h
e5669fe16162  1.0MiB examples/examples-audiokit/streams-memory_pcm-mixer-audiokit/drums.h
5e1ed9c85394  1.2MiB examples/examples-stream/streams-memory_mp3-metadata/sample-12s.h
f02a9928909f  1.3MiB docs/resources/small.raw
0363c2a6446d  1.4MiB tests-cmake/codec/aac-faad/audio.h
783023f1aec3  1.6MiB examples/examples-audiokit/streams-audiokit-fft/esp32.svd
d9274744e4a8  2.1MiB jupyter/Jupyter.ipynb
7f43503ef427  2.6MiB docs/resources/audio.mp3
773d000f71d3  2.8MiB examples/examples-desktop/mp3/BabyElephantWalk60_mp3.h
932f5d796c6e  3.0MiB docs/resources/audio.aac
47332e11a1a3  3.8MiB tests-cmake/codec/hls/miniaudio.h
b2667522cac2  3.8MiB cmake-tests/miniaudio/miniaudio.h
facf124cdfb0  3.8MiB cmake-tests/miniaudio/miniaudio.h
dbe8e4d1e064  3.8MiB tests-cmake/miniaudio/miniaudio.h
2fad6838f191  5.9MiB docs/resources/audio-8000.raw
8a79a3b516eb  6.4MiB examples/sandbox/experiment-flash_midi-a2dp/mgm1.h
258cf9fa7d09  7.8MiB examples/examples-stream/streams-memory_raw-i2s/StarWars30.h
05f19f1fc15b  7.8MiB examples/examples-stream/streams-memory_raw-i2s/StarWars30.h
ad7302c32ee2  8.3MiB examples/examples-stream/streams-memory_wav-pwm/knghtsng.h
06b14d1d1b60   18MiB sandbox/streams-memory_aac-a2dp/audio.h
a7d3d3543f55   33MiB docs/resources/audio.raw
```

pschatzmann, 2024-07-22:
Thanks for this: it seems that I still missed to delete some of  them...
Now the repo is down to 99m!