pschatzmann/arduino-audio-tools #967: arduino-audio-tools-> stream with Inmp441 mic is not working after connection

gauravacad, 2023-08-17:
2:01:28.437 -> starting I2S...
12:01:28.437 -> [I] I2SConfig.h : 98 - rx/tx mode: RX_MODE
12:01:28.437 -> [I] I2SConfig.h : 99 - port_no: 0
12:01:28.437 -> [I] I2SConfig.h : 100 - is_master: Master
12:01:28.437 -> [I] I2SConfig.h : 101 - sample rate: 44100
12:01:28.485 -> [I] I2SConfig.h : 102 - bits per sample: 16
12:01:28.485 -> [I] I2SConfig.h : 103 - number of channels: 2
12:01:28.485 -> [I] I2SConfig.h : 104 - i2s_format: I2S_STD_FORMAT
12:01:28.485 -> [I] I2SConfig.h : 106 - auto_clear:0
12:01:28.485 -> [I] I2SConfig.h : 113 - buffer_count:10
12:01:28.485 -> [I] I2SConfig.h : 114 - buffer_size:512
12:01:28.485 -> [I] I2SConfig.h : 118 - pin_bck: 14
12:01:28.485 -> [I] I2SConfig.h : 119 - pin_ws: 15
12:01:28.485 -> [I] I2SConfig.h : 120 - pin_data: 32
12:01:28.485 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.485 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.531 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.531 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.531 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.531 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.531 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.531 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.578 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.578 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.578 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.578 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.578 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.578 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.578 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.624 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.624 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.624 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.624 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.624 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.624 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.624 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.670 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.670 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.670 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.670 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.670 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.670 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.717 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.717 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.717 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.717 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.717 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.717 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.717 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.765 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.765 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.765 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.765 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.765 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.765 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.812 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.812 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.812 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.812 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.812 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.812 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.812 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.859 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.859 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.859 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.859 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.859 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.859 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.859 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.905 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.905 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.905 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.905 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.905 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.905 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.905 -> [I] AudioCopy.h : 135 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
12:01:28.951 -> [I] AudioCopy.h : 135 - StreamCopy::copy 512 -> 512 -> 512 bytes - in 1 hops
12:20:41.944 -> [W] AudioA2DP.h : 307 - ==> state: Disconnected

pschatzmann, 2023-08-17:
I have no clue what you talk about!

closed 2023-08-17 by pschatzmann

pschatzmann, 2023-08-17:
I don't have any magical powers and the only one who can find out what's wrong is you! 
I already sent you the link how to do this!

- Did you test the microphone ?
- Did you test the your bluetooth speaker ?

Some final hints:
- A2DP logging can be activated in the Tools Menu by defining the core debug level.
- for A2DP I suggest that you prefer to use the basic examples. They are much more efficient.
- If A2DP is connecting and closing the connection after processing some data, you are providing invalid audio

pschatzmann, 2023-08-17:
And don't ignore the information provided in the Wiki about ADCs