pschatzmann/arduino-audio-tools #977: Problem with stream-audikit-sd_wav

BRAINDEX-PierreC, 2023-08-26:
Hello, thank you for your work, 
I'm using the stream-audikit-sd_wav exemple with a Lyrat 4.3. Once recording, there is a .wav files of ~5seconds but the signal is only 0, not even noise (top signal in picture) .

The exemple  stream-audikit-sd-audiokit work fine with .raw files (bottom signal in picture) so I don't think there is a material/configuration problem. 

[image]

I try to switch the input line but it doesn't work.

richyxi, 2023-08-27:
Can you try changing this line?
```
auto cfg = in.defaultConfig(RXTX_MODE);
```

i faced a similar issue

BRAINDEX-PierreC, 2023-08-27:
perfect it worked! thank you really much !

musiq021, 2023-08-27:
Surely this is a bug. Why would RXTX be the solution?

pschatzmann, 2023-08-27:
Today I had some time to look into this. If you check the log you can see that the wrong pin was used to read the data from.
I have committed a correction!

closed 2023-08-27 by pschatzmann