pschatzmann/arduino-audio-tools #979: Play wav with LM386

msrba, 2023-08-26:
I would like to use arduino-audio-tools to play a WAV files on an SD Card connected to a Arduino Nano ESP32 over a LM386.

Unfortunately, I'm having trouble getting started.

As far as I understand, I can use AudioPlayer. The source is SDFat, which is clear. The WAVDecoder decoder is also clear. I'm not sure which stream I have to use.

Does anyone have any help for me? Maybe even a small example?

pschatzmann, 2023-08-26:
Please read the documentation first and there are plenty of examples!
[THIS iS NOT A BUG!](https://github.com/pschatzmann/arduino-audiokit#support)

closed 2023-08-26 by pschatzmann

msrba, 2023-08-26:
Two question: 
1. Where i can ask general question
2. I can not find if "Arduino Nano ESP32" is supported and how i can get it work

LinusHeu, 2023-08-26:
1. Discussions Tab
2. ESP32 is supported, and that's the processor that the Arduino Nano ESP32 uses, so I would expect it to just work. You might have to figure out the correct pins. 
The Datasheet says analog input https://www.ti.com/product/de-de/LM386 so I suppose you would use PWMAudioOutput.
Check out the documentation and examples for how to do it.