pschatzmann/arduino-audio-tools #982: The Stream Example issue

gauravacad, 2023-08-29:
Dear Sir 

Sinwave example for stream on A2dp works with 98% memory of ESp32 Wroom Board
INMP441 Digital Microphone with A2dp Says "Guru Meditation Error"

pschatzmann, 2023-08-29:
You were supposed to check the issues and discussions before posting an issue. Use the A2DP project for A2DP related errors: 
- https://github.com/pschatzmann/ESP32-A2DP/issues/120
Your Guru Mediation Error is missing the necessary detail, but I assume its from the watchdog ?
- https://github.com/pschatzmann/ESP32-A2DP/issues/463

closed 2023-08-29 by pschatzmann

gauravacad, 2023-09-06:
Dear Sir after Right Partition , the code is consuming 67%
But I have correct Sinewave sound from steam generator code.
That means I have correct setup. 
Now using A2DP I2S stream code, I am getting left channel only with high noise in right ear phone. 
please correct me where I am wrong now. 
I check no latency while playing in left, though it also have pause and play but was atleast working, Right channel has huge noise only.
Can You please tell me something , Do I need to change the Philliphs format, 44100 sample rate, or buffer len etc.
Please give me a guide this time.
thank you

pschatzmann, 2023-09-06:
This has already been answered before and is properly documented for the INMP441 in the Wiki

gauravacad, 2023-09-06:
I read : https://github.com/pschatzmann/arduino-audio-tools/wiki/It's-not-working
INMP441 is working when it is near to sound source if we take it away say 10 cms , the sinewave became distorted.

Second in this function statement "ConverterFillLeftAndRight<int16_t> filler(LeftIsEmpty);" 
In the stream-i2s-a2dp.ino
what should I change to make it 44100 sampling rate, stereo sound both left and right channel have individual Microphone, 
and width is 32. 
I tried changing in Audioconfig.h file and audiotools in src. But couldnot find it easy to make it work.

pschatzmann, 2023-09-06:
I am afraid that your question does not make any sense to me: What do you think ConverterFillLeftAndRight is doing ?
Did you make any tests to visualize the result ?

As an alternative you could use 32 bits, but then you will need to convert to 16. This is properly documented in the Wiki!