pschatzmann/arduino-audio-tools #986: Multiple definition of `_ZN11audio_tools4selfE'

Arubinu, 2023-09-01:
Hello, my goal is to read the contents of a MicroSD with your library, using an ESP8266 (in the configuration: ESP12E).

When compiling, everything goes well except at the very end, as you can see.

I don't know enough about it to find the source of the problem. Anyone got a solution?

```
 *  Exécution de la tâche : C:\Users\pyrof\.platformio\penv\Scripts\platformio.exe run --environment ESP8266 

Processing ESP8266 (platform: espressif8266; board: esp12e; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 (4.2.1) > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 @ 3.30102.0 (3.1.2)       
 - tool-esptool @ 1.413.0 (4.13)
 - tool-esptoolpy @ 1.30000.201119 (3.0.0)
 - toolchain-xtensa @ 2.100300.220621 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- FastLED @ 3.6.0
|-- ESP Async WebServer @ 1.2.3+sha.f71e3d4
|-- audio-tools @ 0.9.6+sha.ae5be6e        
|-- libhelix @ 0.1.0+sha.e937240
|-- SD @ 2.0.0
|-- SPI @ 1.0
|-- ESP8266WiFi @ 1.0
|-- ESPAsyncTCP @ 1.2.2
Building in release mode
Linking .pio\build\ESP8266\firmware.elf
c:/users/pyrof/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\ESP8266\src\main.cpp.o:(.bss._ZN11audio_tools4selfE+0x0): multiple definition of `_ZN11audio_tools4selfE'; .pio\build\ESP8266\src\audio.cpp.o:(.bss._ZN11audio_tools4selfE+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\ESP8266\firmware.elf] Error 1
=================================================================================================== [FAILED] Took 10.38 seconds ===================================================================================================

Environment    Status    Duration
-------------  --------  ------------
ESP8266        FAILED    00:00:10.375
============================================================================================== 1 failed, 0 succeeded in 00:00:10.375 ============================================================================================== 

 *  Arrêt du processus de terminal "C:\Users\pyrof\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'ESP8266'". Code de sortie : 1. 
```

pschatzmann, 2023-09-01:
I committed a correction. Could you please help me to test it.
Multiple compilation units have not been tested on the ESP32 only...

Arubinu, 2023-09-01:
I can test of course.

I tried to compile for ESP32 before this issue, and there were no problems.

On the other hand, do you know the syntax to specify the hash of a commit on platformio?

pschatzmann, 2023-09-01:
No, this is my correction: https://github.com/pschatzmann/arduino-audio-tools/commit/3fc0928323b0367ba41a18fcf715cb667909024a

Arubinu, 2023-09-01:
So for https://github.com/pschatzmann/arduino-audio-tools/commit/3fc0928323b0367ba41a18fcf715cb667909024a, the compilation goes smoothly for the ESP8266 (ESP12S).

On the other hand, it is impossible to start the compilation for the ESP32, PlatformIO gives me the following error: Cannot read properties of undefined (reading 'id')

Arubinu, 2023-09-01:
Small note for the ESP8266 (since it's the one I have plugged in at the moment), I don't know if it's due to my code which also uses a web server, but when I try to read a file , I have a crash followed by a reboot.

```
[I] AudioEncoded.h : 362 - virtual void audio_tools::EncodedAudioOutput::end()
[I] MetaDataID3.h : 575 - virtual void audio_tools::MetaDataID3::end()
[I] AudioPlayer.h : 211 - reset codec

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Unhandled C++ exception: OOM

>>>stack>>>

ctx: sys
sp: 3fffece0 end: 3fffffb0 offset: 0010
3fffecf0:  3fffed68 3fffa274 00002800 40220e08  
3fffed00:  3fffed68 3fffa9c4 3fff7f8c 40209628  
3fffed10:  ffffffff ffffffff 3fff50b8 00000002  
3fffed20:  3fffed68 3fffa274 3fff50b8 40209e52  
3fffed30:  3fff9d78 00000010 3fff4cd0 4020f40a  
3fffed40:  3fff9d78 3fffa274 3fffa274 40210d97  
3fffed50:  00000000 3fffa274 00000001 3fffa77c  
3fffed60:  0010001f 80000000 2e383030 0033706d  
3fffed70:  07000000 3fffa274 3fff971c 40215148  
3fffed80:  00000000 000d000f 3fff9d64 402145c0  
3fffed90:  3fffa274 3fffa5ec 3fffa274 40213415  
3fffeda0:  00000000 00000001 3fffa2a8 40220262  
3fffedb0:  3ffe9d4e 3fffa5ec 00000000 00000030  
3fffedc0:  3fffa5ec 3fff61dc 3fffa054 00000002  
3fffedd0:  00000002 3fffa5ec 3fffa274 40213542  
3fffede0:  3fff61bc 3fffa054 3fff61b8 00000000  
3fffedf0:  00000001 00000020 3fff9e94 40100bf0  
3fffee00:  3fffa2a8 00000000 9645a1ca 0039ea1d  
3fffee10:  00007fff 3fffa274 3fffa18c 3fffa3bc  
3fffee20:  3fffee70 3fffa18c 3fffa054 40224690  
3fffee30:  000001ec 3fffa402 3fffa18c 3fffa274  
3fffee40:  00000000 00000566 3ffe8b94 401008a6  
3fffee50:  3fff0000 3fff5960 3fffa054 00000000  
3fffee60:  3fffa3bc 3fffa054 3fffa18c 402246f0  
3fffee70:  3fff9eb8 3fff9eac 00000000 3fff5f18  
3fffee80:  3fff6380 3fff5f18 3fffa3bc 3fff61b8  
3fffee90:  3fffa054 3fff61bc 3fff61bd 4025a719  
3fffeea0:  ffffffff 3fffc6fc 00000000 3fffdab0  
3fffeeb0:  00000018 00000000 3ffef470 00000030  
3fffeec0:  3fff0000 3fffa022 3fffa01a 3fff5f18  
3fffeed0:  3fff6380 3fffa3da 3fffa3bc 4025f67d  
3fffeee0:  00000014 3fff5f18 00000000 4025736d  
3fffeef0:  00000000 00000539 3ffe8b94 401008a6  
3fffef00:  4027bc68 3fff9e94 3fff78ac 3fff9c34  
3fffef10:  3fff5f18 00000008 3fffa3bc 40256a35  
3fffef20:  3fffdc80 3fff78ac 3fff9e94 40256848  
3fffef30:  4027ee19 3fff78ac 3fff9e94 4027ee2b  
3fffef40:  3fffa3cc 3fffa3bc 00000000 3fffdcb0  
3fffef50:  3fff4608 00000000 3fff9e94 40280107  
3fffef60:  40000f49 3fffdab0 ffffff01 40000f49  
3fffef70:  40000e19 00098289 bff00000 0000bfff  
3fffef80:  00000005 aa55aa55 000000f7 40105629
3fffef90:  4010562f bff00000 0000bfff ee4870e4  
3fffefa0:  4010000d bff00000 00098289 401000ab
3fffefb0:  402707c8 3fffef4c 40270779 3fffff38  
3fffefc0:  3fffffd0 00000000 00000000 feefeffe  
3fffefd0:  feefeffe feefeffe feefeffe feefeffe
3fffefe0:  feefeffe feefeffe feefeffe feefeffe  
3fffeff0:  feefeffe feefeffe feefeffe feefeffe
3ffff000:  feefeffe feefeffe feefeffe feefeffe  
3ffff010:  feefeffe feefeffe feefeffe feefeffe
3ffff020:  feefeffe feefeffe feefeffe feefeffe  
3ffff030:  feefeffe feefeffe feefeffe feefeffe
3ffff040:  feefeffe feefeffe feefeffe feefeffe  
[... 231 lines cut ...]
3ffffec0:  40221ae1 00000030 00000010 ffffffff  
3ffffed0:  40221adb 4021f9dc feefeffe feefeffe
3ffffee0:  00000000 00000000 00000000 fffffffe  
3ffffef0:  40223274 3fffc6fc 00000000 3fff5ae4
3fffff00:  40105c81 003ce350 3fff5b10 00000030  
3fffff10:  40105d2e 3fff45c8 003ce350 00000000
3fffff20:  40105f05 003cef96 3fff5ae8 00000000  
3fffff30:  3fff45f0 3fff5ae8 3ffe8b90 3fff5ae8
3fffff40:  0000320b 0000000a 40221888 3fffefb0
3fffff50:  3fff5ae8 0000320b 0000000a 402219e1  
3fffff60:  0000320b 0000000a 0000320b 40221a57
3fffff70:  00000000 0000320b 916872b0 0039dd60
3fffff80:  00000000 3fff5960 3fff5a70 3fff5b10  
3fffff90:  3fffdad0 0000320b 0000000a 40222040
3fffffa0:  3fffdad0 00000000 3fff5ae4 40211168  
<<<stack<<<

last failed alloc call: 40209628(10240)

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

last failed alloc caller: 0x40209628

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)
```

pschatzmann, 2023-09-01:
You need to decide the exception.

pschatzmann, 2023-09-01:
I never tested this scenario on an ESP8266. Maybe you just run out of memory

Arubinu, 2023-09-01:
Okay, thanks for this information.
For the ESP32, I just tried and it actually compiles without problems (the previous problem with PlatformIO was resolved after restarting VSCode).

closed 2023-09-03 by pschatzmann