pschatzmann/arduino-audio-tools #993: Problem using Analog Pin GPIO36 on ESP32 when using i2s

jimruxton, 2023-09-08:
I'm using i2s in audiotools on an esp32 mini module and have no trouble reading analog on GPIO34 however when trying to read analog on GPIO36 when using i2s in audiotools I'm getting incorrect values. I either get 0 or 4095. Wondering how audiotools is manipulating this pin and how I can stop it from interfering with that pin.
Thanks

pschatzmann, 2023-09-08:
I am not doing anything with GPIO36. Can you share how you set this up?

jimruxton, 2023-09-08:
My code is below . If I remove all the code except reading the analog pin I can read GPIO36 or keeping the audiotools code in and using GPIO34 everything works. The version of audio tools I'm using is version=0.9.3 . This is the board I'm using [https://www.amazon.ca/gp/product/B08TBQRMY4/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1](url)
`/**
 * @file example-serial-receive.ino
 * @author Phil Schatzmann
 * @brief Receiving audio via IP and writing to I2S
 * @version 0.1
 * @date 2022-03-09
 * 
 * @copyright Copyright (c) 2022
 */

#include "AudioTools.h"
#include <WiFi.h>
#include "AudioLibs/Communication.h"
#include <driver/adc.h>
  const int potPin = 36; // changed to 36 from 34

// variable for storing the potentiometer value
int potValue = 0;

float volumeSet=0.1; 

uint16_t sample_rate = 16000; 
uint16_t port = 8000;
// Set your Static IP address
IPAddress local_IP(192, 168, 1, 184);
// Set your Gateway IP address
IPAddress gateway(192, 168, 1, 1);

IPAddress subnet(255, 255, 255, 0);
IPAddress primaryDNS(8, 8, 8, 8);   //optional
IPAddress secondaryDNS(8, 8, 4, 4); //optional
uint8_t channels = 1;  // The stream will have 2 channels
WiFiServer server(port);
WiFiClient client; 
UDPStream stream;
I2SStream out;
VolumeStream volume(out);
 
StreamCopy copier(volume, stream); 
// const char *ssid = "ARRIS-C025";
// const char *password = "*****";

const char *ssid = "NETGEAR41";
const char *password = "****";

void connectWifi(){
  // connect to WIFI
  WiFi.begin(ssid, password);
//  if (!WiFi.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS)) {
//    Serial.println("STA Failed to configure");
//  }
  while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
  }
  Serial.println();
  Serial.println(WiFi. localIP());

  // Performance Hack              
  esp_wifi_set_ps(WIFI_PS_NONE);
} 

void setup() {
  Serial.begin(115200);
//  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  connectWifi(); 

  // start server
  stream.begin(8000);

  // start I2S
  Serial.println("starting I2S...");
  auto config = out.defaultConfig(TX_MODE);
  config.sample_rate = sample_rate; 
  config.channels = channels;
  config.bits_per_sample = 16;
  config.buffer_size = 512;
  config.buffer_count = 6;
  out.begin(config);

  // set initial volume
  volume.begin(config); // we need to provide the bits_per_sample and channels
  volume.setVolume(volumeSet);
  
  Serial.println("started...");

 
}

void loop() { 

 

  potValue = analogRead(potPin);
  volumeSet=potValue/4095.0;

    volume.setVolume(volumeSet); 
  Serial.println(volumeSet); 

     copier.copy(); 

 // }
}`

pschatzmann, 2023-09-08:
This has nothing to do with my library, so I am closing this.
I think you just forgot to define the pinmode...

closed 2023-09-08 by pschatzmann