pschatzmann/arduino-audio-tools #1002: How to Read Mp3 file form SPIFFS and play on A2DP

Talhavaival1, 2023-09-24:
### Problem Description

I have sucessfuly connected to my bluetooth speaker and tested its output using the generator a2dp sketch , how do I output an Audio file from spiffs to speaker.

### Device Description

ESP32cam

### Sketch

```cpp
no code
```

### What is your development environment

arduino

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

Talhavaival1, 2023-09-24:
well update , It seems I was download the wrong version , as I didnt get the release file version , but now even the sine test sketch is not working and I cant connect to my bluetooth speaker anymore.

pschatzmann, 2023-09-24:
Check the issues and discussions of A2DP Project to resolve reconnect issues.

There are multiple examples that demonstrate how to play files to A2DP. 
I suggest to have a look at the player documentation...

closed 2023-09-24 by pschatzmann

Talhavaival1, 2023-09-24:
But none of the example show how to play audio files from SPIFFS

pschatzmann, 2023-09-24:
But in the end this does not matter: the mechanism is the same independently on what file system you use

Talhavaival1, 2023-09-25:
Can you mention the example where it is done , Also I have made a sketch where I am reading mp3 from Spiffs but it says , no data to read to read Audioplayer 185 .

pschatzmann, 2023-09-25:
> Can you mention the example where it is done , Also I have made a sketch where I am reading mp3 from Spiffs but it says , no data to read to read Audioplayer 185 .

In the audio player examples. You can also find a link in the audioplayer wiki

pschatzmann, 2023-09-25:
Your sketch will not work! https://github.com/pschatzmann/ESP32-A2DP/wiki/WIFI-and-A2DP-Coexistence