pschatzmann/arduino-audio-tools #1015: Example basic-a2dp-spdif crashes at start

RetroBobba, 2023-10-08:
### Problem Description

Hello,
I've try to get the basic-a2dp-spdif running. But got no luck.

At first, the examples "streams-generator-spdif" works and the "bt_music_receiver_simple" from your ESP32-A2DP lib is working too

Partition sheme: huge app
I've tried different boards (both ESP WROOM 32). Arduino IDE 1.8.12 and 2.2.1. 
Version ESP32-A2DP 1.7.3 and 1.7.4

Depending on combination of libs/IDE I got different errors messages. Here maybe the most helping one:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d362b  PS      : 0x00060f30  A0      : 0x800d40ba  A1      : 0x3ffcba40  
A2      : 0x3ffc3618  A3      : 0x3f400efc  A4      : 0x3ffc36a8  A5      : 0x00000003  
A6      : 0x00000001  A7      : 0x0000e100  A8      : 0x018cba80  A9      : 0x00000000  
A10     : 0x3ffc3618  A11     : 0xffffffff  A12     : 0x00000000  A13     : 0x3ffc4338  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400d3530  LEND    : 0x400d3537  LCOUNT  : 0x00000000  

Backtrace: 0x400d3628:0x3ffcba40 0x400d40b7:0x3ffcbb00 0x400d637e:0x3ffcbb60

### Device Description

ESP WROOM 32

### Sketch

```cpp
basic-a2dp-spdif
```

### What is your development environment

Arduino IDE 1.8.12 and 2.2.1. 

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2023-10-08:
I suggest to decode the stacktrace to find out where the error is coming from.

pschatzmann, 2023-10-08:
I have committed the correction to the sketch

closed 2023-10-08 by pschatzmann

RetroBobba, 2023-10-09:
Thanks, your correction does help a lot. And it works but not stable.

It seemes so, that errors appear randomly. Mostly start up and pairing works fine.
When SPDIF output works, I will get every 1 - 5 Minuts dropouts. At first a was working with a old SPDIF Output Bracket ans it seems so, that was designed for 5 V Signals. That I've made my own voltage devider for working with 3.3 V. But dropouts are still there.
And sometimes, the bluetooth connection got lost. Sometimes it is possible to establish the connection again and sometimes I have du restart the ESP32.

I've tried to decode a stracktrace where the ESP crashed while establishing the bluetooth connection.
Not sure I've done the stacktrace right.

Put stacktrace and basic-a2dp-spdif.ino.elf file to maximeborges.github.io/esp-stacktrace-decoder:

Stacktrace:

```
assert failed: host_recv_pkt_cb hci_hal_h4.c:548 (0)

Backtrace: 0x400838b9:0x3ffe7f10 0x40094751:0x3ffe7f30 0x40099f79:0x3ffe7f50 0x401153fe:0x3ffe8080 0x40170672:0x3ffe80a0 0x4015f6ee:0x3ffe80c0 0x4001791d:0x3ffe80e0 0x4016005d:0x3ffe8100 0x4016013a:0x3ffe8140 0x400188f5:0x3ffe8170 0x4015f97d:0x3ffe81b0 0x401634ad:0x3ffe81d0 0x400869f4:0x3ffe81f0 0x40019d11:0x3ffe8220 0x40055b4d:0x3ffe8240 0x4015df3b:0x3ffe8260 0x4015e5a5:0x3ffe8280

ELF file SHA256: c22db07c70764a7b
```

Output:

```
    0x400838b9: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
    0x40094751: esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
    0x40099f79: __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
    0x401153fe: host_recv_pkt_cb at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/hci/hci_hal_h4.c:548
```

pschatzmann, 2023-10-09:
I am afraid that this error hasn't anything to do with my code. 

I suggest that you try another ESP32 to exclude that there is something wrong with your hardware and different ESP32 core releases e.g  2.0.14 or 1.0.6