pschatzmann/arduino-audio-tools #1021: Volume management with my RTSP stream

Bartiff, 2023-10-16:
### Problem Description

Hi !

I managed to create an RTSP stream with an INMP441 using issue #756. But I can't increase the gain/volume.
Below is my code but VolumeStream doesn't seem to have any effect on volume.

Thanks for your help

### Device Description

ESP32 Wroom

### Sketch

```cpp
#include "AudioTools.h"
#include "AudioLibs/RTSP.h"
#include "AudioStreamer.h"
#include "RTSPServer.h"

#include "secret.h"

int port = 554;

I2SStream i2sStream; // Audio source

AudioInfo from(8000, 1, 32);
AudioInfo to(8000,1,16);
FormatConverterStream converter(i2sStream);
StreamCopy copier(i2sStream, converter);

VolumeStream volume(converter);

RTSPSourceFromAudioStream source(converter); // IAudioSource for RTSP
AudioStreamer streamer = AudioStreamer(&source); // Stream audio via RTSP
RTSPServer rtsp = RTSPServer(&streamer, port);

const char* wifi = WIFI_SSID;
const char* password = WIFI_PASS;

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);
  // Define Converter
  converter.begin(from, to);

  auto cfg = i2sStream.defaultConfig(RX_MODE);

  cfg.i2s_format = I2S_STD_FORMAT;
  cfg.copyFrom(from);
  cfg.pin_ws = 2;
  cfg.pin_bck = 14;
  cfg.pin_data = 15;
  cfg.port_no = 1;
  i2sStream.begin(cfg);

  volume.begin(cfg);
  volume.setVolume(2.0);

  // Start Wifi
  rtsp.begin(wifi, password);
}

void loop() { delay(1000); }
```

### What is your development environment

PlatformIO

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2023-10-16:
Of cause this is not working: : volumeStream is not used anywhere in your chain!

closed 2023-10-16 by pschatzmann

Bartiff, 2023-10-16:
Thank you for your speed but unfortunately I don't see how to integrate it into the chain... :/

pschatzmann, 2023-10-16:
I think something like this should work: i2sStream - converter - volume - source

I2SStream i2sStream; 
FormatConverterStream converter(i2sStream);
VolumeStream volume(converter);
RTSPSourceFromAudioStream source(volume); // IAudioSource for RTSP
AudioStreamer streamer = AudioStreamer(&source); // Stream audio via RTSP
RTSPServer rtsp = RTSPServer(&streamer, port);

Bartiff, 2023-10-16:
I tried that but my RTSP stream no longer outputs sound:

```
I2SStream i2sStream; // Audio source

AudioInfo from(8000, 1, 32);
AudioInfo to(8000, 1, 16);
FormatConverterStream converter(i2sStream);
StreamCopy copier(i2sStream, converter);

VolumeStream volume(converter);

RTSPSourceFromAudioStream source(volume); // IAudioSource for RTSP
AudioStreamer streamer = AudioStreamer(&source); // Stream audio via RTSP

RTSPServer rtsp = RTSPServer(&streamer, port);
```

pschatzmann, 2023-10-16:
Of cause not, you messed up with the begin !

Bartiff, 2023-10-16:
I really turned the thing around and I don't understand why it doesn't work...

```
#include "AudioTools.h"
#include "AudioLibs/RTSP.h"
#include "AudioStreamer.h"
#include "RTSPServer.h"

#include "secret.h"

int port = 554;

I2SStream i2sStream; // Audio source

AudioInfo from(8000, 1, 32);
AudioInfo to(8000, 1, 16);
FormatConverterStream converter(i2sStream);

VolumeStream volume(converter);

RTSPSourceFromAudioStream source(volume); // IAudioSource for RTSP
AudioStreamer streamer = AudioStreamer(&source); // Stream audio via RTSP
RTSPServer rtsp = RTSPServer(&streamer, port);

const char* wifi = WIFI_SSID;
const char* password = WIFI_PASS;

void setup() {
  Serial.begin(9600);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);
  
  // Define Converter
  converter.begin(from, to);

  auto cfg = i2sStream.defaultConfig(RX_MODE);
  cfg.i2s_format = I2S_STD_FORMAT;
  cfg.copyFrom(from);
  cfg.pin_ws = 2;
  cfg.pin_bck = 14;
  cfg.pin_data = 15;
  cfg.port_no = 1;

  i2sStream.begin(cfg);

  // Define Volume
  volume.begin(cfg);
  volume.setVolume(2.0);

  // Start Wifi
  rtsp.begin(wifi, password);
}

void loop() { delay(1000); }
```

My logs :

> connect to rtsp://192.168.101.142:554
> [ 15673][E][RTSPFormat.h:152] format(): unsupported audio type
> [ 15673][E][RTSPFormat.h:136] payloadFormat(): unsupported audio type
> [I] RTSP.h : 129 - virtual void audio_tools::RTSPSourceFromAudioStream::start()
> [I] VolumeStream.h : 187 - setVolume: 1.000000
> [I] RTSP.h : 138 - virtual void audio_tools::RTSPSourceFromAudioStream::stop()
> [ 31570][E][RTSPFormat.h:152] format(): unsupported audio type
> [ 31570][E][RTSPFormat.h:136] payloadFormat(): unsupported audio type
> [I] RTSP.h : 129 - virtual void audio_tools::RTSPSourceFromAudioStream::start()
> [I] VolumeStream.h : 187 - setVolume: 1.000000
> [I] RTSP.h : 138 - virtual void audio_tools::RTSPSourceFromAudioStream::stop()

pschatzmann, 2023-10-16:
You need to provide the correct AudioInfo. If you have the volume after the conversion you need to tell it it's 16 bits and not 32!

Bartiff, 2023-10-17:
Ok but that doesn't seem to be enough. I'm having trouble understanding what could be happening. Could it be because of my conversion?
Otherwise, adjusting the volume before conversion is a mistake in your opinion?

pschatzmann, 2023-10-17:
That should work as well: It's just a little bit less efficient because the volume needs to calculate on 32 values instead of 16 bits. The critical thing is really that the VolumeStream needs to know the correct format!