pschatzmann/arduino-audio-tools #1025: VBAN throttle bug

vorosj, 2023-10-21:
### Problem Description

Using a Python program I tested the VBAN UDP stream of the streams-generator-vban example.
I measured the bitrate in Python, and found, that the example program's bitrate varies way too much.

Your intention probaly was, that the throttle setting acts as fine tuning in the 0.1% range. But it's brutal, throttle +-1 resulted in a few percents. The random deviation is big too, prevents any slow adaptive sync to keep up.

So I think it's a bug in the throttle control. 

### Device Description

ESP 32S, PlatformIO

### Sketch

```cpp
The unchanged streams-generator-vban.ino produces the speed issue.
```

### Other Steps to Reproduce

Measure the sent bitrate with any method, I did it using a Python test program. I also tested the possible UDP errors, there was no missung or errenous UDP packet durint my tests.

### What is your development environment

PlatformIO

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2023-10-21:
I suggest to change the implementation to use microseconds instead of milliseconds.
That should be quite easy to do

vorosj, 2023-10-21:
Not so easy... :-)

I printed out the values in the following function:

> void delaySamples(size_t samples) {
    int durationMsEff = millis() - start_time;
    int durationToBe = (samples * 1000) / info.sample_rate;
    int waitMs = durationToBe - durationMsEff + info.correction_ms;
    LOGI("wait: %d", waitMs);
    if (waitMs > 0) {
      //delayMicroseconds(5*waitMs);
      Serial.print(millis());
      Serial.print("  ");
      Serial.print(start_time);
      Serial.print("  ");and - 
      Serial.println(durationMsEff);
    }
  }

and - to my surprise - non of the variables are correct. 
"start_time" is increasing with millis(), so int durationMsEff==0
samples == 512 (not increasing)
This way the correction fails, and waitMs depends on only the throttle, and slows down the packet rate brutally.  

Hope I can fix it in some minutes...

vorosj, 2023-10-21:
I wrote a quick&dirty fix, now the timing is perfect, and working without glitches with either the Python program and the VB-Audio VBANReceptor.

The fix:
1. modify startDelay() to record the program start time once and forever,  or call itt only once and not in cycle in the write() function
2. make a cumulative variable to sum the samples in delaySamples(size_t samples), and use it:  int64_t durationToBe = (sum_samples * 1000) / info.sample_rate;
3. use int64_t (or a less dirty way) to prevent overflow of the msec and sum_samples variables

This way the code keeps the samplerate, and the "throttle" is not speed control, but more like controls how fast it acts. In order to make it a fine speed control, it needs a mod to insert the throttle value in a cumulative way into durationToBe. 

I didn't wrote a polished code, as you do it way better in your style.

pschatzmann, 2023-10-21:
https://github.com/pschatzmann/arduino-audio-tools/commit/8089a7bb2c2d9d0b1861cb215b2c5f42ca75333a

closed 2023-10-21 by pschatzmann