pschatzmann/arduino-audio-tools #1028: Analog Input to SD Card Recording Skipping

cjoweb12, 2023-10-24:
### Problem Description

I've been trying for about a week now to get a consistent recording on an SD card in the form of a WAV file.

I am able to record no problem, and there is very little noise in the recording on the SD card. However, sometimes the recording will have a skip in it. I can't figure out why this would happen randomly. For example, sometimes the file written for a ten second recording at 44.1kHz, 2 channel, 16 bits is 1,774KB, and other times it is 1,600KB and the recording has skips in it. The recording will vary anywhere from 9 seconds to 10 seconds long, but should consistently be 10 seconds.

I thought it could be the buffer size being too low, but changing the value does not make much of a difference. It is definitely worse at smaller buffer sizes, but not better at larger sizes. 

### Device Description

ESP32 WROOM, MAX9814 with electret microphone, SD card adapter

### Sketch

```cpp
#include <SD.h>
#include <SPI.h>

#include "AudioTools.h"
#include "AudioCodecs/CodecWAV.h"

#define USE_AUDIO_LOGGING false

const char* file_name = "/rec.wav";
AudioInfo info(44100, 2, 16);
AnalogAudioStream in;
File file;  // final output stream
EncodedAudioStream out(&file, new WAVEncoder());
StreamCopy copier(out, in);  // copies data
uint64_t timeout;

void setup() {
  Serial.begin(115200);
//  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // setup input
  auto cfg = in.defaultConfig(RX_MODE);
  cfg.copyFrom(info);
  in.begin(cfg);
  
  // Open SD drive
  if (!SD.begin(5)) {
    Serial.println("Initialization failed!");
    stop();
  }
  
  // Cleanup if necessary
  if (SD.exists(file_name))
    SD.remove(file_name);

  // open file
  file = SD.open(file_name, FILE_WRITE);
  if (!file){
    Serial.println("file failed!");
    stop();
  }

  Serial.println(copier.bufferSize());

  // setup output
  out.begin(info);
  

  // set timeout - recoring for 5 seconds
  timeout = millis() + 20000;
}

void loop() {
  if (millis() < timeout) {
    // record to wav file
    copier.copy();
  } else {
    // close file when done
    if (file) {
      file.flush();
      Serial.print("File has ");
      Serial.print(file.size());
      Serial.println(" bytes");
      file.close();
    }
  }
}
```

### Other Steps to Reproduce

If I record as RAW, I don’t experience the skipping but the audio is quite noisy. WAV gets rid of the noise, but has skipping. 

### What is your development environment

Arduino IDE, using latest versions of all libraries.

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2023-10-24:
My gut feeling is that you need to look into the write speed of your sd card. 
Just try to decrease the sampling rate or set the channels to 1 (which would half the data thruput)

cjoweb12, 2023-10-24:
Thanks for your reply! I have tried increasing the SPI speed for the SD card, and changing the sample rate and channels. It doesn't appear to make much of a difference, but I will try to test more formally.

Would it make sense that writing the RAW file doesn't have this issue, but the WAV does?

pschatzmann, 2023-10-24:
Not really, WAV is just writing a header in the beginning and then it does the same thing like raw...

cjoweb12, 2023-10-24:
Interesting! Then the noise difference is very odd as well... I get a ton of noise when recording RAW but essentially none when recording WAV.

pschatzmann, 2023-10-24:
That does not make any sense...

cjoweb12, 2023-10-24:
Sounds like it makes no sense. So calling file.flush() after every copy would make sure no data is lost?

cjoweb12, 2023-10-25:
Update: If I put a short delay before each recording, it seems to solve the problem most of the time. 

I also noticed that it's usually a problem on the FIRST recording after resetting the ESP32. Because of how my code was setup, I was resetting the ESP32 to make a new recording. That meant it was always the "first" recording.

When I changed the code to add the delay and to make many recordings without resetting the ESP32, 99% of them were perfect without skips...

Kundan-Saha, 2023-11-26:
I am able to record audio sound, so can you mention? What are the pinouts for the recording audio sound using MAXX4466/MAXX9614 microphone , SD card and ESP 32 WROOM?

closed 2023-12-11 by pschatzmann