pschatzmann/arduino-audio-tools #1034: few buggs!

josef2600, 2023-11-02:
### Problem Description

In file:
`arduino-audio-tools-main\examples\examples-communication\vban\streams-audiokit-vban`
you forgot the file extension. It should be` streams-audiokit-vban.ino`
in files:
```
arduino-audio-tools-main\examples\examples-communication\rtsp\communication-codec-rtsp
arduino-audio-tools-main\examples\examples-communication\ip\communication-ip-receive
arduino-audio-tools-main\examples\examples-communication\ip\communication-ip-receive_measure
```
you have used:
```
  if (XXX)
  {
    copier.copy();
  }
```
This causes crash if `if == 0` then memory goes up after few seconds it crashes. You can’t use “if”. The copier.copy(); has to always be running. I assume you are creating tasks that are not cleaning the memory and you have memory leakage!

### Device Description

all boards.

### Sketch

```cpp
arduino-audio-tools-main\examples\examples-communication\vban\streams-audiokit-vban
arduino-audio-tools-main\examples\examples-communication\rtsp\communication-codec-rtsp
arduino-audio-tools-main\examples\examples-communication\ip\communication-ip-receive
arduino-audio-tools-main\examples\examples-communication\ip\communication-ip-receive_measure
```

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

josef2600, 2023-11-02:
and base on few tests i have doe, I2S buffer length at most should be 1024, otherwise it will disturb the audio or wont connect. too large buffers can cause that too.

josef2600, 2023-11-02:
i don't know why, but RTSP samples are not compiling. i have installed all libs including "AudioLibs/AudioClientRTSP.h"
`arduino-audio-tools-main\examples\examples-communication\rtsp\communication-rtsp-audiokit`

```
BasicUsageEnvironment.hh:1:1: error: expected unqualified-id before '.' token
 ../BasicUsageEnvironment/include/BasicUsageEnvironment.hh
```

```
src/AudioLibs/AudioClientRTSP.h:139:5: error: 'audiotools_rtsp' does not name a type; did you mean 'audio_tools'?
     audiotools_rtsp::OurRTSPClient *client() {
```
```
arduino-audio-tools-main\src/AudioLibs/AudioClientRTSP.h:144:5: error: 'audiotools_rtsp' does not name a type; did you mean 'audio_tools'?
     audiotools_rtsp::OurRTSPClient* rtsp_client;
```
```
arduino-audio-tools-main\src/AudioLibs/AudioClientRTSP.h:145:5: error: 'UsageEnvironment' does not name a type
     UsageEnvironment* env=nullptr;
```
```
arduino-audio-tools-main\src/AudioLibs/AudioClientRTSP.h:146:5: error: 'TaskScheduler' does not name a type; did you mean 'vTaskEndScheduler'?
     TaskScheduler* scheduler=nullptr;
```

```
     vTaskEndScheduler
\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/AudioClientRTSP.h: In member function 'void audio_tools::AudioClientRTSP::setBufferSize(int)':
\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/AudioClientRTSP.h:78:7: error: 'audiotools_rtsp' has not been declared
       audiotools_rtsp::rtspSinkReceiveBufferSize = size;
```

and in `arduino-audio-tools-main\examples\examples-communication\rtsp\communication-rtsp-i2s` example:

```
communication-rtsp-i2s.ino:15:
C:\Users\Josef\Documents\Arduino\libraries\arduino-live555-master\src/RTSPSimpleClient.hh:1:1: error: expected unqualified-id before '.' token
 ../arduino/RTSPSimpleClient.hh
 ^

```
```
mmunication-rtsp-i2s:18:29: error: 'i2s' was not declared in this scope
 EncodedAudioStream out_mp3(&i2s, new MP3DecoderHelix());  // Decoding stream
```
```
mmunication-rtsp-i2s:19:1: error: 'RTSPSimpleClient' does not name a type; did you mean 'ESPLwIPClient'?
 RTSPSimpleClient rtsp;
```
```
communication-rtsp-i2s\communication-rtsp-i2s.ino: In function 'void setup()':
communication-rtsp-i2s:29:18: error: 'i2s' was not declared in this scope
   auto cfg_i2s = i2s.defaultConfig(TX_MODE);
              ^~~
communication-rtsp-i2s:37:14: error: 'rtsp' was not declared in this scope
   auto cfg = rtsp.defaultConfig();

note: suggested alternative: 'stop'
   auto cfg = rtsp.defaultConfig();
              ^~~~
              stop

```
```

communication-rtsp-i2s.ino: In function 'void loop()':
communication-rtsp-i2s:50:3: error: 'rtsp' was not declared in this scope
   rtsp.singleStep();
```

and many more.

pschatzmann, 2023-11-03:
I could reproduce your iban solution and this has been resolved.

For the other topics I am missing the  information how to reproduce your issues. 
I am also asking you, not to mix different issues: I will not be able to track this properly. 

Your assumption is also wrong: I am just using plain vanilla Arduino functionality that should work on any other processor as well, so you might bark up the wrong tree!

closed 2023-11-03 by pschatzmann

pschatzmann, 2023-11-03:
ps. The namespace audiotools_rtsp is part of the AudioTools project, so if you get compile errors you are not using the latest release

josef2600, 2023-11-03:
shit! 409 files!!! please don't tell me you only simulate codes and not compiling!
in file `C:\Users\Josef\Documents\Arduino\libraries\arduino-live555-master\src\BasicUsageEnvironment.hh` you have:
`../BasicUsageEnvironment/include/BasicUsageEnvironment.hh`
this is wrong. it should be:
`#include "../BasicUsageEnvironment/include/BasicUsageEnvironment.hh"`
then i got error for `UsageEnvironment.hh` and fixed that and ...
i have fixed all 409 files, and now i get a new error:
```

In file included from D:\micro\2023\SDR Home Automation\esp32-audio cam\arduino-audio-tools-main\examples\examples-communication\rtsp\communication-rtsp-audiokit\communication-rtsp-audiokit.ino:16:
C:\Users\Josef\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/AudioClientRTSP.h: In member function 'void audio_tools::AudioClientRTSP::loop()':
C:\Users\Josef\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/AudioClientRTSP.h:127:80: error: 'class TaskScheduler' has no member named 'singleStep'
       if (audiotools_rtsp::rtspEventLoopWatchVariable==0) env->taskScheduler().singleStep();

```
[src.zip](https://github.com/pschatzmann/arduino-audio-tools/files/13251615/src.zip)
i don't know what to do now.

pschatzmann, 2023-11-03:
If you check the readme it tells you : Please note that this library only works properly for environments that can handle symbolic links correctly.

I guess you work with Windows, so you need to figure this one out yourself!

josef2600, 2023-11-03:
yes i am!

pschatzmann, 2023-11-03:
If you want to contribute you can clone the project and replace all symbolic links with include files that point to the right original files.

I guess you work with windows, so maybe there are other ways to deal with this.  On linux and osx this is working w/o problems.

josef2600, 2023-11-03:
what about `if (audiotools_rtsp::rtspEventLoopWatchVariable==0) env->taskScheduler().singleStep();  ` ?
now i get that error:
```
C:\Users\Josef\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/AudioClientRTSP.h: In member function 'void audio_tools::AudioClientRTSP::loop()':
C:\Users\Josef\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/AudioClientRTSP.h:127:80: error: 'class TaskScheduler' has no member named 'singleStep'
       if (audiotools_rtsp::rtspEventLoopWatchVariable==0) env->taskScheduler().singleStep();
```

josef2600, 2023-11-03:
P.S. since you have told me few times, it exist only in your wiki:
```
Protected Member Functions
 	BasicTaskScheduler (unsigned maxSchedulerGranularity)
 
void 	schedulerTickTask ()
 
virtual void 	SingleStep (unsigned maxDelayTime=0)
 
virtual void 	setBackgroundHandling (int socketNum, int conditionSet, BackgroundHandlerProc *handlerProc, void *clientData)
 
virtual void 	moveSocketHandling (int oldSocketNum, int newSocketNum)

```
and the source code is in the file:
`D:\micro\2023\SDR Home Automation\esp32-audio cam\arduino-live555-master\BasicUsageEnvironment\BasicTaskScheduler.cpp`
i guess i have to fined out myself!

pschatzmann, 2023-11-04:
I just found this: https://www.pschatzmann.ch/home/2022/08/07/symlinks-and-git/

pschatzmann, 2023-11-04:
And have you any news about the symlink support in your environment ?

josef2600, 2023-11-05:
> And have you any news about the symlink support in your environment ? I suggest to look into git: (git config --global core.symlinks true)

thank you. as i said, i changed all those files and i sent them to you too. if you like, please check them.
https://github.com/pschatzmann/arduino-audio-tools/files/13251615/src.zip
they don't cause problem about linkage. but i dont know why i get that error:  `error: 'class TaskScheduler' has no member named 'singleStep'`
mostly i thought you got tired of me! so i moved to HTTP protocol. and finally i got it to work. not that your code had problem, no, i just couldn't figure out how i can work with it. i still need to change few things in your source code about task managing. i like to make it work with 2 cores. any way, it was so oddly easy that i was like , woooooooooo! m.t.f ! i didn't say anything mostly, i thought you got tired of me, and i found how to do it in windows. and it is beautiful! now my esp32 has its own IP and any devices in net can send audio to it via esp32 IP and it play's. it is built in, in Windows 10 version 1803 and later. i just found it and haven't search for android or Linux, but i am sure they have it too. the info link [](https://curl.se/windows/microsoft.html)
and lets say i have my mp3 file in drive d: 
`curl -X POST -H "Content-Type: audio/mpeg" --data-binary @D:\k\5.mp3 http://192.168.1.10:7000/audio`
it works like a charm. and works with mp3 lame codec. it has jitter about 1% , but i don't care! so far, it is what i have don.
but you still have the problem with:
```
if (!client)
{
    client = server.available();  
}
```
thank you.

pschatzmann, 2023-11-05:
In the meantime I have also executed the script that I have sent to you and committed the resulting changes, so that windows is supported now as well. The compile times were bad, but now they are absolutely horrible after this change...

pschatzmann, 2023-11-05:
Oh, I see the problem now: the commits were not on the master branch. I merged my changes to master and deleted the working branch...

Did you finally manage to implement the solution with the AsynchServer ?

josef2600, 2023-11-05:
thank you again. i don't want to use AsynchServer. it was just for the starting point. i have my own AsynchServer, it doesn't hang! and doesn't leak either! but again, i don't want to use that way. because it would be slower than i like. i want to use multi tasking without use of mutex. the spi (I2S) can handle it, since it is a blocking function in Arduino. it remind of something, i have checked your source code, but i couldn't fined how to control the speed of spi(I2S). apparently, you are using some division for it and i cant fined out what is going on! it would be great if you could make it they way user can change the speed, like speed = 200000. as i understand, correctly, you are using 10mhz speed. i thing it is slow and 20mhz would be a lot better. ofcourse, it is too soon for me to give you a finished code, but i will. 
thank you again.

pschatzmann, 2023-11-05:
I dont quite understand what you mean with speed ? SPI and I2S are not the same thing!  

The sampling rate is reported by the decoder and each change is calling the setAudioInfo() method on the output which has been defined in the EncodedAudioStream.

josef2600, 2023-11-05:
so sorry, my bad. but in the page 310 of esp32 technical reference manual, it say's how the clocking of I2S works:
```
12.3 The Clock of I2S Module
As is shown in Figure 12-2, I2Sn_CLK, as the master clock of I2S module, is derived from the 160 MHz clock
PLL_F160M_CLK or the configurable analog PLL output clock APLL_CLK. The serial clock (BCK) of the I2S module
is derived from I2Sn_CLK. The I2S_CLKA_ENA bit of register I2S_CLKM_CONF_REG is used to select either
PLL_F160M_CLK or APLL_CLK as the clock source for I2Sn. PLL_F160_CLK is used as the clock source for
I2Sn, by default.
```
thank you.

pschatzmann, 2023-11-05:
The API is more relevant here: in my API it's part of the [config](https://pschatzmann.github.io/arduino-audio-tools/classaudio__tools_1_1_i2_s_config_e_s_p32.html) this is just driving the flag of the corresponding ESP32 I2S API