pschatzmann/arduino-audio-tools #1036: Task watchdog got triggered

josef2600, 2023-11-04:
### Problem Description

in example `communication-ip-receive.ino` if there is no client (noting connected to esp32), it crashes. reason being this:
```
  if (client.connected())
  {
    copier.copy();
  }
```
the ram gets full and boom crashhhh.
```

[   470][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
...............
192.168.1.10
starting I2S...
[I] AudioTypes.h : 96 - sample_rate: 16000
[I] AudioTypes.h : 97 - channels: 1
[I] AudioTypes.h : 98 - bits_per_sample: 16
[I] I2SConfigESP32.h : 90 - rx/tx mode: TX_MODE
[I] I2SConfigESP32.h : 91 - port_no: 0
[I] I2SConfigESP32.h : 92 - is_master: Master
[I] I2SConfigESP32.h : 93 - sample rate: 16000
[I] I2SConfigESP32.h : 94 - bits per sample: 16
[I] I2SConfigESP32.h : 95 - number of channels: 1
[I] I2SConfigESP32.h : 96 - signal_type: Digital
[I] I2SConfigESP32.h : 98 - i2s_format: I2S_STD_FORMAT
[I] I2SConfigESP32.h : 100 - auto_clear: true
[I] I2SConfigESP32.h : 102 - use_apll: true
[I] I2SConfigESP32.h : 107 - buffer_count:6
[I] I2SConfigESP32.h : 108 - buffer_size:512
[I] I2SConfigESP32.h : 113 - pin_bck: 14
[I] I2SConfigESP32.h : 115 - pin_ws: 15
[I] I2SConfigESP32.h : 117 - pin_data: 22
started...
E (14834) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (14834) task_wdt:  - IDLE (CPU 0)
E (14834) task_wdt: Tasks currently running:
E (14834) task_wdt: CPU 0: loopTask
E (14834) task_wdt: CPU 1: IDLE
E (14834) task_wdt: Aborting.

abort() was called at PC 0x400e51a5 on core 0

Backtrace: 0x40083a5d:0x3ffbecbc |<-CORRUPTED

```
if i remove` if (client.connected())` it wont crash. meaning, `copier.copy(); `has to always be running. can you fix it? and it is not just in here, anywhere that `copier.copy();` is in an if , it crashes if it is not running it. ether i have to end `I2SStream `, or run `copier.copy(); `for ever.
and i am using latest of all your libraries. 

### Device Description

esp32 - audio kit

### Sketch

```cpp
communication-ip-receive.ino
```

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2023-11-04:
No, I think your diagnosis is wrong: if there is nothing to copy you are just starving the task. 
I think the right way to deal with this is to use something like: if (!copy()) delay(100);

pschatzmann, 2023-11-05:
I could not reproduce your issue: I executed the example and let it run for quite some time.
Anyhow I committed a correction where I feed the dog if there is nothing to do...

If you check in the Board Manager: what ESP32 core version are you using ?

closed 2023-11-05 by pschatzmann

josef2600, 2023-11-05:
this is weird! wright now i tested it again. i changed nothing, but the ssid password. and yet:
```
started...
E (10337) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10337) task_wdt:  - IDLE (CPU 0)
E (10337) task_wdt: Tasks currently running:
E (10337) task_wdt: CPU 0: loopTask
E (10337) task_wdt: CPU 1: IDLE
E (10337) task_wdt: Aborting.

abort() was called at PC 0x400e3f81 on core 0

Backtrace: 0x40083a5d:0x3ffbec1c |<-CORRUPTED

```
i am using board manager v2.0.14

pschatzmann, 2023-11-05:
Was this after my correction ? 
What are the exact steps to reproduce this ?

josef2600, 2023-11-05:
i am using your yesterday code. no different.

pschatzmann, 2023-11-06:
Did you try to add the delay ?
I am just starting this sketch w/o server and I am getting:
```

09:39:20.612 -> entry 0x400805f0
09:39:21.583 -> .
09:39:21.584 -> 192.168.1.35
09:39:21.584 -> starting I2S...
09:39:21.584 -> [I] AudioTypes.h : 96 - sample_rate: 16000
09:39:21.584 -> [I] AudioTypes.h : 97 - channels: 1
09:39:21.584 -> [I] AudioTypes.h : 98 - bits_per_sample: 16
09:39:21.584 -> [I] I2SConfigESP32.h : 90 - rx/tx mode: TX_MODE
09:39:21.584 -> [I] I2SConfigESP32.h : 91 - port_no: 0
09:39:21.584 -> [I] I2SConfigESP32.h : 92 - is_master: Master
09:39:21.615 -> [I] I2SConfigESP32.h : 93 - sample rate: 16000
09:39:21.615 -> [I] I2SConfigESP32.h : 94 - bits per sample: 16
09:39:21.615 -> [I] I2SConfigESP32.h : 95 - number of channels: 1
09:39:21.615 -> [I] I2SConfigESP32.h : 96 - signal_type: Digital
09:39:21.615 -> [I] I2SConfigESP32.h : 98 - i2s_format: I2S_STD_FORMAT
09:39:21.615 -> [I] I2SConfigESP32.h : 100 - auto_clear: true
09:39:21.615 -> [I] I2SConfigESP32.h : 102 - use_apll: true
09:39:21.646 -> [I] I2SConfigESP32.h : 107 - buffer_count:6
09:39:21.646 -> [I] I2SConfigESP32.h : 108 - buffer_size:512
09:39:21.646 -> [I] I2SConfigESP32.h : 113 - pin_bck: 14
09:39:21.646 -> [I] I2SConfigESP32.h : 115 - pin_ws: 15
09:39:21.646 -> [I] I2SConfigESP32.h : 117 - pin_data: 22
09:39:21.646 -> started...
```
So what are you doing differently ?

josef2600, 2023-11-07:
thank you. right now i am working on my multi server, i just got this message!
`[I] StreamCopy.h : 74 - buffer_size=-1208848456`
funny! i don't know why!
it doesn't hang either!

josef2600, 2023-11-09:
thank you. sorry i forgot to tell you. apparently, it is fixed! exactly the day i last downloaded it, you changed few files. now it doesn't reset with your example. thank you. currently, i am looking for the fastest decoder so i can receive more than 1 connection audio. i know mp3 is not good, it is very slow. i don't know why i cant make PCM to work! (some times i get really stooped!). but i don't think it would be a good choice, since it has a high throughout data rate. i am going for libg7xx to see what will happens.

pschatzmann, 2023-11-09:
I suggest to have a look at adpcm.

josef2600, 2023-11-09:
i am using GSM, but it doesn't play audio and only shows this massage:
```
[E] CodecGSM.h : 103 - gsm_decode
[E] CodecGSM.h : 103 - gsm_decode
[E] CodecGSM.h : 103 - gsm_decode
[E] CodecGSM.h : 103 - gsm_decode
[E] CodecGSM.h : 103 - gsm_decode
[E] CodecGSM.h : 103 - gsm_decode
[E] CodecGSM.h : 103 - gsm_decode
[E] CodecGSM.h : 103 - gsm_decode
```
i have change buffer to 128 and 512 , but no changes. this is the audio file:
[12.zip](https://github.com/pschatzmann/arduino-audio-tools/files/13306239/12.zip)

pschatzmann, 2023-11-09:
Did you read the spec. If i remember right tmit supports only 1 channel with a rate of 8000.

josef2600, 2023-11-09:
yes, but i dont see any setting for converter. i am using Switch Sound File Converter .

josef2600, 2023-11-09:
i am trying for [ADPCM](https://github.com/pschatzmann/adpcm#adpcm-library)

josef2600, 2023-11-09:
no luck for that too. i am using .wav file with 2 channel and 16000 sample rate and 16bit int and pcm uncompressed. this is the loge:
```
ERROR: step_index[0] = 21587
Free heap:   210272
ERROR: step_index[0] = 13617
Free heap:   203424
ERROR: step_index[1] = 256
ERROR: step_index[0] = 256
ERROR: step_index[0] = -513
ERROR: step_index[0] = -513
ERROR: step_index[0] = 1024
ERROR: step_index[0] = 1024
ERROR: step_index[1] = -1
ERROR: step_index[0] = -256
ERROR: step_index[0] = -1281
ERROR: step_index[0] = -769
ERROR: step_index[0] = 1536
ERROR: step_index[0] = 1536
ERROR: step_index[0] = -1281
ERROR: step_index[0] = -1025
ERROR: step_index[0] = 768
ERROR: step_index[0] = 2560
ERROR: step_index[0] = 1024
ERROR: step_index[0] = -1793
ERROR: step_index[0] = -2817
ERROR: step_index[0] = -769
ERROR: step_index[0] = 2560
ERROR: step_index[0] = 3328
ERROR: step_index[0] = 767
ERROR: step_index[0] = -1537
ERROR: step_index[0] = -1793
ERROR: step_index[0] = -257
ERROR: step_index[0] = 2048
ERROR: step_index[0] = 1536
ERROR: step_index[0] = 255

```
and this is my code:
```

//curl -X POST -H "Content-Type: audio/mpeg" --data-binary @D:\k\5.mp3 http://192.168.1.10:7000/audio
//curl -X POST -H "Content-Type: audio/wav" --data-binary @D:\k\14.wav http://192.168.1.10:7000/audio

/**
 * @file example-serial-receive.ino
 * @author Phil Schatzmann
 * @brief Receiving audio via IP and writing to I2S
 * @version 0.1
 * @date 2022-03-09
 * 
 * @copyright Copyright (c) 2022
 */

/**
 * @file AudioKitSettings.h
 * @author Phil Schatzmann
 * @brief AUDIOKIT_BOARD selects a specic board:
 *  1) lyrat_v4_3
 *  2) lyrat_v4_2
 *  3) lyrat_mini_v1_1
 *  4) esp32_s2_kaluga_1_v1_2
 *  5) ai_thinker (ES8388) 2957 3478 A149
 *  6) ai_thinker (AC101) 2473 2762 2957
 *  7) ai_thinker (ES8388) 2957
 *  8) esp32_s3_box
 *  9) esp32_s3_box_lite
 *  10) generci_es8388
 *  11) generic_es8311
 *  12) generic (no defaults)
 * @version 0.1
[... 153 lines cut ...]
      
    }
    else if (my_clients[i])
    {
      my_clients[i].stop();
    }
  }

  for (uint8_t i = 0; i < MAX_CLIENTS; i++)
  {
    copiers[i]->copy();
  }

  if ((lastCheckedTime + 3000) < millis())
  {
    lastCheckedTime = millis();
    Serial.print("Free heap:   ");
    Serial.println(ESP.getFreeHeap());
    //delay(1);
  }

  //delay(100);
}
```
i dont know what i am doing wrong? it works with MP3, but nothing more. and for mp3 if more than 1 client connects to it, the audio gets very bad.

pschatzmann, 2023-11-09:
This will be tricky to find: It seems that somehow some bytes get lost or added. MP3 does not care too much about this, but other decoders are quite picky...

josef2600, 2023-11-09:
i found something that is not very nice, in my opinion. in file:
`arduino-audio-tools-main\src\AudioHttp\URLStreamBuffered.h`
you have created a task, and in line 122 you have this:
`delay(100);`
why this much delay? it is not necessary at all. 1 ms is fine. but more important thing is in line 128. where is the end of task? lets say if ` if (*(self->p_stream) && available_to_write>0)` is always true, then `delay`  never runs and after 3 second we get a nice reset! please do not add another delay. just add `taskYIELD();` at the end of while, in line 128.

josef2600, 2023-11-09:
and i was wondering, if should i continue writing farther my info and test in here or create a new subject? i am very bad at GitHub!

pschatzmann, 2023-11-09:
You should create new discussions for new topics other then the task watchdog...