pschatzmann/arduino-audio-tools #1039: Is AnalogAudioStream supported on ESP32-S3?

uutzinger, 2023-11-06:
### Problem Description

When I open basic-adc-serial and I attempt to compile it I get:
error: 'AnalogAudioStream' does not name a type

My goals is to ready with the internal A/D converter.

### Device Description

Hardware: Adafruit Feather ESP32-S3 2MB PSRAM
Board Manager: esp32 by Espressif Systems version 2.0.14
Selected Board: ESP32S3 Dev Module (not the Adafruit defined board)

### Sketch

```cpp
The example program base-adc-serial
```

### Other Steps to Reproduce

Not applicable.

### What is your development environment

Arduino IDE 2.2.1
Selected Board: ESP32S3 Dev Module (not the Adafruit defined board)

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2023-11-06:
To use the ADC functionality of the S3 you need to have the new ADC API available from Espressif. This is not the case in the version you are using: you can find the version in the Board Manager.

This is only available starting from ESP32 3.0. Fortunately there is a pre-release available.

The other option is to add a USE_ANALOG_ARDUINO in the AudioConfig.h and you might get away with this if you keep the sampling rates very low.

uutzinger, 2023-11-07:
Thanks very much for your support.

I tried with alpha2. Using base-adc-serial unfortunately I get 
src/AudioAnalog/AnalogConfigESP32V1.h:62:5: error: 'dac_channel_mask_t' does not name a type; did you mean 'adc_channel_t'?

I guess its almost working.

P.S. When I used the new AnalogReadContinous example in 3.x I can measure with ESP32-S3 at about 20Khz but regularly it drops one reading.  At 10kHz its about once a second. I am looking forward trying your AudioStream instead.

pschatzmann, 2023-11-07:
Yes, this error was introduced with one of the last changes to make the dac pin selectable.
The correction has been committed...

uutzinger, 2023-11-07:
Thank you very much! 
I tried to compile some of your examples on the different board options for Arduino ESP32 Core version 3.0.0 alpha2 **before your fix** and found that ESP32C3, ESP32S3, ESP32C6 and ESP32H2 had issues. 
**After the fix** _C3 and S3 is working_, but there are remaining issues with ESP32C6  with 'ADC_CONV_MODE', 'ADC_OUTPUT_TYPE', 'ADC_CHANNELS', 'touchRead', and ESP32H2 with 'i2s_pdm_rx_slot_config_t', 'i2s_pdm_rx_clk_config_t', 'i2s_pdm_rx_config_t', 'pdm_rx_cfg' 'i2s_channel_init_pdm_rx_mode' and 'touchRead'. I don't have those models but I guess sooner or later others will be asking...

pschatzmann, 2023-11-07:
I added the missing variants to the AudioConfig.h

uutzinger, 2023-11-07:
Cool. 
Unfortunately when I use 

```
uint16_t sample_rate = 100;
uint8_t  channels = 2;
uint8_t  bits_per_sample = 12;
AudioInfo info(sample_rate, channels, bits_per_sample);
```
I get regardless of sample_rate

`E (228) adc_continuous: adc_continuous_config(511): ADC sampling frequency out of range`

When I use the stock base-adc-serial example (default config) I get

```
E (227) adc_continuous: adc_continuous_config(511): ADC sampling frequency out of range
E (238) adc_continuous: adc_continuous_read(414): The driver is already stopped
```

Almost there.

pschatzmann, 2023-11-07:
Well 100 does not make any sense. 
In audio values start usually with 8000

uutzinger, 2023-11-07:
Yes I started with 44100 and went down with the values. So no value between 100 and 44100 are accepted.

pschatzmann, 2023-11-08:
You should be ablee to determine the available range for your processor by printing the following values

#define SOC_ADC_SAMPLE_FREQ_THRES_HIGH          (2*1000*1000)
#define SOC_ADC_SAMPLE_FREQ_THRES_LOW           (20*1000)

The values above are for the ESP32, so I expect it to be different for your board

uutzinger, 2023-11-08:
Phil
I looked into this some further by digging into the source code (yours, Arduino-ESP32 and ESP33 IDL) and when I tried to run your code on an older ESP32 Dev Kit, I realized this is a problem not specific to ESP32S3 but to all version on attempting Arduino ESP32 Core version 3.0.0 alpha. I will create separate ticket explaining what I found so far. For example the numbers above SOC_ADC_SAMPLE_FREQ_THRES_LOW/HIGH actually translate to like 600 something and 80000 somethin when you add a boundary check in AnalogAudioESP32V1.h and print them (e.g. when you set sample frequency to 20000 it states its 633). What puzzles me is that the Example provided by Espressif for Continuous A/DC is able to set the continuous configuration without error while it does not work with your code, although code section under the hood for configuring are similar.

Or in short: At this time your base-adc_serial  compiles but fails to set sampling rate with ESP32 Core Version 3.0.0 alpha2.

pschatzmann, 2023-11-09:
Strange, I used the example from Espressif as starting point for my implementation.
I double check if I can find the difference...

pschatzmann, 2023-11-09:
Oops, there were some pretty ugly bugs: e.g. I did not set the sampling rate!

I committed a correction and a [test case](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/tests/adc/read/read.ino). But there is still something wrong: I am measuring a sampling rate which is much higer. But at least it seems that we are getting some data now...

pschatzmann, 2023-11-09:
I committed some more corrections: I removed the shadowed channels and bits_per_sample and all conversions so that we can better understand the provided data.
One last topic: I assume the result will be unsigned, but in audio we prefer to have signed values. So there might be one final conversion necessary...

pschatzmann, 2023-11-17:
https://github.com/pschatzmann/arduino-audio-tools/pull/1049

mehran-mohammadi, 2023-11-23:
Hi, I have an issue with esp32-s3, more specifically the board I am using is lilygo t-display s3. I installed ESP version 3.0 alpha 2 and this solved the compile issue. however, After uploading the code, I get no sound! I have checked all the pins on the board but there is no sound...
Do I need to change anything in the configuration files or the GPIO pins, since ESP32-S3 does not have default pins (25 and 26) on board?

pschatzmann, 2023-11-23:
I am not really sure what functionality you are talking about, but I have the impression that you try to use the __non existing DAC__ of the ESP32-S3! 

If you want to output sound you need to use I2S or PWM on this processor...

mehran-mohammadi, 2023-11-23:
Thank you for your answer. I will try them.

closed 2023-12-11 by pschatzmann