pschatzmann/arduino-audio-tools #1054: Recording a series of WAV files

fugufisch, 2023-11-14:
### Problem Description

I have a an issue with recording a series of files to an SD card. The first file works flawlessly, but the following files are corrupted (the data is there, but my suspicion is that the header is missing).
I adapted your example code from [here](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/streams-audiokit-sd-audiokit/streams-audiokit-sd-audiokit.ino).

As you can see in the code, I set the `out` stream and the copier back to `begin()` and I also tried re-initiating the AudioKitStream.

I apologize if this isn't a bug, but just my lack of understanding or misuse of the library, but then maybe it's worth showing an example for this.

### Device Description

ESP32-A1S AudioKit

### Sketch

```cpp
#include "AudioTools.h"
#include "AudioLibs/AudioKit.h"
#include <SD.h>

void record_start(bool pinStatus, int pin, void *ref)
{
  String newFileName = generateFileName(); 
  Serial.println("Recording to " + newFileName + "...");
  file = SD.open(newFileName.c_str(), FILE_WRITE);
  if (!file)
  {
    Serial.println("Failed to open file for recording");
    return;
  }
  out.begin(info);
  copier.begin(out, in);
  recording = true;
}

void record_end(bool pinStatus, int pin, void *ref)
{
  if (recording == true)
  {
    Serial.println("Stop.");
    copier.end();
    file.close();
    recording = false;
    Serial.println("Press key1 to record.");
  }
}

void setup()
{
  Serial.begin(115200);

  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // setup input
  auto cfg = in.defaultConfig(RX_MODE);
  cfg.sd_active = true;
  cfg.copyFrom(info);
  cfg.input_device = AUDIO_HAL_ADC_INPUT_ALL; // microphone
  in.begin(cfg);

  // Open SD drive
  if (!SD.begin(PIN_AUDIO_KIT_SD_CARD_CS))
  {
    Serial.println("Initialization failed!");
    stop();
  }

  // setup output
  out.begin(info);

  // record when key 1 is pressed
  in.audioActions().add(PIN_KEY1, record_start, record_end);
  Serial.println("Press Key 1 to record");
}

void loop()
{
  // record to wav file
  if (recording)
  {
    copier.copy();
  }

  // Process keys
  in.processActions();
}
```

### What is your development environment

PlatformIO

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2023-11-14:
You are not writing any wav files but just binary files!
If you need wav you will need to write them thru a wav encoder...

closed 2023-11-14 by pschatzmann

fugufisch, 2023-11-14:
Sorry, I forgot to add that part to the encoder. I tried to leave in only the important bits.
Here's the declarations:
```
// audio
const char *file_name = "/rec.wav";
AudioInfo info(16000, 1, 16);
AudioKitStream in;
File file; // final output stream
EncodedAudioStream out(&file, new WAVEncoder());
StreamCopy copier; // copies data
uint64_t timeout;
bool recording = false; // flag to make sure that close is only executed one
uint64_t end_time;      // trigger to call endRecord
```

pschatzmann, 2023-11-14:
Try to display the file header e.g. with a hex dump to confirm that the header is really missing.
Not sure if this is necessary, but I would also tell out that the file has changed with out.setOutput(&file)

fugufisch, 2023-11-14:
Ok, looking at the hex confirms my suspicion. First file has the header, next is missing it.
Here are the logs that I get for two consecutive key presses/recordings. There is also no logging of the header write.

```
[I] AudioTypes.h : 104 - sample_rate: 16000
[I] AudioTypes.h : 105 - channels: 1
[I] AudioTypes.h : 106 - bits_per_sample: 16
[I] AudioTypes.h : 103 - Info from AudioKitStream:
[I] AudioTypes.h : 104 - sample_rate: 16000
[I] AudioTypes.h : 105 - channels: 1
[I] AudioTypes.h : 106 - bits_per_sample: 16
[I] I2SConfigESP32.h : 90 - rx/tx mode: RX_MODE
[I] I2SConfigESP32.h : 91 - port_no: 0
[I] I2SConfigESP32.h : 92 - is_master: Master
[I] I2SConfigESP32.h : 93 - sample rate: 16000
[I] I2SConfigESP32.h : 94 - bits per sample: 16
[I] I2SConfigESP32.h : 95 - number of channels: 1
[I] I2SConfigESP32.h : 96 - signal_type: Digital
[I] I2SConfigESP32.h : 98 - i2s_format: I2S_STD_FORMAT
[I] I2SConfigESP32.h : 100 - auto_clear: true
[I] I2SConfigESP32.h : 102 - use_apll: true
[I] I2SConfigESP32.h : 107 - buffer_count:6
[I] I2SConfigESP32.h : 108 - buffer_size:512
[I] I2SConfigESP32.h : 111 - pin_mck: 0
[I] I2SConfigESP32.h : 113 - pin_bck: 27
[I] I2SConfigESP32.h : 115 - pin_ws: 26
[I] I2SConfigESP32.h : 117 - pin_data: 35
[W] AudioKit.h : 63 - sd_active = true
[I] AudioTypes.h : 104 - sample_rate: 16000
[I] AudioTypes.h : 105 - channels: 1
[I] AudioTypes.h : 106 - bits_per_sample: 16
[I] I2SConfigESP32.h : 90 - rx/tx mode: RX_MODE
[I] I2SConfigESP32.h : 91 - port_no: 0
[I] I2SConfigESP32.h : 92 - is_master: Master
[I] I2SConfigESP32.h : 93 - sample rate: 16000
[I] I2SConfigESP32.h : 94 - bits per sample: 16
[I] I2SConfigESP32.h : 95 - number of channels: 1
[I] I2SConfigESP32.h : 96 - signal_type: Digital
[I] I2SConfigESP32.h : 98 - i2s_format: I2S_STD_FORMAT
[I] I2SConfigESP32.h : 100 - auto_clear: true
[I] I2SConfigESP32.h : 102 - use_apll: true
[I] I2SConfigESP32.h : 107 - buffer_count:6
[I] I2SConfigESP32.h : 108 - buffer_size:512
[I] I2SConfigESP32.h : 111 - pin_mck: 0
[I] I2SConfigESP32.h : 113 - pin_bck: 27
[I] I2SConfigESP32.h : 115 - pin_ws: 26
[I] I2SConfigESP32.h : 117 - pin_data: 35
[I] AudioKit.h : 605 - void audio_tools::AudioKitStream::setupActions()
[W] AudioKit.h : 612 - Mode Button ignored because of conflict: 13 
[W] AudioKit.h : 620 - Headphone detection ignored because of conflict: 5 
[I] AudioKit.h : 355 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 42 - ActionLogic::add pin: 5 / logic: 0
[I] AudioActions.h : 52 - pin 5 -> INPUT_PULLUP
[I] AudioKit.h : 355 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 42 - ActionLogic::add pin: 18 / logic: 0
[I] AudioActions.h : 52 - pin 18 -> INPUT_PULLUP
[I] CodecWAV.h : 539 - sample_rate: 16000
[I] CodecWAV.h : 540 - channels: 1
[I] CodecWAV.h : 549 - is_streamed! because length is 2147418112
[I] AudioActions.h : 42 - ActionLogic::add pin: 36 / logic: 0
[I] AudioActions.h : 52 - pin 36 -> INPUT_PULLUP
Press Key 1 to record
Recording to /rec_1.wav...
[I] StreamCopy.h : 82 - buffer_size=1024
[I] CodecWAV.h : 588 - Writing Header
[I] StreamCopy.h : 140 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] StreamCopy.h : 140 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] StreamCopy.h : 140 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] StreamCopy.h : 140 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] StreamCopy.h : 140 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[... 16 lines cut ...]
[I] StreamCopy.h : 140 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] StreamCopy.h : 140 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] StreamCopy.h : 140 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
Stop.
Press key1 to record.
Recording to /rec_2.wav...
[I] StreamCopy.h : 82 - buffer_size=1024
[I] StreamCopy.h : 140 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] StreamCopy.h : 140 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] StreamCopy.h : 140 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] StreamCopy.h : 140 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] StreamCopy.h : 140 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] StreamCopy.h : 140 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] StreamCopy.h : 140 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] StreamCopy.h : 140 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] StreamCopy.h : 140 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] StreamCopy.h : 140 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] StreamCopy.h : 140 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] StreamCopy.h : 140 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] StreamCopy.h : 140 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] StreamCopy.h : 140 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] StreamCopy.h : 140 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
Stop.
Press key1 to record.
```

pschatzmann, 2023-11-14:
Try to use log level debug to figure out why this is not called

pschatzmann, 2023-11-14:
Just double checked the code:   out.begin(info); is not doing anything if it is already open!
If you add out.end() in your record_end() it should work...

fugufisch, 2023-11-14:
Thank you so much! That solved it. 
Sorry to bother you with a false alarm. Really appreciate your work on this!