pschatzmann/arduino-audio-tools #1063: Include file 'CodecAPTX.h' tries to include file that does not exist anymore

Razzorback225, 2023-11-20:
### Problem Description

Hi, when I tried to compile my code including the header 'AudioCodecs/CodecAPTX.h', Arduino IDE gives me error saying that the following file does not exist : "AudioBasic/Int24x.h".

While navigating to the repo I found, indeed, that it does not exist (or not anymore). I replaced in the header the include of Int24x.h by the following 'Int24_4bytes_t.h' and the sketch compiled without error.

### Device Description

Custom ESP32 audio board.

### Sketch

```cpp
#include 'AudioCodecs/CodecAPTX.h'
```

### What is your development environment

Arduino IDE 1.8.x

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2023-11-20:
Thanks for letting me know: I must have missed this one...
The correction has been committed

Razzorback225, 2023-11-20:
Thanks I'll close the issue then :)

closed 2023-11-20 by Razzorback225