pschatzmann/arduino-audio-tools #1067: Is APTX, AAC and other codec are really supported on A2DP Sink?

Razzorback225, 2023-11-22:
### Problem Description

Regarding some informations gathered from an Espressif document, the ESP32 should only be capable of using SBC codec for Bluetooth. 

If so, then the ESP will always advertise other peripherals that it is compatible with SBC only and you should not be able to get APTX encoded audio?

[image]

I already tried to use the APTX codec in this way, but it only caused reboot or connection failure when I tried to connect my phone to the ESP :

A2DPSink stream > APTXDecoder > I2SStream

But accordingly to what I discovered it would make sense.

So, do I misunderstood the chain of Stream and components works or am I correct?

Thanks in advance.

N.B. : This is not an issue but I don't know where to post this question

### Device Description

Custom ESP32 board using WROOM32UE-H16

### Sketch

```cpp
Not relevant
```

### What is your development environment

Arduino IDE 1.8.x

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2023-11-22:
Espressif is only supporting SBC! 

If I remember right, I have seen a project on Github which was extending the IDF to support other codecs for A2DP. 
But this is nothing that would work under Arduino!

In any case from an API point of view you will always need to provide and you get PCM data

closed 2023-11-22 by pschatzmann

Razzorback225, 2023-11-22:
Thanks for these explainations :)