pschatzmann/arduino-audio-tools #1068: max9814 not loud enough

sufius, 2023-11-22:
### Problem Description

recording is working, but it should be louder or how to make it louder?

### Device Description

I'm using an esp32-DevKitC v4 connected to max9814 Microphone Amplifier at port 34 (default) and a normal sd card module at 5,18,19,23 (default)

### Sketch

```cpp
#include <SD.h>
#include <SPI.h>

#include "AudioTools.h"
#include "AudioCodecs/CodecWAV.h"
#include "JC_Button.h" // https://github.com/JChristensen/JC_Button

// I2S Connections
#define PMB_PIN_I2S_LRC 26
#define PMB_PIN_I2S_BCLK 27
#define PMB_PIN_I2S_DIN 25

const int ledRedPin = 4;
const int recordButtonPin = 15;
bool recordMode = false;
bool recording = false;

Button recordButton(recordButtonPin);
Button button13(13);

File audioFile; // final output stream

I2SStream speaker; // final output of decoded stream
VolumeStream volume(speaker);
EncodedAudioStream decoder(&volume, new WAVDecoder()); // Decoding stream
StreamCopy player;

AnalogAudioStream mic;
EncodedAudioStream encoder(&audioFile, new WAVEncoder());
StreamCopy recorder(encoder, mic); // copies data

void startRecording(const char *fileName)
{
  recording = true;
  // Cleanup if necessary
  if (SD.exists(fileName))
    SD.remove(fileName);

  // open file
  audioFile = SD.open(fileName, FILE_WRITE);
  if (!audioFile)
  {
    Serial.println("file operation failed!");
    recording = false;
    recordMode = false;
    stop();
  }

  Serial.println(recorder.bufferSize());
}

void stopRecording()
{
  digitalWrite(ledRedPin, LOW);
  recordMode = false;
  recording = false;
  // close file when done
  if (audioFile)
  {
    audioFile.flush();
    Serial.print("File has ");
    Serial.print(audioFile.size());
    Serial.println(" bytes");
  }
  else
  {
    Serial.println("closing failed");
  }
}

void setup()
{
  Serial.begin(115200);
  //  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  pinMode(ledRedPin, OUTPUT);
  digitalWrite(ledRedPin, LOW);

  recordButton.begin();
  button13.begin();

  // Open SD drive
  if (!SD.begin(5))
  {
    Serial.println("Initialization failed!");
    stop();
  }
  else
  {
    Serial.println("Initialization successful!");
  }

  int sample_rate = 32000;
  // int sample_rate = 16000; // using this making it recording in slow motion
  int bits_per_sample = 32;

  // // for playback
  auto speakerConfig = speaker.defaultConfig(TX_MODE);
  speakerConfig.channels = 1;
  speakerConfig.sample_rate = sample_rate;
  speakerConfig.bits_per_sample = bits_per_sample;
  speakerConfig.pin_data = PMB_PIN_I2S_DIN;
  speakerConfig.pin_ws = PMB_PIN_I2S_LRC;
  speakerConfig.pin_bck = PMB_PIN_I2S_BCLK;
  speakerConfig.i2s_format = I2S_LSB_FORMAT;
  speakerConfig.port_no = 1; // Since micConfig initializes port_no on 0, we have to change either the mic or speaker port numbers.
  speaker.begin(speakerConfig);

  auto volume_config = volume.defaultConfig();
  volume.begin(volume_config);
  volume.setVolume(1);

  decoder.setNotifyAudioChange(speaker);
  decoder.begin();

  // // for recording
  auto micConfig = mic.defaultConfig(RX_MODE);
  micConfig.sample_rate = sample_rate;
  micConfig.bits_per_sample = bits_per_sample;
  micConfig.channels = 1;
  mic.begin(micConfig);

  auto cfg_out = encoder.defaultConfig();
  cfg_out.channels = 1;
  cfg_out.sample_rate = sample_rate;
  cfg_out.bits_per_sample = bits_per_sample;
  encoder.begin(cfg_out);

  recorder.setCheckAvailableForWrite(false);
  recorder.begin(encoder, mic);
}

void loop()
{
  recordButton.read();
  if (!recordMode && recordButton.pressedFor(500))
  {
    Serial.println("Record started");
    recordMode = true;
    digitalWrite(ledRedPin, HIGH);
  }

  button13.read();
  if (recordMode && button13.isPressed())
  {
    if (!recording)
    {
      startRecording("/00003.wav");
    }
    else
    {
      Serial.println("recording");
      recorder.copy();
    }
  }
  else if (recordMode && recording && button13.wasReleased())
  {
    Serial.println("recording stopped");
    stopRecording();
  }
  else if (!recordMode && !recording && button13.wasReleased())
  {
    Serial.println("button13.wasReleased");
    audioFile = SD.open("/00003.wav");
    player.begin(decoder, audioFile);
  }

  if (!recordMode && !recording)
  {
    if (!player.copy())
    {
      audioFile.close(); // close file after playback
    }
  }
}
```

### What is your development environment

Working with platformIO

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2023-11-22:
Just read the documentation about [volume control](https://github.com/pschatzmann/arduino-audio-tools/wiki/Volume-Control)!
You might need to update to use the latest version to use the amplification functionality!

closed 2023-11-22 by pschatzmann

sufius, 2023-11-23:
I'm sorry for annyoing again, I've red the documentation about VolumeControl, and I tried really everything, even this approach with the logarithmic volume control. I added the VolumeStream to the input side like u adviced above. when setting the volume at input side on f.e. 0.3 it is getting more silent - thus it is working, but when putting a value above 1.0 (yes, i've set allow_boost on true) it is not getting louder than 1.0, although the debug screen is showing the correct values. what im doing wrong. playing  mp3 files are playbacked very very loud, so the problem is not the output i guess. thank in advance. btw i think im running the newest version: 0.9.6.

```
I2SStream speaker; // final output of decoded stream
VolumeStream volumeSpeaker(speaker);
EncodedAudioStream decoder(&volumeSpeaker, new WAVDecoder()); // Decoding stream
StreamCopy player;

I2SStream mic;
VolumeStream volumeMic(mic);
EncodedAudioStream encoder(&audioFile, new WAVEncoder());
StreamCopy recorder; // copies data

...
...

  int samples_per_second = 16000;
  // int samples_per_second = 22050;
  // int samples_per_second = 30000;
  // int samples_per_second = 32000;
  // int samples_per_second = 44100;
  int bits_per_sample = 16;
  int channels = 1;
  float volume = 1.0;

  // // for playback
  // setup audio output
  auto speakerConfig = speaker.defaultConfig(TX_MODE);
  speakerConfig.channels = channels;
  speakerConfig.sample_rate = samples_per_second;
  speakerConfig.bits_per_sample = bits_per_sample;
  speakerConfig.pin_data = PMB_PIN_I2S_DIN;
  speakerConfig.pin_ws = PMB_PIN_I2S_LRC;
  speakerConfig.pin_bck = PMB_PIN_I2S_BCLK;
  speakerConfig.i2s_format = I2S_LSB_FORMAT;
  speakerConfig.port_no = 1; // Since micConfig initializes port_no on 0, we have to change either the mic or speaker port numbers.
  speaker.begin(speakerConfig);

  auto decoderConfig = decoder.defaultConfig();
  decoderConfig.channels = channels;
  decoderConfig.sample_rate = samples_per_second;
  decoderConfig.bits_per_sample = bits_per_sample;
  decoder.setNotifyAudioChange(speaker);
  decoder.begin(decoderConfig);

  auto volumeSpeakerConfig = volumeSpeaker.defaultConfig();
  volumeSpeakerConfig.allow_boost = true;
  volumeSpeakerConfig.channels = channels;
  volumeSpeakerConfig.sample_rate = samples_per_second;
  volumeSpeakerConfig.bits_per_sample = bits_per_sample;
  volumeSpeaker.begin(volumeSpeakerConfig);
  volumeSpeaker.setVolume(volume);

  // // for recording
  // setup audio input
  auto micConfig = mic.defaultConfig(RX_MODE);
  micConfig.pin_ws = 32;
  micConfig.pin_bck = 33;
  micConfig.pin_data = 15;
  micConfig.sample_rate = samples_per_second;
  micConfig.bits_per_sample = bits_per_sample;
  micConfig.channels = channels;
  mic.begin(micConfig);

  // setup codec
  auto encoderConfig = encoder.defaultConfig();
  encoderConfig.channels = channels;
  encoderConfig.sample_rate = samples_per_second;
  encoderConfig.bits_per_sample = bits_per_sample;
  encoder.begin(encoderConfig);

  auto volumeMicConfig = volumeMic.defaultConfig();
  volumeMicConfig.allow_boost = true;
  volumeMicConfig.channels = channels;
  volumeMicConfig.sample_rate = samples_per_second;
  volumeMicConfig.bits_per_sample = bits_per_sample;
  volumeMic.begin(volumeMicConfig);
  volumeMic.setVolume(volume);

  recorder.setCheckAvailableForWrite(false);
  recorder.begin();
```

pschatzmann, 2023-11-23:
Did you update the library to the latest version ?

sufius, 2023-11-23:
Thank you, after a `git pull` everything works fine, Thank you Sir, you made my day.