pschatzmann/arduino-audio-tools #1073: MedianFilter example not found/working

sufius, 2023-11-25:
### Problem Description

I'm trying to reduce the noise in the input of my microphone by the help of the MedianFilter-Class. Unfortunately I'm not able to find an example, just examples of the FIR-Filter class. When trying to addapt the FIR-example to the MedianFilter-example I'm getting following error:

`MedianFilter.h:18:35: error: cannot convert 'audio_tools::Vector<audio_tools::MedianFilter<float>::sMedianNode_t>' to 'audio_tools::MedianFilter<float>::sMedianNode_t*' in assignment`

I'm sure it is a very stupid thing which could be solved in seconds^^ I'm apriory sorry for inconvenience

### Device Description

I'm using the ESP32 with the newest audio-library version 0.9.6 (git pull included)

### Sketch

```cpp
int channels = 1;
AnalogAudioStream mic;
VolumeStream volumeMic(mic);
FilteredStream<int16_t, float> filtered(volumeMic, channels);  // Defiles the filter as BaseConverter
EncodedAudioStream encoder(&audioFile, new WAVEncoder());
StreamCopy recorder;

...

void startRecording(const char *fileName)
{
  recording = true;
  recorder.begin(encoder, filtered);
}

void setup()
{
...
  // //             _                      _                      
  // //   _ __ ___ (_) ___ _ __ ___  _ __ | |__   ___  _ __   ___ 
  // //  | '_ ` _ \| |/ __| '__/ _ \| '_ \| '_ \ / _ \| '_ \ / _ \
  // //  | | | | | | | (__| | | (_) | |_) | | | | (_) | | | |  __/
  // //  |_| |_| |_|_|\___|_|  \___/| .__/|_| |_|\___/|_| |_|\___|
  // //                             |_|                           
  filtered.setFilter(0, new MedianFilter<float>());
  // setup audio input
  auto micConfig = mic.defaultConfig(RX_MODE);
 ...
```

### What is your development environment

PlatformIO

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2023-11-26:
I started to work on this, but never got it completed.

Today I committed some corrections and the missing [test case](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/tests/filters/test-median-filter/test-median-filter.ino).

This is working amazingly well: it removed all the cracking sound from the test noise, but I am not sure if this will help with your microphone input...

closed 2023-11-26 by pschatzmann

sufius, 2023-11-26:
I indeed can hear a difference. Thank you very much.