pschatzmann/arduino-audio-tools #1075: Audiokit example crashes when decoding AAC-stream

LieBtrau, 2023-11-26:
### Problem Description

Hi,
I'm using a ESP32 Audio Kit V2.2 (2762).  I few months ago, I tried [this example](https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-audiokit/streams-url_aac-audiokit) and it worked fine.  I created [my own example](https://github.com/LieBtrau/esp32-web-radio/tree/main/firmware/esp32/platformio/esp32a1s-1stTest), based on it.  It worked as well.

---
This morning I cleaned my build directory, upon which the most recent versions of the libraries (https://github.com/pschatzmann/arduino-audiokit,     https://github.com/pschatzmann/arduino-audio-tools and https://github.com/pschatzmann/arduino-libhelix) have been downloaded.  
My code crashes.  I tried the latest version of your streams-url_aac-audiokit example and it crashed as well.  Below you can find the serial logging.

FYI : Decoding MP3-stream still works fine.

```
ELF file SHA256: 461950aabbe5302a

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[    31][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[I] AudioTypes.h : 104 - sample_rate: 44100
[I] AudioTypes.h : 105 - channels: 2
[I] AudioTypes.h : 106 - bits_per_sample: 16
[I] AudioTypes.h : 103 - Info from AudioKitStream:
[I] AudioTypes.h : 104 - sample_rate: 44100
[I] AudioTypes.h : 105 - channels: 2
[I] AudioTypes.h : 106 - bits_per_sample: 16
[I] I2SConfigESP32.h : 90 - rx/tx mode: TX_MODE
[I] I2SConfigESP32.h : 91 - port_no: 0
[I] I2SConfigESP32.h : 92 - is_master: Master
[I] I2SConfigESP32.h : 93 - sample rate: 44100
[I] I2SConfigESP32.h : 94 - bits per sample: 16
[I] I2SConfigESP32.h : 95 - number of channels: 2
[I] I2SConfigESP32.h : 96 - signal_type: Digital
[I] I2SConfigESP32.h : 98 - i2s_format: I2S_STD_FORMAT
[I] I2SConfigESP32.h : 100 - auto_clear: true
[I] I2SConfigESP32.h : 102 - use_apll: true
[I] I2SConfigESP32.h : 107 - buffer_count:6
[I] I2SConfigESP32.h : 108 - buffer_size:512
[I] I2SConfigESP32.h : 111 - pin_mck: 0
[I] I2SConfigESP32.h : 113 - pin_bck: 27
[I] I2SConfigESP32.h : 115 - pin_ws: 25
[I] I2SConfigESP32.h : 117 - pin_data: 26
[W] AudioKit.h : 63 - sd_active = true
[   139][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=33 scl=32 freq=100000
[I] AudioTypes.h : 104 - sample_rate: 44100
[I] AudioTypes.h : 105 - channels: 2
[I] AudioTypes.h : 106 - bits_per_sample: 16
[I] I2SConfigESP32.h : 90 - rx/tx mode: TX_MODE
[I] I2SConfigESP32.h : 91 - port_no: 0
[I] I2SConfigESP32.h : 92 - is_master: Master
[I] I2SConfigESP32.h : 93 - sample rate: 44100
[I] I2SConfigESP32.h : 94 - bits per sample: 16
[I] I2SConfigESP32.h : 95 - number of channels: 2
[I] I2SConfigESP32.h : 96 - signal_type: Digital
[I] I2SConfigESP32.h : 98 - i2s_format: I2S_STD_FORMAT
[I] I2SConfigESP32.h : 100 - auto_clear: true
[I] I2SConfigESP32.h : 102 - use_apll: true
[I] I2SConfigESP32.h : 107 - buffer_count:6
[I] I2SConfigESP32.h : 108 - buffer_size:512
[I] I2SConfigESP32.h : 111 - pin_mck: 0
[I] I2SConfigESP32.h : 113 - pin_bck: 27
[I] I2SConfigESP32.h : 115 - pin_ws: 25
[I] I2SConfigESP32.h : 117 - pin_data: 26
[I] AudioKit.h : 605 - void audio_tools::AudioKitStream::setupActions()
[W] AudioKit.h : 612 - Mode Button ignored because of conflict: 13 
[I] AudioActions.h : 42 - ActionLogic::add pin: 39 / logic: 2
[I] AudioActions.h : 55 - pin 39 -> INPUT
[I] AudioKit.h : 355 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 42 - ActionLogic::add pin: 5 / logic: 0
[I] AudioActions.h : 52 - pin 5 -> INPUT_PULLUP
[I] AudioKit.h : 355 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 42 - ActionLogic::add pin: 18 / logic: 0
[I] AudioActions.h : 52 - pin 18 -> INPUT_PULLUP
[I] AudioKit.h : 260 - virtual void audio_tools::AudioKitStream::setAudioInfo(audio_tools::AudioInfo)
[I] URLStream.h : 85 - virtual bool audio_tools::URLStream::begin(const char*, const char*, MethodID, const char*, const char*): http://peacefulpiano.stream.publicradio.org/peacefulpiano.aac
[I] Url.h : 74 - Url::parse
[I] Url.h : 110 - url->http://peacefulpiano.stream.publicradio.org/peacefulpiano.aac
[I] Url.h : 111 - host->peacefulpiano.stream.publicradio.org
[I] Url.h : 112 - protocol->http
[I] Url.h : 113 - path->/peacefulpiano.aac
[I] Url.h : 114 - port->80
[I] URLStream.h : 379 - void audio_tools::URLStream::login()
[   367][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   460][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[   464][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 2 - STA_START
[   474][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
..[  1104][V][WiFiGeneric.cpp:355] _arduino_event_cb(): STA Connected: SSID: 3WO8kiazq8Ok17E, BSSID: 2c:79:d7:67:b1:4e, Channel: 6, Auth: WPA2_PSK
[  1106][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  1259][V][WiFiGeneric.cpp:369] _arduino_event_cb(): STA Got New IP:192.168.1.21
[  1259][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  1262][D][WiFiGeneric.cpp:1098] _eventCallback(): STA IP: 192.168.1.21, MASK: 255.255.255.0, GW: 192.168.1.1

[I] URLStream.h : 353 - WiFiClient
[I] HttpRequest.h : 231 - process connecting to host peacefulpiano.stream.publicradio.org port 80
[I] HttpRequest.h : 329 - connected 1 timeout 60000
[I] HttpRequest.h : 242 - Free heap: 98592
[I] HttpHeader.h : 276 - HttpHeader::write
[I] HttpHeader.h : 425 - -> GET /peacefulpiano.aac HTTP/1.1
[I] HttpHeader.h : 199 -  -> Host: peacefulpiano.stream.publicradio.org 
[I] HttpHeader.h : 199 -  -> Connection: keep-alive 
[I] HttpHeader.h : 199 -  -> Accept-Encoding: identity 
[I] HttpHeader.h : 199 -  -> Accept: audio/aac 
[I] HttpHeader.h : 341 -  -> <CR LF> 
[I] HttpRequest.h : 283 - Request written ... waiting for reply
[I] HttpHeader.h : 245 - Waiting for data...
[I] HttpHeader.h : 254 - Data available: 757
[I] HttpRequest.h : 173 - no CONTENT_LENGTH found in reply
[I] URLStream.h : 110 - size: 0
[I] URLStream.h : 242 - Request written ... waiting for reply
[W] CodecAACHelix.h : 154 - sample_rate: 44100
[I] AudioKit.h : 260 - virtual void audio_tools::AudioKitStream::setAudioInfo(audio_tools::AudioInfo)
[I] StreamCopy.h : 143 - StreamCopy::copy  1024 -> 1024 -> 1024 bytes - in 1 hops

assert failed: spinlock_acquire spinlock.h:122 (result == core_id || result == SPINLOCK_FREE)

Backtrace: 0x4008384d:0x3ffcdea0 0x4008d7b5:0x3ffcdec0 0x40092e21:0x3ffcdee0 0x40090acf:0x3ffce010 0x4008e2cd:0x3ffce050 0x40114e75:0x3ffce090 0x40103f9e:0x3ffce0c0 0x40115fcd:0x3ffce0f0 0x401161fe:0x3ffce110 0x40101cf7:0x3ffce160 0x401033f1:0x3ffce1a0 0x401034c6:0x3ffce200 0x400e4d39:0x3ffce220 0x400e4e85:0x3ffce250 0x400e4ed4:0x3ffce270 0x400d61fd:0x3ffce2b0 0x400d6228:0x3ffce2d0 0x400d81b4:0x3ffce2f0 0x400d831a:0x3ffce330 0x400ec195:0x3ffce350
```

### Device Description

I'm using a ESP32 Audio Kit V2.2 (2762).

### Sketch

```cpp
see (https://github.com/LieBtrau/esp32-web-radio/tree/main/firmware/esp32/platformio/esp32a1s-1stTest)
```

### What is your development environment

I'm building with platformio.  My code is [here](https://github.com/LieBtrau/esp32-web-radio/tree/main/firmware/esp32/platformio/esp32a1s-1stTest).

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2023-11-26:
I am pretty sure that I did not do any changes which would cause this and I am pretty sure that if you decode the stacktrace, you will see that the source of your trouble is not with my code.

I also tested with the latest ESP32 2.14 kernal version and could not reproduce your error with [my example](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/streams-url_aac-audiokit/streams-url_aac-audiokit.ino).

I guess you must be picking some buggy kernal

LieBtrau, 2023-11-26:
Yes, you're right about the ESP32-kernel.  The latest version of your libraries work using an older 5.2.0 version of the ESP32 platform.
```
[env:esp32dev]
platform = espressif32 @ 5.2.0
board = esp32dev
framework = arduino
lib_deps = 
    https://github.com/pschatzmann/arduino-audiokit
    https://github.com/pschatzmann/arduino-audio-tools
    https://github.com/pschatzmann/arduino-libhelix
lib_ldf_mode = deep+
build_flags = -DCORE_DEBUG_LEVEL=5 -DAUDIOKIT_BOARD=5 ; uses 5, not 6 as mentioned in the wiki
monitor_speed = 115200
```
I'll have a further look into it.