pschatzmann/arduino-audio-tools #1076: SynchronizedBuffers available method is not thread safe

lligallo, 2023-11-27:
### Problem Description

Hi Phil, 
First congrats on your project, it is awesome.

I run into a problem caused by [SyncrhonizeBuffer](https://github.com/pschatzmann/arduino-audio-tools/blob/main/src/AudioTools/SynchronizedBuffers.h). I'm using a RingBuffer to communicate with a server. One thread is receiving data and writing it in the RingBuffer (via the Synchronized wrapper) and another is calling copy with StreamCopy. 

The available() method in Synchronized is not thread-safe (the only one not using LockGuard). So when StreamCopy::copy gets the available bytes, while another thread (on the other core) is pushing data through the writeArray (threadSafe) then the number of bytes is "non-correct". This is especially noticeable when the data you are pushing is in some kind of structure (for example an MP3 chunk with headers) since then StreamCopy::copy will call AudioStream::readBytes with the length that is only a piece of the packet structure.

I would say a LockGuard should be added.

Again, great work!

### Device Description

ESP32-S3, using Platformio, nothing fancy.

### Sketch

```cpp
Description.
```

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2023-11-27:
If you use the ESP32, I recommend that you use the SynchronizedBufferRTOS. 
This is giving by far the best performance and it is really save.

So far, I have left it out of the available and availableForWrite because I thought it would not matter too much....

pschatzmann, 2023-11-27:
I have committed the correction

lligallo, 2023-11-27:
Thanks Phil for the correction and the recommendation.

closed 2023-11-27 by lligallo