pschatzmann/arduino-audio-tools #1080: ESP32 high frequency PWM overflows int16_t

akumpf, 2023-11-30:
### Problem Description

Quick bug fix. For ESP32, PWM frequency can be greater than 65kHz, and even though this variable is only set indirectly, because `pwm_frequency` is of type `int16_t` in `PWMAudioBase.h`, it overflows and produces the wrong frequency. Changing it to type `int32_t` fixed it for me. :)

### Device Description

Tested on ESP32S3

### Sketch

```cpp
See https://github.com/pschatzmann/arduino-audio-tools/blob/bdc3713fd30878b196e381d1b462c35ecd77b768/src/AudioPWM/PWMAudioBase.h#L46
```

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

akumpf, 2023-11-30:
I'll look into it. It compiles fine and doesn't report an error with the `int32_t`, but looks like the PWM doesn't actually output for me. It's a minor edge case, but hopefully i can resolve it fully, and then follow up later with a pull request.

pschatzmann, 2023-12-11:
Thanks for [your pull request](https://github.com/pschatzmann/arduino-audio-tools/pull/1081)

closed 2023-12-11 by pschatzmann