pschatzmann/arduino-audio-tools #1097: Memory leak in HttpHeader.h

pschatzmann, 2023-12-11:
### Discussed in https://github.com/pschatzmann/arduino-audio-tools/discussions/1088

<div type='discussions-op-text'>

<sup>Originally posted by **pete8804** December  4, 2023</sup>
After updating my web radio project to the most recent version of arduino_audio_tools, 
I noticed that my project was consuming ca. 4k of memory on each restart of an URL stream. 
After some investigation I was able to identify the cause:
On 2023-09-22, method **clear() in HttpRequest.h** was changed as follows:
 ```
 /// clears the data - usually we do not delete but we just set the active flag
        HttpHeader& clear(bool activeFlag=true) {
            is_written = false;
            is_chunked = false;
            url_path = "/";
            for (auto it = lines.begin() ; it != lines.end(); ++it){
                if (activeFlag){
                    (*it)->active = false;
                } else {
                    (*it)->active = false;  <- added 2023-09-22
                    //delete *it;  // this leads to exceptions <- commented out 2023-09-22
                }
            }
            if (!activeFlag){
                lines.clear(); // BUG: memory leak if delete *it is commented out
            }
            return *this;
```            
When _lines.clear()_ is executed without preceding _delete *it,_ the existing header lines become inaccessible, but memory is not released. Then, on the next start of an URL stream, new header lines are created instead of re-activating the existing ones.

I fixed the problem locally by commenting out the _lines.clear()_ statement, and successfully retested it.
I hope you can confirm my observation and implement the fix.

By the way: arduino_audio_tools is a great piece of work. I really appreciate it!</div>

pschatzmann, 2023-12-11:
https://github.com/pschatzmann/arduino-audio-tools/pull/1098

closed 2023-12-11 by pschatzmann

svenbrill, 2023-12-11:
@pschatzmann 
I cannot compile anymore. You changed the signature of `waitForData()` to `waitForData(int timeout)` in `URLStream.h`.
The old signature is used in `HLSStream.h`. I got this error message:

```
In file included from .pio/libdeps/heltec_wifi_lora_32_V3/audio-tools/src/AudioHttp/AudioHttp.h:13,
                 from .pio/libdeps/heltec_wifi_lora_32_V3/audio-tools/src/AudioTools.h:91,
                 from src/audio.cpp:1:
.pio/libdeps/heltec_wifi_lora_32_V3/audio-tools/src/AudioHttp/HLSStream.h: In member function 'void audio_tools::URLLoader::bufferRefill()':
.pio/libdeps/heltec_wifi_lora_32_V3/audio-tools/src/AudioHttp/HLSStream.h:126:29: error: no matching function for call to 'audio_tools::URLStream::waitForData()'
       p_stream->waitForData();
                             ^
In file included from .pio/libdeps/heltec_wifi_lora_32_V3/audio-tools/src/AudioHttp/AudioHttp.h:8,
                 from .pio/libdeps/heltec_wifi_lora_32_V3/audio-tools/src/AudioTools.h:91,
                 from src/audio.cpp:1:
.pio/libdeps/heltec_wifi_lora_32_V3/audio-tools/src/AudioHttp/URLStream.h:240:22: note: candidate: 'virtual bool audio_tools::URLStream::waitForData(int)'
         virtual bool waitForData(int timeout) {
                      ^~~~~~~~~~~
.pio/libdeps/heltec_wifi_lora_32_V3/audio-tools/src/AudioHttp/URLStream.h:240:22: note:   candidate expects 1 argument, 0 provided
```

pschatzmann, 2023-12-12:
OK, I committed the remaining open changes as well to prevent any inconsistencies...