pschatzmann/arduino-audio-tools #1104: HTML Not found

Ephesian, 2023-12-19:
### Problem Description

Helo
I dont think this works, pops an error in browser, and I dont see html folder in the source.

https://pschatzmann.github.io/arduino-audio-tools/html/modules.html

### Device Description

General

### Sketch

```cpp
Browser HTML
```

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2023-12-19:
I don't quite understand: where did you get this link from ?
All the documented links in the Readme of this project are working w/o any issues...

Ephesian, 2023-12-20:
Hello 
in the doc folder->ext->index.html  .  Hope this make sense

pschatzmann, 2023-12-22:
Just use the online documentation.
If you want to use it locally, you need to generate it with doxygen yourself

closed 2023-12-23 by pschatzmann