pschatzmann/arduino-audio-tools #1129: compilation error in BluetoothA2DPSource.cpp:968 when Logging is enabled

cheops, 2023-12-29:
### Problem Description

ESP32-A2DP\src\BluetoothA2DPSource.cpp:968
compilation error, because `rc->psth_rsp.rsp_code` is not defined in `avrc_ct_psth_rsp_param`

corrected code:
```
  /* when passthrough responsed, this event comes */
  case ESP_AVRC_CT_PASSTHROUGH_RSP_EVT: {
    ESP_LOGI(
        BT_RC_CT_TAG,
        "AVRC passthrough response: key_code 0x%x, key_state %d",
        rc->psth_rsp.key_code, rc->psth_rsp.key_state);
    break;
  }
```

logging enabled in `platformio.ini`
```
build_flags = 
	-DCORE_DEBUG_LEVEL=5
```

### Device Description

`platformio.ini`
```
platform = espressif32
board = featheresp32
framework = arduino

lib_deps = 
	https://github.com/pschatzmann/ESP32-A2DP
	adafruit/Adafruit VS1053 Library@^1.4.0
```

### Sketch

```cpp
just try to compile with debugging enabled
```

### What is your development environment

PlatformIO
with platform espressif32
board featheresp32
framework arduino

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2023-12-29:
I can't reproduce your issue: Both the latest Arduino and Platformio versions are compiling w/o problems.
The latest Arduino core version is 2.0.14 and the one that comes with Platformio is quite older: 2.0.11.

If you want to use an even older version, please correct the version dependent error yourself and submit a pull request.

closed 2023-12-29 by pschatzmann

cheops, 2023-12-29:
You are correct, I had an older version, after upgrading everything is fine.
Thanks for the quick response!