pschatzmann/arduino-audio-tools #1177: I2S problem causing constant reboots

mooballs, 2024-01-10:
### Problem Description

Previously working code that I have come back to after 6 months or so now just reboots. Have tried different boards, cables, flash modes, esp core versions (2.0.11, 2.0.14, 3.alpha). Used the latest version of Audio Tools, plus tried some older ones.

Please Help!

### Device Description

ESP32 Dev Module

### Sketch

```cpp
#include <AudioTools.h>

#include "AudioLibs/AudioA2DP.h"
#include "AudioConfigLocal.h"
#include "BluetoothA2DPSink.h"

BluetoothA2DPSink a2dp_sink;
I2SStream i2s;
//A2DPStream in;
Equilizer3Bands eq(i2s);
//StreamCopy copier(eq, i2s); // copy in to equilizer
ConfigEquilizer3Bands cfg_eq;

// Potentiometer is connected to GPIO 35
const int potPin = 35;
// variable for storing the potentiometer value
int potValue = 0;

void read_data_stream(const uint8_t *data, uint32_t length) {
  eq.write(data, length);
}

// Arduino Setup
void setup(void) {

  Serial.begin(115200);

  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  //i2s_out esp32 external DAC
  auto config = i2s.defaultConfig(TX_MODE);
  config.pin_bck = 26;
  config.pin_ws = 25;
  config.pin_data = 23;
  config.sample_rate = a2dp_sink.sample_rate();
  config.channels = 2;
  config.bits_per_sample = 16;
  i2s.begin(config);

  cfg_eq = eq.defaultConfig();
  cfg_eq.setAudioInfo(config); // use channels, bits_per_sample and sample_rate from kit

  cfg_eq.gain_medium = 1.0;
  cfg_eq.gain_high = 1.0;
  eq.begin(cfg_eq);

  // register callback
  //a2dp_sink.set_pin_config(my_pin_config);
  a2dp_sink.set_stream_reader(read_data_stream, false);

  // Start Bluetooth Audio Receiver
  a2dp_sink.set_auto_reconnect(true);
  a2dp_sink.start("Poo");  //bluetooth name

}

// Arduino loop
void loop() {
  //copier.copy();
  //delay(100);
  potValue = float(analogRead(potPin)) / 4095;
  Serial.println(potValue);
  cfg_eq.gain_low = potValue;
  delay(500);
}
```

### Other Steps to Reproduce

**_From serial Monitor:_**

ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
[I] AudioTypes.h : 104 - sample_rate: 0
[I] AudioTypes.h : 105 - channels: 2
[I] AudioTypes.h : 106 - bits_per_sample: 16
[I] I2SConfigESP32.h : 90 - rx/tx mode: TX_MODE
[I] I2SConfigESP32.h : 91 - port_no: 0
[I] I2SConfigESP32.h : 92 - is_master: Master
[I] I2SConfigESP32.h : 93 - sample rate: 0
[I] I2SConfigESP32.h : 94 - bits per sample: 16
[I] I2SConfigESP32.h : 95 - number of channels: 2
[I] I2SConfigESP32.h : 96 - signal_type: Digital
[I] I2SConfigESP32.h : 98 - i2s_format: I2S_STD_FORMAT
[I] I2SConfigESP32.h : 100 - auto_clear: true
[I] I2SConfigESP32.h : 102 - use_apll: true
[I] I2SConfigESP32.h : 107 - buffer_count:6
[I] I2SConfigESP32.h : 108 - buffer_size:512
[I] I2SConfigESP32.h : 113 - pin_bck: 26
[I] I2SConfigESP32.h : 115 - pin_ws: 25
[I] I2SConfigESP32.h : 117 - pin_data: 23
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d8974  PS      : 0x00060030  A0      : 0x800d9674  A1      : 0x3ffcf180  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffcae40  A5      : 0x00000000  
A6      : 0x3ffcae40  A7      : 0x3ffc3d70  A8      : 0x00000100  A9      : 0x00000002  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x3ffc3d70  
A14     : 0xffffffbf  A15     : 0x00000008  SAR     : 0x0000001a  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400914d8  LEND    : 0x400914e3  LCOUNT  : 0x00000000  

Backtrace: 0x400d8971:0x3ffcf180 0x400d9671:0x3ffcf1d0 0x400d3626:0x3ffcf210 0x400d37f1:0x3ffcf290 0x400d38ed:0x3ffcf300 0x400d437c:0x3ffcf370 0x400d6256:0x3ffcf450

ELF file SHA256: 34414cc9cb465256

**_Which I fed into exception decoder_**

EXCVADDR: 0x00000000

Decoding stack results
0x400d8971: i2s_set_clk at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/driver/i2s.c line 1171
0x400d9671: i2s_driver_install at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/driver/i2s.c line 2027
0x400d3626: audio_tools::I2SDriverESP32::begin(audio_tools::I2SConfigESP32, int, int) at C:\Users\marky\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioI2S/I2SESP32.h line 185
0x400d37f1: audio_tools::I2SDriverESP32::begin(audio_tools::I2SConfigESP32) at C:\Users\marky\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioI2S/I2SESP32.h line 54
0x400d38ed: audio_tools::I2SStream::begin(audio_tools::I2SConfigESP32) at C:\Users\marky\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioI2S/I2SStream.h line 61
0x400d437c: setup() at C:\Users\marky\OneDrive\Documents\Arduino\WorkingEQ/WorkingEQ.ino line 43
0x400d6a5a: uartBegin at C:\Users\marky\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32\esp32-hal-uart.c line 213

### What is your development environment

Arduino 2..2.1 and 1.8

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-01-10:
As you can see, the issue comes from the missing sample rate: either assign 44100 or open a2dp first, in order to mske it available

closed 2024-01-10 by pschatzmann

mooballs, 2024-01-11:
I wondered about that but just panicked as it had worked previously, many thanks.