pschatzmann/arduino-audio-tools #1179: Build errors - platformio, esp32 s3

olup, 2024-01-11:
### Problem Description

While trying to build, having a certain number of errors:

```
In file included from .pio/libdeps/esp32-s3-devkitc-1/audio-tools/src/AudioLibs/AudioSourceSDMMC.h:5,
                 from src/player/player.cpp:8:
.pio/libdeps/esp32-s3-devkitc-1/audio-tools/src/AudioTools/AudioSource.h:50:48: error: expected ',' or '...' before '(' token
     virtual bool setMetadataCallback(void (*fn)(MetaDataType info, const char* str, int len), ID3TypeSelection sel=SELECT_ICY) {
                                                ^
In file included from .pio/libdeps/esp32-s3-devkitc-1/audio-tools/src/AudioLibs/AudioSourceSDMMC.h:5,
                 from src/player/player.cpp:8:
.pio/libdeps/esp32-s3-devkitc-1/audio-tools/src/AudioTools/AudioSource.h:159:37: error: expected ')' before '&' token
     AudioSourceURL(AbstractURLStream& urlStream, T(&urlArray)[N], const char* mime, int startPos = 0) {
                   ~                 ^
                                     )
.pio/libdeps/esp32-s3-devkitc-1/audio-tools/src/AudioTools/AudioSource.h:241:40: error: expected ',' or '...' before '(' token
     bool setMetadataCallback(void (*fn)(MetaDataType info, const char* str, int len), ID3TypeSelection sel=SELECT_ICY) {
                                        ^
.pio/libdeps/esp32-s3-devkitc-1/audio-tools/src/AudioTools/AudioSource.h:248:5: error: 'AbstractURLStream' does not name a type
     AbstractURLStream* actual_stream = nullptr;
     ^~~~~~~~~~~~~~~~~
.pio/libdeps/esp32-s3-devkitc-1/audio-tools/src/AudioTools/AudioSource.h: In member function 'virtual Stream* audio_tools::AudioSourceURL::selectStream(int)':
.pio/libdeps/esp32-s3-devkitc-1/audio-tools/src/AudioTools/AudioSource.h:187:22: error: 'actual_stream' was not declared in this scope
         if (started) actual_stream->end();
                      ^~~~~~~~~~~~~
.pio/libdeps/esp32-s3-devkitc-1/audio-tools/src/AudioTools/AudioSource.h:187:22: note: suggested alternative: 'NullStream'
         if (started) actual_stream->end();
                      ^~~~~~~~~~~~~
                      NullStream
.pio/libdeps/esp32-s3-devkitc-1/audio-tools/src/AudioTools/AudioSource.h:188:9: error: 'actual_stream' was not declared in this scope
         actual_stream->begin(urlArray[pos], mime);
         ^~~~~~~~~~~~~
.pio/libdeps/esp32-s3-devkitc-1/audio-tools/src/AudioTools/AudioSource.h:188:9: note: suggested alternative: 'NullStream'
         actual_stream->begin(urlArray[pos], mime);
         ^~~~~~~~~~~~~
         NullStream
.pio/libdeps/esp32-s3-devkitc-1/audio-tools/src/AudioTools/AudioSource.h: In member function 'virtual Stream* audio_tools::AudioSourceURL::selectStream(const char*)':
.pio/libdeps/esp32-s3-devkitc-1/audio-tools/src/AudioTools/AudioSource.h:216:22: error: 'actual_stream' was not declared in this scope
         if (started) actual_stream->end();
                      ^~~~~~~~~~~~~
.pio/libdeps/esp32-s3-devkitc-1/audio-tools/src/AudioTools/AudioSource.h:216:22: note: suggested alternative: 'NullStream'
         if (started) actual_stream->end();
                      ^~~~~~~~~~~~~
                      NullStream
.pio/libdeps/esp32-s3-devkitc-1/audio-tools/src/AudioTools/AudioSource.h:217:9: error: 'actual_stream' was not declared in this scope
         actual_stream->begin(path, mime);
         ^~~~~~~~~~~~~
.pio/libdeps/esp32-s3-devkitc-1/audio-tools/src/AudioTools/AudioSource.h:217:9: note: suggested alternative: 'NullStream'
         actual_stream->begin(path, mime);
         ^~~~~~~~~~~~~
         NullStream
.pio/libdeps/esp32-s3-devkitc-1/audio-tools/src/AudioTools/AudioSource.h: In member function 'virtual void audio_tools::AudioSourceURL::setTimeout(int)':
.pio/libdeps/esp32-s3-devkitc-1/audio-tools/src/AudioTools/AudioSource.h:232:9: error: 'actual_stream' was not declared in this scope
         actual_stream->setTimeout(millisec);
         ^~~~~~~~~~~~~
.pio/libdeps/esp32-s3-devkitc-1/audio-tools/src/AudioTools/AudioSource.h:232:9: note: suggested alternative: 'NullStream'
         actual_stream->setTimeout(millisec);
         ^~~~~~~~~~~~~
         NullStream
.pio/libdeps/esp32-s3-devkitc-1/audio-tools/src/AudioTools/AudioSource.h: In member function 'virtual bool audio_tools::AudioSourceURL::setMetadataCallback(void*)':
.pio/libdeps/esp32-s3-devkitc-1/audio-tools/src/AudioTools/AudioSource.h:243:16: error: 'actual_stream' was not declared in this scope
         return actual_stream->setMetadataCallback(fn);
                ^~~~~~~~~~~~~
.pio/libdeps/esp32-s3-devkitc-1/audio-tools/src/AudioTools/AudioSource.h:243:16: note: suggested alternative: 'NullStream'
         return actual_stream->setMetadataCallback(fn);
                ^~~~~~~~~~~~~
                NullStream
Compiling .pio/build/esp32-s3-devkitc-1/lib182/WiFi/WiFi.cpp.o
Compiling .pio/build/esp32-s3-devkitc-1/lib182/WiFi/WiFiAP.cpp.o
In file included from .pio/libdeps/esp32-s3-devkitc-1/audio-tools/src/AudioTools.h:93,
                 from src/player/player.cpp:9:
.pio/libdeps/esp32-s3-devkitc-1/audio-tools/src/AudioTools/AudioPlayer.h: In member function 'virtual void audio_tools::AudioPlayer::setMetadataCallback(void (*)(audio_tools::MetaDataType, const char*, int), audio_tools::ID3TypeSelection)':
.pio/libdeps/esp32-s3-devkitc-1/audio-tools/src/AudioTools/AudioPlayer.h:399:47: error: invalid conversion from 'void (*)(audio_tools::MetaDataType, const char*, int)' to 'void*' [-fpermissive]
```

### Device Description

ESP32S3 and MAX98357A

### Sketch

```cpp
PIO config (`ttps://github.com/schreibfaul1/ESP32-audioI2S.git` is still in the deps - I want to compare sound output between libs)

[env]
platform_packages = 
	toolchain-riscv32-esp @ 8.4.0+2021r2-patch5

[env:esp32-s3-devkitc-1]
platform = espressif32@6.4.0
board = esp32-s3-devkitc-1
framework = arduino
lib_ldf_mode = deep+
lib_deps = 
	bblanchon/ArduinoJson@^6.21.3
	https://github.com/schreibfaul1/ESP32-audioI2S.git
	bodmer/TFT_eSPI@^2.5.33
	lennarthennigs/Button2@^2.2.4
	ESP Async WebServer
	bodmer/JPEGDecoder@^2.0.0
	esp32-targz@1.2.2
	https://github.com/pschatzmann/arduino-audio-tools
	https://github.com/pschatzmann/arduino-libhelix
monitor_speed = 115200
monitor_rts = 0
monitor_dtr = 0
build_flags = 
	-D SOC_SDMMC_HOST_SUPPORTED
	-D ARDUINO_USB_MODE=1
	-D ARDUINO_USB_CDC_ON_BOOT=1
	-D BOARD_HAS_PSRAM
	-Wno-unused-variable -Wno-unused-but-set-variable -Wno-unused-function -Wno-format-extra-args
	
	-D USER_SETUP_LOADED=1
	
	-D ILI9341_2_DRIVER=1
	-D TFT_WIDTH=240
	-D TFT_HEIGHT=320
	-D TFT_MOSI=35
	-D TFT_SCLK=37
	-D TFT_DC=36
	-D TFT_CS=38
	-D AUDIO_LOG=true
	
	-D SPI_FREQUENCY=27000000
```
```

### What is your development environment

vscode platformio, using arduino framework.

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

olup, 2024-01-11:
Same using `platform = https://github.com/platformio/platform-espressif32.git`

pschatzmann, 2024-01-11:
You are funny: How do you think I will be able to reproduce this ?

The relevant documentation is
- https://github.com/pschatzmann/arduino-audio-tools/wiki
- https://github.com/pschatzmann/arduino-audio-tools/wiki/Working-with-PlatformIO

I tried to compile the sdmmc player sketch with the platformio.ini from the documentation just changing the board and could not reproduce any issues
```
Processing esp32dev (platform: https://github.com/platformio/platform-espressif32.git; board: esp32-s3-devkitc-1; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (6.5.0+sha.ee1dd5a) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- audio-tools @ 0.9.6+sha.2375de1
|-- libhelix @ 0.1.0+sha.cf81f03
Building in release mode
Compiling .pio/build/esp32dev/src/main.cpp.o
Building .pio/build/esp32dev/bootloader.bin
Generating partitions .pio/build/esp32dev/partitions.bin
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/aacdec.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/aactabs.c.o
esptool.py v4.5.1
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/bitstream.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/buffers.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/dct4.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/decelmnt.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/dequant.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/fft.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/filefmt.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/huffman.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/hufftabs.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/imdct.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/noiseless.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/pns.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/sbr.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/sbrfft.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/sbrfreq.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/sbrhfadj.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/sbrhfgen.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/sbrhuff.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/sbrimdct.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/sbrmath.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/sbrqmf.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/sbrside.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/sbrtabs.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/stproc.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/tns.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-aac/trigtabs.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-mp3/bitstream.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-mp3/buffers.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-mp3/dct32.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-mp3/dequant.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-mp3/dqchan.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-mp3/huffman.c.o
Compiling .pio/build/esp32dev/libe8d/libhelix/libhelix-mp3/hufftabs.c.o
[... 82 lines cut ...]
      ^~~~~~~~~~~
Compiling .pio/build/esp32dev/FrameworkArduino/libb64/cdecode.c.o
Compiling .pio/build/esp32dev/FrameworkArduino/libb64/cencode.c.o
Compiling .pio/build/esp32dev/FrameworkArduino/main.cpp.o
Compiling .pio/build/esp32dev/FrameworkArduino/stdlib_noniso.c.o
Compiling .pio/build/esp32dev/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/esp32dev/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/esp32dev/libFrameworkArduino.a
Indexing .pio/build/esp32dev/libFrameworkArduino.a
Linking .pio/build/esp32dev/firmware.elf
Retrieving maximum program size .pio/build/esp32dev/firmware.elf
Checking size .pio/build/esp32dev/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   6.9% (used 22756 bytes from 327680 bytes)
Flash: [=         ]  13.7% (used 458065 bytes from 3342336 bytes)
Building .pio/build/esp32dev/firmware.bin
esptool.py v4.5.1
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
================================================ [SUCCESS] Took 60.85 seconds
```
You must be doing something seriously wrong when your environment does not support any function pointers!

If you want to compare any APIs, I suggest to also include the one that comes with the ESP32 Arduino. But if you expect to see any quality differences, I will have to disappoint you: all are using the same [Espressif I2S API](https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32/api-reference/peripherals/i2s.html)

closed 2024-01-11 by pschatzmann

olup, 2024-01-12:
Hey, thanks, I guess ?
I thought I'd had linked the repo's branch : https://github.com/olup/grigri/tree/new-sound

You are probably right, but I know esp32-audio-i2s manipulates the sound wave as it offers filters - and someone in a github issue said your lib sounds better. Might be total BS, but I get saturated sound over 50% software volume and wanted to compare. Probably won't do anything, and I'll have to understand deeper how sound works in the context of the max98357 DAC and amp.

Also, I get troubles with the seek feature of this other lib - yours does not implement it but let the user in control - I can then build my own working seek feature.