pschatzmann/arduino-audio-tools #1230: I2SStream does not name a type.

poetaster, 2024-01-20:
### Problem Description

The compiler is not including / seeing I2SStream ? 

### Device Description

Arduino Nano, PT8211, VS1053 (etc).

### Sketch

```cpp
streams-generator-i2s example sketch.
```

### Other Steps to Reproduce

I also read through the AudioConfig.h file for hints and have the feeling you just don't support 

### What is your development environment

Arduino 1.8.12

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-01-21:
The Arduino Nano is not rally suited for audio processing
- it is too slow
- has not enough RAM and Progmem
- does not support I2S

Your only chance is to use it together with VS1053, so that the only thing it needs to do is to copy data..

closed 2024-01-21 by pschatzmann

poetaster, 2024-01-21:
Well, I use it all the time: https://github.com/poetaster/rampart   but it certainly isn't ideal. I'm doing I2S via wire/spi  using VS1003/VS1053 in DAC mode, which works quite well with the nano. But I wanted to try your library. I'm assuming it's not compiling because it detects 'no I2S'?  

For high end stuff I use Ksoloti ( rebirth of Axoloti). There, I often find that I've underutilized which is expensive (I have two axolotis and two Ksolotis). https://github.com/ksoloti/ksoloti , perhaps it's of interest to you.

Which board ( < 20 Euros) would you recomend? Aside from the ones in the wiki? Thanks for your feedback!

poetaster, 2024-01-21:
Oh, just found a NodeMCU ESP8266 in a box. Is that appropriate for testing your library?

pschatzmann, 2024-01-21:
Not sure about the limitations: I did only some minimum testing with the ESP8266...

You can't use any examples which use classes that are not available for your processor. If you have an I2SStream using a nano, you will get compile errors, because the class does not exist but a PWMStream should work if you keep the sample rate low enough.

The VS1003 examples can be found [here](https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-vs1053), but they will not work with the Nano because they will not fit into the memory. But they should work e.g. with a Uno R4.

I am usually working with an ESP32 based [AudioKit](https://github.com/pschatzmann/arduino-audiokit). So any ESP32 based solution provides the maximum of functionality. A RP2040 should work for most of the cases as well....

poetaster, 2024-01-22:
Thanks very much for the responses. The ESP8266 doesn't have enough ADC, so it's out. I found RP2040s are very affordable, so that may be my goto for 'simple things'. 

I just noticed that I can get the VS1003 (not the 1053) into dac mode, but can not load mozzi, so, as you suggest, memory is too constrained. I noticed that your  VS1053Stream.h class does not seem to rely on the dac mode user program. I think I need to read up on the 1053 some more. 

Thanks again for your efforts.

pschatzmann, 2024-01-22:
The VS1053Stream just automatically adds a wav header when you send pcm data. 
This was quite easy to implement with my available functionality...

mkoenings, 2024-01-22:
we are also testing the Pico, for Audio Input and output ( connected wirelessly, with nrf24). For the audio input, we are trying the WM8960 ( Just because I had one available) but I can't make it work, the I2S only returns random garbage values. 
If somebody wants to try:
The I2C lines need pullup resistors ( and maybe also short wires as Phil mentioned in his blog post). It worked for me with 1.5k.
Which ADC chip would you recommend instead? We need line-in, 32khz, 16bit Mono. I ordered the [es8388](https://pcbartists.com/product/es8388-module/) boards from India, because we had good results with the LyraT boards, that carry this chip. Any other recommendations ? I have a V1053, and also a teensy audio Board. The Teensy board (SGTL5000) works fine ( on a Teensy) but I dont know how to use it (witout the teensy audio lib) in this szenario: a Pico with my favorite audio library.. ;-) . 
Thanks.

poetaster, 2024-01-23:
> The VS1053Stream just automatically adds a wav header when you send pcm data. This was quite easy to implement with my available functionality.

Ah, duh! Why didn't I think of that :) The chip tends to be in mp3 decoder mode by default but supports a bunch of formats. Sigh. Thanks!

> The ESP32 supports up to 18 measurement channels which should be more then enough! 

Ok, I'll have to read the spec again, but the nodemcu I have only has one analog channel. I usually need at least four (potentiometers). It could work for other projects. Thanks again!