pschatzmann/arduino-audio-tools #1231: base-adc-serial example crash on ESP32-S3

romain145, 2024-01-22:
### Problem Description

Default max_buffer_len is 1024 in the example. Increasing it to 4096 causes a crash and reboot of the ESP32-S3:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce3818,len:0x438
load:0x403c9700,len:0x4
load:0x403c9704,len:0x9ac
load:0x403cc700,len:0x2758
entry 0x403c9868
[   136][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42006aa8
[   147][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42006a74
[   158][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42006a40
[   170][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42006a0c
[   181][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42006aa8
[   192][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42006a74
[   204][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42006a40
[   215][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42006a0c
[   226][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42006aa8
[   238][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42006a74
[   249][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42006a40
[   260][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42006a0c
[   281][V][esp32-hal-periman.c:154] perimanSetPinBus(): Pin 44 successfully set to type UART_RX (2) with bus 0x3fc94000
[   292][V][esp32-hal-periman.c:154] perimanSetPinBus(): Pin 43 successfully set to type UART_TX (3) with bus 0x3fc94000
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-S3
  Package           : 0
  Revision          : 1
  Cores             : 2
  Frequency         : 240 MHz
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   393404 B ( 384.2 KB)
  Free Bytes        :   361756 B ( 353.3 KB)
  Allocated Bytes   :    26664 B (  26.0 KB)
  Minimum Free Bytes:   356568 B ( 348.2 KB)
  Largest Free Block:   327668 B ( 320.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         : 33554432 B (32 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : OPI
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  3264.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00340000, size:  3264.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00670000, size:  1536.0 KB, type: DATA, subtype: SPIFFS
[... 132 lines cut ...]
description = Audio Example
default_envs = esp32dev

[env:esp32dev]
platform = https://github.com/sgryphon/platform-espressif32.git#sgryphon/add-esp32-arduino-libs
platform_packages =
    platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#master
    platformio/framework-arduinoespressif32-libs @ https://github.com/espressif/esp32-arduino-libs.git#idf-release/v5.1
board = esp32-s3-devkitc-1
framework = arduino
board_build.arduino.memory_type = opi_opi
lib_deps = https://github.com/pschatzmann/arduino-audio-tools
build_flags = -DCORE_DEBUG_LEVEL=5 -Wno-unused-variable -Wno-unused-but-set-variable -Wno-unused-function -Wno-format-extra-args 
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
```

### What is your development environment

PlatformIO with Arduino ESP32 alpha v3.0.0 based on ESP-IDF v5.1 for ESP32-S3 support in platformIO.
see https://github.com/platformio/platform-espressif32/pull/1281

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-01-22:
Why would you want to use such a big max_buffer_len ?

romain145, 2024-01-22:
Good question. I'll stick with 1024 then. Still, this should throw an error instead of crashing?

Does the rest of the setup look OK? I'm trying to evaluate the performance of the ESP32-S3 ADC.
Thanks

pschatzmann, 2024-01-22:
For audio I would would change the following
  adcConfig.is_auto_center_read = __true__;

pschatzmann, 2024-01-23:
I can move the allocation for the temporary memory from the stack to the heap.
Another way to deal with this is just to increase the stack size by increasing the CONFIG_ARDUINO_LOOP_STACK_SIZE!

pschatzmann, 2024-09-10:
closed due to inactivity

closed 2024-09-10 by pschatzmann