pschatzmann/arduino-audio-tools #1274: AudioLibs/VS1053Stream.h produces no sound on the rp2040 pi pico (w/o wifi).

poetaster, 2024-01-31:
### Problem Description

streams-generator-vs1053 compiles and executes but produces no sound. Using the arduino-vs1053 driver alone this minimal sketch works: https://github.com/poetaster/VS1003/tree/master/examples/vs1053_wavcpm_sine_nomozzi_pico

### Device Description

RP2040 I'll try another device (RP2040 and and esp) as soon as I find some time.

### Sketch

```cpp
It's the example sketch without changes. I did, however set the AudioConfig.h:
// Default Pins for VS1053

#define VS1053_CS 17
#define VS1053_DCS 9
#define VS1053_DREQ 10
#define VS1053_RESET 11 
#define VS1053_CS_SD -1
```
```

### What is your development environment

Arduino 1.8.19, Linux (ubuntu 22). 

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-01-31:
Did you try with the [documented pins](https://github.com/pschatzmann/arduino-vs1053/wiki/Pinouts-for-Processors-and-Tested-Boards) ?

poetaster, 2024-01-31:
> Most likely you did not get all pin definitions right: Did you double check the [documented pins](https://github.com/pschatzmann/arduino-vs1053/wiki/Pinouts-for-Processors-and-Tested-Boards) ?

Yes, I did. But you also have other settings on conf files. In any  case. I did ONCE have sound (a clean sine) from that example sketch. The example I posted always works. It's crappy (crappy osc., insufficient buffering) but it works with your driver. Hence, the pin assignments must be correct since you driver works with it. Every time. 

```
#define VS1053_CS     17
#define VS1053_DCS    9
#define VS1053_DREQ   10
```

Are the currently funtional ones. Miso and Mosi I set to RP defaults, but that obviously isn't the problem. Otherwise my sketch with your driver wouldn't work.

> My initial tests were with an ESP32 and my last tests with an UNO R4. What information do you get in the log with log level Info ?

The logging, though verbose, is often misleading. I did get the  
      ```  VS1053_LOGW("VS1053 not properly installed!")```
message, at some point Added  some delay and:
```    data_mode_on();```

 Switched to my sketch (with only the VS1053Driver) and it just worked. 

My other tess with I2S and the audio-tools library, although they did work, were not very promising (1502, two different DACs, both which are clean with other projects). I'm going to put pins on another RP2040 just to rule out hardware, but PWM with Mozzi on the test chip are fine.

pschatzmann, 2024-02-01:
I can't reproduce your issue. For me this is working perfectly.
So please activate the logging on info level with

```C++
void begin() {
  Serial.begin(115200);
  while(!Serial);
  AudioLogger::instance().begin(Serial, AudioLogger::Info); // Info is causing a lot of noise  
  VS1053Logger.begin(Serial, VS1053Info);
```

closed 2024-02-01 by pschatzmann

poetaster, 2024-02-01:
> Can you double check if calling SPI.begin(); makes a difference ? I noticed that the library is currently missing this call and you have it in your working sketch...

I'll try this now.

pschatzmann, 2024-02-01:
Yes, it's the Arduino pin numbers which are different then the numbers printed on the board. E.g. 0 is D0

poetaster, 2024-02-01:
All rewired as per instructions (you really should qualify and state GP or PIN) and It works. Perhaps it's because of how you treat Mosi/Miso? My sketch also works with these pin settings, so, fine by me. Or Perhaps some file was mangled on a git pull.

Thanks for your efforts!