pschatzmann/arduino-audio-tools #1318: Keys / Actions don't get fired

robert1x2, 2024-02-10:
### Problem Description

Keys / Actions don't get fired

### Device Description

AI Thinker ESP32 Audio Kit V2.2 A252

### Sketch

```cpp
/**
 * @file player-sd-audiokit.ino
 * @brief see https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/player-sd-audiokit/README.md
 * Make sure that the pins are set to off, on, on, off, off
 * @author Phil Schatzmann
 * @copyright GPLv3
 */

#include "AudioTools.h"
#include "AudioLibs/AudioBoardStream.h"
#include "AudioLibs/AudioSourceSD.h" // or AudioSourceIdxSD.h
#include "AudioCodecs/CodecMP3Helix.h"

const char *startFilePath="/";
const char* ext="mp3";
AudioSourceSD source(startFilePath, ext, PIN_AUDIO_KIT_SD_CARD_CS);
AudioBoardStream kit(AudioKitEs8388V2);
MP3DecoderHelix decoder;  // or change to MP3DecoderMAD
AudioPlayer player(source, kit, decoder);

void next(bool, int, void*) {
  Serial.println("next");
   player.next();
}

void previous(bool, int, void*) {
  Serial.println("prev");
   player.previous();
}

void startStop(bool, int, void*) {
  Serial.println("startstop");
   player.setActive(!player.isActive());
}

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // setup output
  auto cfg = kit.defaultConfig(TX_MODE);
  // sd_active is setting up SPI with the right SD pins by calling 
  // SPI.begin(PIN_AUDIO_KIT_SD_CARD_CLK, PIN_AUDIO_KIT_SD_CARD_MISO, PIN_AUDIO_KIT_SD_CARD_MOSI, PIN_AUDIO_KIT_SD_CARD_CS);
  cfg.sd_active = true;
  kit.begin(cfg);

  // setup additional buttons 
  kit.addAction(kit.getKey(1), startStop);
  kit.addAction(kit.getKey(4), next);
  kit.addAction(kit.getKey(3), previous);

  // setup player
  player.setVolume(0.7);
  player.begin();

  // select file with setPath() or setIndex()
  //player.setPath("/ZZ Top/Unknown Album/Lowrider.mp3");
  //player.setIndex(1); // 2nd file

}

void loop() {
  player.copy();
  kit.processActions();

}
```

### Other Steps to Reproduce

I checked in DriversPins.h file and also used digitalRead(36) function. digitalRead did work and gave output like 0 and 1 when pressed or not. But the actions do not get fired.

### What is your development environment

Arduino IDE

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

robert1x2, 2024-02-10:
I checked this documationtation as well: https://docs.ai-thinker.com/en/esp32-audio-kit

pschatzmann, 2024-02-11:
You are right: The redesign introduced quite a few issues:
- The pins are active low -> I corrected the settings in the audio-driver library
- This was breaking the SD from working because of some pin conflicts on this board, so I do not activate the conflicting keys if the sd is active.
- In your case, the action did not get fired because the debounceTimeout variable was not initialized properly. 

I took the opportunity to change the default action registration. In the old logic, they were set up automatically. Now they will need to be added with kit.addDefaultActions(); 

The changes have been committed.

closed 2024-02-11 by pschatzmann

robert1x2, 2024-02-11:
Thank you. The keys seem to come up but still I hasitate to get the right board config. I think the PIN assignment still is wrong with the defaults on my end. Do you see a good option to identify the key GPIOs and ... there is some pin 13 issue I am screwed. Its not playing any music from the card.

I added the new addDefaultActions() function here:

```
  cfg.sd_active = true;
  kit.begin(cfg);

  kit.addDefaultActions();

  // setup additional buttons 
  kit.addAction(kit.getKey(1), startStop);
  kit.addAction(kit.getKey(4), next);
  kit.addAction(kit.getKey(3), previous);
```
And this is output: :-/
```
13:36:20.293 -> [I] AudioBoardStream.h : 101 - static void audio_tools::AudioBoardStream::actionVolumeUp(bool, int, void*)
13:36:20.330 -> Error:   ->p_wire->endTransmission: 2
13:36:21.320 -> Error:   ->p_wire->requestFrom 1->0
13:36:21.320 -> [I] AudioBoardStream.h : 92 - incrementVolume: 0.000000 -> 2.000000
13:36:21.320 -> Error:   ->p_wire->endTransmission: 2
13:36:21.320 -> Error:   ->p_wire->endTransmission: 2
13:36:21.320 -> Error:   ->p_wire->endTransmission: 2
13:36:21.362 -> Error:   ->p_wire->endTransmission: 2
13:36:21.362 -> Error:   ->p_wire->endTransmission: 2
13:36:21.362 -> Error:   ->p_wire->endTransmission: 2
13:36:21.395 -> [I] StreamCopy.h : 143 - StreamCopy::copy  1024 -> 1024 -> 1024 bytes - in 1 hops
13:36:21.427 -> [I] StreamCopy.h : 143 - StreamCopy::copy  1024 -> 1024 -> 1024 bytes - in 1 hops
13:36:21.462 -> next
13:36:21.462 -> [I] AudioPlayer.h : 509 - void audio_tools::AudioPlayer::writeEnd()
13:36:21.462 -> [I] Fade.h : 106 - faded out 1152 frames to volume 0.000000
13:36:21.541 -> [I] StreamCopy.h : 143 - StreamCopy::copy  1024 -> 1024 -> 1024 bytes - in 1 hops
13:36:21.541 -> [I] AudioSourceSD.h : 63 - nextStream: 1
13:36:21.541 -> [I] AudioSourceSD.h : 68 - selectStream: 2
13:36:21.541 -> [I] AudioSourceSD.h : 72 - Using file /sound3.mp3
13:36:21.578 -> [I] AudioEncoded.h : 356 - virtual void audio_tools::EncodedAudioOutput::end()
13:36:21.578 -> [I] MetaDataID3.h : 575 - virtual void audio_tools::MetaDataID3::end()
13:36:21.578 -> [I] AudioPlayer.h : 204 - reset codec
13:36:21.578 -> [I] MetaDataID3.h : 569 - virtual void audio_tools::MetaDataID3::begin()
13:36:21.578 -> [I] StreamCopy.h : 74 - buffer_size=1024
13:36:21.610 -> [I] AudioPlayer.h : 233 - sample_rate: 44100
13:36:21.610 -> [I] AudioPlayer.h : 234 - bits_per_sample: 16
13:36:21.610 -> [I] AudioPlayer.h : 235 - channels: 2
13:36:21.610 -> [I] AudioTypes.h : 104 - sample_rate: 44100
13:36:21.610 -> [I] AudioTypes.h : 105 - channels: 2
13:36:21.610 -> [I] AudioTypes.h : 106 - bits_per_sample: 16
13:36:21.610 -> [I] I2SCodecStream.h : 83 - virtual void audio_tools::I2SCodecStream::setAudioInfo(audio_tools::AudioInfo)
13:36:21.645 -> [I] AudioTypes.h : 104 - sample_rate: 44100
13:36:21.645 -> [I] AudioTypes.h : 105 - channels: 2
13:36:21.645 -> [I] AudioTypes.h : 106 - bits_per_sample: 16
13:36:21.645 -> Warning: Pin '13' not set up because of conflict
13:36:22.617 -> Error:   ->p_wire->endTransmission: 5
13:36:22.649 -> Error:   ->p_wire->endTransmission: 2
13:36:22.649 -> Error:   ->p_wire->endTransmission: 2
13:36:22.649 -> Error:   ->p_wire->endTransmission: 2
13:36:22.649 -> Error:   ->p_wire->endTransmission: 2
```

pschatzmann, 2024-02-11:
If I2C (Wire) is giving error messages, then the most likely root cause is, that you did not specify the correct board.

pschatzmann, 2024-02-11:
I just discovered that the AudioKitEs8388V2 has some i2c pin conflicts with a button. I extended the checks and changed the sequence that I activate the buttons first and then i2c and spi. This makes sure that at least i2c and spi will work when there is a conflict!

robert1x2, 2024-02-11:
Dear @pschatzmann thank you a lot. Your fix and the v1 hint solved my issues. I tought test using v1 but probably got confused the bug.