pschatzmann/arduino-audio-tools #1406: How to Read Mp3 file form URL and play on I2S

GoFuuu, 2024-03-01:
### Problem Description

Thank you for the wonderful library!
I am currently trying out one of the example project: "streams-url_mp3_mad-i2s“.And I'm going to use the ESP32 to play the static mp3 on my server

The only changes done to the sketch are:

> Pin-out for the i2S
> WIFI credentials
> URL which points to a static mp3 file on my web server.

 I've already tested the url from the example("http://stream.srg-ssr.ch/m/rsj/mp3_128") .It works ,but when try to ues my own URL which points to a static mp3 file on my web server. It can't work and reboot again and again.

I guess it because run out of memory because the output of the serial monitor :

>  "Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled."

I'm wondering if it might be because I need streaming URL rather than URL that point to fixed mp3 files.

How can I play the static mp3 file on ESP32 successfully?

here is the full serial output info:

`ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[I] AudioTypes.h : 104 - sample_rate: 44100
[I] AudioTypes.h : 105 - channels: 2
[I] AudioTypes.h : 106 - bits_per_sample: 16
[I] I2SConfigESP32.h : 90 - rx/tx mode: TX_MODE
[I] I2SConfigESP32.h : 91 - port_no: 0
[I] I2SConfigESP32.h : 92 - is_master: Master
[I] I2SConfigESP32.h : 93 - sample rate: 44100
[I] I2SConfigESP32.h : 94 - bits per sample: 16
[I] I2SConfigESP32.h : 95 - number of channels: 2
[I] I2SConfigESP32.h : 96 - signal_type: Digital
[I] I2SConfigESP32.h : 98 - i2s_format: I2S_STD_FORMAT
[I] I2SConfigESP32.h : 100 - auto_clear: true
[I] I2SConfigESP32.h : 102 - use_apll: true
[I] I2SConfigESP32.h : 107 - buffer_count:6
[I] I2SConfigESP32.h : 108 - buffer_size:512
[I] I2SConfigESP32.h : 113 - pin_bck: 27
[I] I2SConfigESP32.h : 115 - pin_ws: 32
[I] I2SConfigESP32.h : 117 - pin_data: 33
[I] AudioEncoded.h : 269 - virtual void audio_tools::EncodedAudioOutput::setNotifyAudioChange(audio_tools::AudioInfoSupport&)
[I] URLStream.h : 86 - virtual bool audio_tools::URLStream::begin(const char*, const char*, MethodID, const char*, const char*): 139.155.158.174:8001/media/outaudio/apologies.mp3
[I] Url.h : 74 - Url::parse
[I] URLStream.h : 417 - void audio_tools::URLStream::login()
.
[I] URLStream.h : 392 - WiFiClient
[I] HttpRequest.h : 233 - Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008af45  PS      : 0x00060730  A0      : 0x801820e0  A1      : 0x3ffb1d20
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffb1ff0
A10     : 0x00000003  A11     : 0x00060723  A12     : 0x00060720  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000017  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4008af45  LEND    : 0x4008af55  LCOUNT  : 0xffffffff  

Backtrace: 0x4008af42:0x3ffb1d20 0x401820dd:0x3ffb1d30 0x4017f025:0x3ffb2040 0x400d8cc0:0x3ffb2100 0x400d3c45:0x3ffb2130 0x400d821a:0x3ffb2160 0x400d8378:0x3ffb21a0 0x400d84cd:0x3ffb21c0 0x400e58a2:0x3ffb2290

ELF file SHA256: 3380edbdc0047efa

Rebooting...`

### Device Description

ESP32 WROOM

### Sketch

```cpp
/**
 * @file streams-url_mp3-i2s.ino
 * @author Phil Schatzmann
 * @brief decode MP3 stream from url and output it on I2S
 * @version 0.1
 * @date 2021-96-25
 * 
 * @copyright Copyright (c) 2021
 */

// install https://github.com/pschatzmann/arduino-libmad.git

#include "AudioTools.h"
#include "AudioCodecs/CodecMP3MAD.h"

URLStream url("0329","123456789");
I2SStream i2s; // final output of decoded stream
EncodedAudioStream dec(&i2s, new MP3DecoderMAD()); // Decoding stream
StreamCopy copier(dec, url); // copy url to decoder

void setup(){
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);  

  // setup i2s
  auto config = i2s.defaultConfig(TX_MODE);
  // you could define e.g your pins and change other settings
  config.pin_bck = 27;
  config.pin_ws = 32;
  config.pin_data = 33;
  //config.i2s_format = I2S_STD_FORMAT;
  i2s.begin(config);

  // setup I2S based on sampling rate provided by decoder
  dec.setNotifyAudioChange(i2s);
  dec.begin();

// mp3 radio
  //url.begin("http://stream.srg-ssr.ch/m/rsj/mp3_128","audio/mp3");
  url.begin("139.155.158.174:8001/media/outaudio/apologies.mp3","audio/mp3");

}

void loop(){
  copier.copy();
}
```

### What is your development environment

PIO

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-03-01:
You should try to analyse where the issue is!

- Decode the stack-trace: that will lead you to the line of code that is failing.
- Set the log level to debug: that will also give you a hint how far the processing goes successfully

I am never using MAD myself and prefer LibHelix. You might also try that...

GoFuuu, 2024-03-03:
I found that it was my network error

closed 2024-03-03 by GoFuuu