pschatzmann/arduino-audio-tools #1409: AACEncoderFDK.h : 234 - Encoding failed: Handle passed to function call was invalid

franki29, 2024-03-02:
### Problem Description

Using the sketch above I get following error:[

I] StreamCopy.h : 356 - try write  - 21 (open 1024 bytes) 
[E] AACEncoderFDK.h : 234 - Encoding failed: Handle passed to function call was invalid

### Device Description

ESP32 DEV Module

### Sketch

```cpp
**
 * @file streams-i2s-webserver_wav.ino
 *
 *  This sketch reads sound data from I2S. The result is provided as WAV stream which can be listened to in a Web Browser
 *
 * @author Phil Schatzmann
 * @copyright GPLv3
 */
//#define USE_FDK
#include "AudioTools.h"
#include "AudioCodecs/CodecAACFDK.h"

AudioEncoderServer server(new AACEncoderFDK(),"ssid","password");
I2SStream i2sStream; // Access I2S as stream

void setup() {
Serial.begin(115200);
AudioLogger::instance().begin(Serial, AudioLogger::Info);
// start i2s input with default configuration
Serial.println("starting I2S...");
auto config = i2sStream.defaultConfig(RX_MODE);
config.i2s_format = I2S_STD_FORMAT; // if quality is bad change to I2S_MSB_FORMAT #23
config.sample_rate = 44100;
config.channels = 2;
config.is_master = 1;
config.bits_per_sample = 16;
config.use_apll = true;
config.fixed_mclk = 0 ;
config.pin_mck = 0;
i2sStream.begin(config);
Serial.println("I2S started");

// start data sink
server.begin(i2sStream, config);
}

// Arduino loop
void loop() {
// Handle new connections
  server.copy();
}
```

### What is your development environment

Arduino IDE 2.3.2

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-03-02:
My gut feeling is that you are just running out of memory. I think using WIFI with more recent ESP32 core versions just does not provide enough free RAM any more.
Did you try the Decoder w/o WIFI ?

pschatzmann, 2024-03-02:
Just added a [test case to confirm that the codec is working](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/tests/codecs/test-codec-aac-fdk/test-codec-aac-fdk).

So it must be the limited RAM...

pschatzmann, 2024-03-02:
If you have PSRAM you can get it to work by [using PSRAM](https://www.pschatzmann.ch/home/2022/05/30/esp32-and-psram/). 

I got it to work with e.g.   heap_caps_malloc_extmem_enable(1024);

pschatzmann, 2024-03-02:
Actually, comparing with my [other examples](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/streams-audiokit-webserver_aac/streams-audiokit-webserver_aac.ino), you just forgot to configure the codec to use less RAM

closed 2024-03-02 by pschatzmann