pschatzmann/arduino-audio-tools #1411: Waveshare ESP32-C3 Zero running player-url-i2s

vbguyny, 2024-03-03:
### Problem Description

Waveshare ESP32-C3 Zero is crashing when I try to run the example: https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-player/player-url-i2s/player-url-i2s.ino

I have the kit connected to a MAX98357A.

```
�#�?�/
Core  0 register dump:
MEPC    : 0x4200300c  RA      0x42003000
3fc925b0: 0x00000014 0x3fc92540 0x3fc923c8 0x42003000 0x00000000 0x3fc92540 0x3fc923c8 0x42003000
3fc925d0: 0x3fc9263c 0x3fc92540 0x3fc923c8 0x42003000 0x00000010 0x3fc92540 0x3fc923c8 0x42003000
3fc925f0: 0x00000000 0x3fc92540 0x3fc923c8 0x42003000 0x3c0f16e4 0x3fc92540 0x3fc923c8 0x42003000
3fc92610: 0x3c0f148c 0x3fc92540 0x3fc923c8 0x42003000 0x3c0f148c 0x3fc92540 0x3fc923c8 0x42003000
3fc92630: 0x00000000 0x3fc92540 0x3fc923c8 0x42003000 0x00000000 0x3fc92540 0x3fc923c8 0x42003000
3fc92650: 0x3c0f1dfc 0x3fc92540 0x3fc923c8 0x42003000 0x3fc9dde4 0x3fc92540 0x3fc923c8 0x42003000
3fc92670: 0x00000010 0x3fc92540 0x3fc923c8 0x42003000 0x00000000 0x3fc92540 0x3fc923c8 0x42003000
3fc92690: 0x00000000 0x3fc92540 0x3fc923c8 0x42003000 0x00000014 0x3fc92540 0x3fc923c8 0x42003000
3fc926b0: 0x00000000 0x3fc92540 0x3fc923c8 0x42003000 0x00000000 0x3fc92540 0x3fc923c8 0x42003000
3fc926d0: 0x00000000 0x3fc92540 0x3fc923c8 0x42003000 0x00000000 0x3fc92540 0x3fc923c8 0x42003000
3fc926f0: 0x00000000 0x3fc92540 0x3fc923c8 0x42003000 0x3c0f16d4 0x3fc92540 0x3fc923c8 0x42003000
3fc92710: 0x3fc92418 0x3fc92540 0x3fc923c8 0x42003000 0x3c0f1c40 0x3fc92540 0x3fc923c8 0x42003000
3fc92730: 0x00000000 0x3fc92540 0x3fc923c8 0x42003000 0x00000000 0x3fc92540 0x3fc923c8 0x42003000
3fc92750: 0x00000000 0x3fc92540 0x3fc923c8 0x42003000 0x00000000 0x3fc92540 0x3fc923c8 0x42003000
3fc92770: 0x00000000 0x3fc92540 0x3fc923c8 0x42003000 0x00000000 0x3fc92540 0x3fc923c8 0x42003000
3fc92790: 0x00000000 0x3fc92540 0x3fc923c8 0x42003000 0x00000000 0x3fc92540 0x3fc923c8 0x42003000
3fc927b0: 0x00000000 0x3fc92540 0x3fc923c8 0x42003000 0x00000000 0x3fc92540 0x3fc923c8 0x42003000

ELF file SHA256: c823c93f00300042

Re-entered core dump! Exception happened during core dump!
Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382230
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x25f4
entry 0x403cc710
```

### Device Description

Waveshare ESP32-C3 Zero

### Sketch

```cpp
...

URLStream urlStream(wifi, password);
AudioSourceURL source(urlStream, urls, "audio/mp3");
I2SStream i2s;
MP3DecoderHelix decoder;
AudioPlayer player(source, i2s, decoder); // <- Crashes here

...
```
```

### What is your development environment

PlatformIO

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-03-03:
I was doing some major cleanup for the codecs. It seems that I have introduced an endless loop here.
Can you try with the latest version of this library ?

vbguyny, 2024-03-05:
@pschatzmann  thanks! It appears to have fixed the crash. But I don't hear anything out of my device. But that might because I don't know what I'm doing 😵‍💫

pschatzmann, 2024-03-05:
When you set the log level to info, what output do you get ?
If you see constently occuring copy lines, the data is processed properly and you need to double check your pin connections....

closed 2024-03-08 by pschatzmann