pschatzmann/arduino-audio-tools #1430: "missing initializer for member" warning on ESP32, espidf, platformio

codegrue, 2024-03-08:
### Problem Description

Compiling on platformio with espidf, this library generates hundreds of these warnings:

.pio/libdeps/project/audio-tools/src/AudioI2S/I2SESP32.h:183:7: warning: missing initializer for member 'i2s_driver_config_t::total_chan' [-Wmissing-field-initializers]

### Device Description

ESP32 development board

### Sketch

```cpp
just including the library in any project causes this at compile time

platformio.ini:

[env]
framework = arduino, espidf
platform = espressif32
```

### What is your development environment

VSCode/Platformio, Windows

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-03-08:
Please read the documentation in the Wiki and at least indicate why you don't follow the instructions.
It would also help if you ESP32 core version that pulled by your installation: 2.0.14 would be the most recent version!

closed 2024-03-08 by pschatzmann

codegrue, 2024-03-09:
I'm not sure what part of the wiki you were alluding to, but I reviewed it again and inferred this, which seems to have solved the warnings:

build_flags = -Wno-missing-field-initializers

pschatzmann, 2024-03-09:
There is a chapter for PlatfomIO: https://github.com/pschatzmann/arduino-audio-tools/wiki/Working-with-PlatformIO