pschatzmann/arduino-audio-tools #1451: Example "streams-audiokit-webserver_aac" plays faster then pause

dl9sec, 2024-03-17:
### Problem Description

Hi Phil,

first of all, thank you very much for the great audio libraries stuff and your effort.

My intention is to get an "analog audio to web stream" device, where I found the "streams-audiokit-webserver_aac" example most suitable. After coming through the same problem described at #1442, I got audio with the latest commit of the "arduino-audio-tools.

The problem now is, that the streamed audio (streamed with latest Windows Firefox or Chrome browser) seems to be played twice the speed, then (maybe if buffer is empty) a pause, the again audio twice the speed, then a pause and so on. There is currently no continous audio stream.

The serial log looks like this ("AudioLogger::instance().begin(Serial, AudioLogger::Info);"):

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
starting AudioKit...
Warning: Pin '13' not set up because of conflict
[I] AudioTypes.h : 104 - sample_rate: 44100
[I] AudioTypes.h : 105 - channels: 2
[I] AudioTypes.h : 106 - bits_per_sample: 16
[I] I2SConfigESP32.h : 90 - rx/tx mode: RX_MODE
[I] I2SConfigESP32.h : 91 - port_no: 0
[I] I2SConfigESP32.h : 92 - is_master: Master
[I] I2SConfigESP32.h : 93 - sample rate: 44100
[I] I2SConfigESP32.h : 94 - bits per sample: 16
[I] I2SConfigESP32.h : 95 - number of channels: 2
[I] I2SConfigESP32.h : 96 - signal_type: Digital
[I] I2SConfigESP32.h : 98 - i2s_format: I2S_STD_FORMAT
[I] I2SConfigESP32.h : 100 - auto_clear: false
[I] I2SConfigESP32.h : 102 - use_apll: true
[I] I2SConfigESP32.h : 107 - buffer_count:6
[I] I2SConfigESP32.h : 108 - buffer_size:512
[I] I2SConfigESP32.h : 111 - pin_mck: 0
[I] I2SConfigESP32.h : 113 - pin_bck: 27
[I] I2SConfigESP32.h : 115 - pin_ws: 25
[I] I2SConfigESP32.h : 117 - pin_data: 35
AudioKit started
[I] AACEncoderFDK.h : 335 - aacEncOpen

[I] C:\Users\thors\Documents\Arduino\libraries\arduino-fdk-aac\src\libSYS\genericStds.cpp : 210 - ==> calloc_align_1(1,600) -> 0x3ffb5aa8 [available MEMORY 8BIT : 4415915 ; 32BIT : 4455579]
[I] C:\Users\thors\Documents\Arduino\libraries\arduino-fdk-aac\src\libSYS\genericStds.cpp : 210 - ==> calloc_align_1(2048,2) -> 0x3ffae89c [available MEMORY 8BIT : 4411803 ; 32BIT : 4451467]
[I] C:\Users\thors\Documents\Arduino\libraries\arduino-fdk-aac\src\libSYS\genericStds.cpp : 210 - ==> calloc_align_1(1,268) -> 0x3ffb5d10 [available MEMORY 8BIT : 4411519 ; 32BIT : 4451183]
[I] C:\Users\thors\Documents\Arduino\libraries\arduino-fdk-aac\src\libSYS\genericStds.cpp : 210 - ==> calloc_align_4(1,85628) -> 0x3f800884 [available MEMORY 8BIT : 4325875 ; 32BIT : 4365539]
[I] C:\Users\thors\Documents\Arduino\libraries\arduino-fdk-aac\src\libSYS\genericStds.cpp : 210 - ==> calloc_align_1(1,3808) -> 0x3ffbbca0 [available MEMORY 8BIT : 4322051 ; 32BIT : 4361715]
[I] C:\Users\thors\Documents\Arduino\libraries\arduino-fdk-aac\src\libSYS\genericStds.cpp : 210 - ==> calloc_align_1(1,8) -> 0x3ffb5e2c [available MEMORY 8BIT : 4322027 ; 32BIT : 4361691]
[I] C:\Users\thors\Documents\Arduino\libraries\arduino-fdk-aac\src\libSYS\genericStds.cpp : 210 - ==> calloc_align_1(1,8) -> 0x3ffb5e44 [available MEMORY 8BIT : 4322003 ; 32BIT : 4361667]
[I] C:\Users\thors\Documents\Arduino\libraries\arduino-fdk-aac\src\libSYS\genericStds.cpp : 210 - ==> calloc_align_1(1,3540) -> 0x3ffcb8bc [available MEMORY 8BIT : 4318447 ; 32BIT : 4358111]
[I] C:\Users\thors\Documents\Arduino\libraries\arduino-fdk-aac\src\libSYS\genericStds.cpp : 210 - ==> calloc_align_1(2048,2) -> 0x3ffcc6a0 [available MEMORY 8BIT : 4314335 ; 32BIT : 4353999]
[I] C:\Users\thors\Documents\Arduino\libraries\arduino-fdk-aac\src\libSYS\genericStds.cpp : 210 - ==> calloc_align_1(1,3540) -> 0x3ffcd6b0 [available MEMORY 8BIT : 4310779 ; 32BIT : 4350443]
[I] C:\Users\thors\Documents\Arduino\libraries\arduino-fdk-aac\src\libSYS\genericStds.cpp : 210 - ==> calloc_align_1(2048,2) -> 0x3ffce494 [available MEMORY 8BIT : 4306667 ; 32BIT : 4346331]
[I] C:\Users\thors\Documents\Arduino\libraries\arduino-fdk-aac\src\libSYS\genericStds.cpp : 210 - ==> calloc_align_1(1,64) -> 0x3ffb5e5c [available MEMORY 8BIT : 4306587 ; 32BIT : 4346251]
[I] C:\Users\thors\Documents\Arduino\libraries\arduino-fdk-aac\src\libSYS\genericStds.cpp : 210 - ==> calloc_align_1(1,2120) -> 0x3ffcf4a4 [available MEMORY 8BIT : 4304451 ; 32BIT : 4344115]
[I] C:\Users\thors\Documents\Arduino\libraries\arduino-fdk-aac\src\libSYS\genericStds.cpp : 210 - ==> calloc_align_1(1,2120) -> 0x3ffcfcfc [available MEMORY 8BIT : 4302315 ; 32BIT : 4341979]
[I] C:\Users\thors\Documents\Arduino\libraries\arduino-fdk-aac\src\libSYS\genericStds.cpp : 210 - ==> calloc_align_1(1,256) -> 0x3ffb5eac [available MEMORY 8BIT : 4302043 ; 32BIT : 4341707]
[I] C:\Users\thors\Documents\Arduino\libraries\arduino-fdk-aac\src\libSYS\genericStds.cpp : 210 - ==> calloc_align_1(1,256) -> 0x3ffb5fbc [available MEMORY 8BIT : 4301771 ; 32BIT : 4341435]
[I] C:\Users\thors\Documents\Arduino\libraries\arduino-fdk-aac\src\libSYS\genericStds.cpp : 210 - ==> calloc_align_1(1,160) -> 0x3ffb60cc [available MEMORY 8BIT : 4301595 ; 32BIT : 4341259]
[I] C:\Users\thors\Documents\Arduino\libraries\arduino-fdk-aac\src\libSYS\genericStds.cpp : 210 - ==> calloc_align_1(1,2020) -> 0x3ffd0554 [available MEMORY 8BIT : 4299559 ; 32BIT : 4339223]
[I] C:\Users\thors\Documents\Arduino\libraries\arduino-fdk-aac\src\libSYS\genericStds.cpp : 210 - ==> calloc_align_1(1,2020) -> 0x3ffd0d48 [available MEMORY 8BIT : 4297523 ; 32BIT : 4337187]
[I] C:\Users\thors\Documents\Arduino\libraries\arduino-fdk-aac\src\libSYS\genericStds.cpp : 210 - ==> calloc_align_1(1,96) -> 0x3ffb617c [available MEMORY 8BIT : 4297411 ; 32BIT : 4337075]
[I] C:\Users\thors\Documents\Arduino\libraries\arduino-fdk-aac\src\libSYS\genericStds.cpp : 210 - ==> calloc_align_1(1,104) -> 0x3ffb61ec [available MEMORY 8BIT : 4297291 ; 32BIT : 4336955]
[I] C:\Users\thors\Documents\Arduino\libraries\arduino-fdk-aac\src\libSYS\genericStds.cpp : 210 - ==> calloc_align_1(1,92) -> 0x3ffb6264 [available MEMORY 8BIT : 4297183 ; 32BIT : 4336847]
[I] C:\Users\thors\Documents\Arduino\libraries\arduino-fdk-aac\src\libSYS\genericStds.cpp : 210 - ==> calloc_align_1(1,92) -> 0x3ffb62d0 [available MEMORY 8BIT : 4297075 ; 32BIT : 4336739]
[... 71 lines cut ...]
Another interesting thing ist the "Warning: Pin '13' not set up because of conflict" although "config.sd_active = false;" (shouldn't the warning disappear if SD card is deactivated?).

What am I doing wrong?

Thanks and regards,
Thorsten

### Device Description

ESP32 Audio Kit V2.2 A402 (seems to be "AudioKitEs8388V1", no I2C/I2S errors with that)

### Sketch

```cpp
https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/streams-audiokit-webserver_aac/streams-audiokit-webserver_aac.ino
```

### What is your development environment

Arduino 1.8.19, latest companion libraries (arduino-audio-driver, arduino-fdk-aac) for arduino-audio-tools installed.
Espressif ESP32 2.0.14, Board: "ESP Dev Module", PSRAM: enabled, Partition scheme: Huge App, Rest: defaults.

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-03-17:
I could not reproduce your issue, however I was testing with an updated example and the latest version of the decoder which supports psram now.

dl9sec, 2024-03-17:
Hmmm. ok. Thank you very much, let me know, if you need further information.
FYI: I disassembled both mics, audio is coming from LINEIN. The WAV streamer example works just fine (apart from a slightly crackling noise in audio).
Updated the arduino-audio-driver library, the warning disappeared, great stuff :-)

pschatzmann, 2024-03-17:
Somehow this does not make any sense and your log does not match with the [example](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/streams-audiokit-webserver_aac/streams-audiokit-webserver_aac.ino)!
```
[I] I2SConfigESP32.h : 102 - use_apll: true
[I] I2SConfigESP32.h : 107 - buffer_count:6
[I] I2SConfigESP32.h : 108 - buffer_size:512
[I] I2SConfigESP32.h : 111 - pin_mck: 0
[I] I2SConfigESP32.h : 113 - pin_bck: 27
[I] I2SConfigESP32.h : 115 - pin_ws: 25
[I] I2SConfigESP32.h : 117 - pin_data: 35
AudioKit started
.
IP address: 192.168.1.33
Server started
```

closed 2024-03-17 by pschatzmann

dl9sec, 2024-03-17:
Yes, I used the example before your commit df5292b and didn't realize, that there are slight modifications.
Brought all to the latest versions, now it works.
Thanks.