pschatzmann/arduino-audio-tools #1467: Any sketch using mixer causes ESP32-S3-Wroom1 to crash

MJenkings, 2024-03-27:
### Problem Description

I cannot use the mixer function in any method as it causes the ESP32-S3 to crash in a loop without running the sketch, which means I get no error log.   I can successfully play all media types from memory or SD card if I don't use the mixer so I know that my hardware setup works and that the basic structure of the sketch is fine.  My aim is to play multiple PCM files through the mixer  (3 max) from an SD card.  All the sample sketches for single streams seem to work straight away but I have not yet manage to get a sketch with the mixer to even run. 

The serial output during the crash loop is below, this loops approx. once per second 

10:00:50.386 -> ESP-ROM:esp32s3-20210327
10:00:50.387 -> Build:Mar 27 2021
10:00:50.387 -> rst:0x8 (TG1WDT_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
10:00:50.387 -> Saved PC:0x40379730
10:00:50.387 -> SPIWP:0xee
10:00:50.387 -> mode:DIO, clock div:1
10:00:50.387 -> load:0x3fce3808,len:0x44c
10:00:50.420 -> load:0x403c9700,len:0xbd8
10:00:50.421 -> load:0x403cc700,len:0x2a80
10:00:50.421 -> entry 0x403c98d0

Many thanks

### Device Description

ESP32-S3-Wroom1 with 16Mb Flash - Audio chip is a MAX98357A, 

### Sketch

```cpp
/**
 * @file streams-mixer-audiokit.ino
 * @author Phil Schatzmann
 * @brief Simple Demo for mixing 2 input streams
 * @version 0.1
 * @date 2022-11-15
 *
 * @copyright Copyright (c) 2022
 *
 */
#include "AudioTools.h"
#include "AudioLibs/AudioKit.h"
#include "drums.h"
#include "guitar.h"

InputMixer<int16_t> mixer;
AudioKitStream kit;
MemoryStream drums(drums_raw, drums_raw_len);
MemoryStream guitar(guitar_raw, guitar_raw_len);
StreamCopy copier(kit, mixer);

void setup() {

  // auto restart when MemoryStream has ended
  drums.setLoop(true);
  guitar.setLoop(true);

  // setup output
  auto cfg = kit.defaultConfig(TX_MODE);
  cfg.pin_ws = 14;
  cfg.pin_bck = 13;
  cfg.pin_data = 11;
  cfg.channels = 1;
  cfg.sample_rate = 8000;
  kit.begin(cfg);
  // max volume
  kit.setVolume(1.0);

  mixer.add(drums);
  mixer.add(guitar);
  mixer.begin(cfg);

}

void loop() { copier.copy(); }
```

### What is your development environment

Arduino IDE

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2024-03-27:
Activate the logger: do you get any error messages.
If not set the log level to info or debug...

I don't quite understand why you use the AudioKitStream with an ESP32-S3!

closed 2024-03-27 by pschatzmann

MJenkings, 2024-03-27:
Thank you that has solved it, my fault completely. 
I swapped the line "AudioKitStream kit;" to "I2SStream kit;" and it works